mu-editor/mu #829: Pre-installed packages cannot be upgraded using 3rd party package installer

martinohanlon, 2019-05-08:
If you are reporting a bug, we would like to know:

* What you were trying to do,
Install latest version of guizero

* What steps you took to make this happen,
 + Opened Mu Administration
 + Added guizero to "3rd party packages"
 + Tried to use latest version of guizero

* What you expected to happen,
That I would be able to use the latest version of guizero

* What actually happened,
The version of guizero was still the old version that was pre-installed with Mu

* Why this difference is problematic (it may not be a bug!),
I am unable to use the new features.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
 + Mu 1.0.0.alpha.1

I am assuming that Mu continues to use the "default" installed version of guizero rather than the version installed by the package manager.

martinohanlon, 2019-06-17:
> However, I intend to remove all the non-essential packages from Mu in 1.1. I.e. if you want guizero (which Mu doesn't need to run itself), then you should install as a package.

In hindsight I think this is the right approach.

It delivers consistency across all platforms and install mechanisms.

ntoll, 2021-07-10:
I believe this is fixed..? We don't have `guizero` included in the Mu installer, but it's very easy to use Mu's package manager to install the latest version should you need it.

Closing (and many thanks for the contributions folks). :+1:

closed 2021-07-10 by ntoll