mu-editor/mu #832: i18n: QMessageBox buttons are not translated

wayoda, 2019-05-09:
I'm running Mu in the german locale `de_DE`. While all the titles and the information presented to the user in dialogs are translated, this is not he case for places in the code that make use of a `QMessageBox`

For instance this is the diaolg that is shown to a (german) user when he quits Mu while there are unsaved changes:    
[image]
As you can see the labels on the buttons read `Ok` and `Cancel` while it should be `Ok` and `Abbrechen`. 

To make this work all the code that displays a QMessageBox should set the text on the StandardButtons. Here is piece of code from the quit-dialog that allows us to translate the button-text in `mu/locale/de_DE/LC_MESSAGES/mu.po` for german users. 

``` 
message_box = QMessageBox(self)
message_box.setText(message)
message_box.setWindowTitle(_('Mu'))
if information:
    message_box.setInformativeText(information)
if icon and hasattr(message_box, icon):
    message_box.setIcon(getattr(message_box, icon))
else:
    message_box.setIcon(message_box.Warning)
    message_box.setStandardButtons(message_box.Cancel | message_box.Ok)
    message_box.setDefaultButton(message_box.Cancel)
    message_box.button(message_box.Ok).setText(_('Yes, please.'))  
    message_box.button(message_box.Cancel).setText(_('No, thanks.')) 
```
The last 2 lines make this happen. The dialog now looks like this (and can later be translated for other languages by updating the po file ) 
[image]

I choose the actual text on the buttons just to illustrate he problem : -)

Cheers 
Eberhard

wayoda, 2019-05-09:
The same problem applies to  ` QDialogButtonBox` instances used in the code. 
This could probably be solved by sub-classing the ` QDialogButtonBox` and translating the ButtonText in just one place.

tmontes, 2019-05-09:
On a nearly-related topic, Qt UI localisation, there's also #756, about the context-menus.

wayoda, 2019-05-09:
Oops, didn't see this because I always use keyboard shortcuts in the Editor. But I will have a look (on Linux)

wayoda, 2019-05-09:
Got this working by loading the translations shipped with the PyQt-Lib manually. But I'm not sure if this works for everybody (different OS, all languages) I get errors for instance for locale `fr_FR` while its works for `de_DE` and `es_ES`. Needs some investigation. Here is some code I placed in `mu/app.py` to make it work.

```
   # The app object is the application running on your computer.
    app = QApplication(sys.argv)
    #load default translations from Qt-Library
    translator = QTranslator()
    print(translator.load(QLocale.system(), "qtbase","_",  os.path.abspath(QLibraryInfo.location(QLibraryInfo.TranslationsPath)), ".qm"))
    print(translator.load(QLocale.system(), "qscintilla","_",  os.path.abspath(QLibraryInfo.location(QLibraryInfo.TranslationsPath)), ".qm"))
    app.installTranslator(translator)

```

wayoda, 2019-05-10:
I close this issue in favour of a broader discussion in #834

closed 2019-05-10 by wayoda