mu-editor/mu #834: Incomplete translation of the Mu user interface

wayoda, 2019-05-10:
### Incomplete translation of the Mu user interface
Hi, I dug a little deeper into the Mu sources to find out why some parts of the
user interface are not translated when Mu is run in a non-english environment.
This is a kind of summary what parts of the app are are affected and what are
the options to fully translate Mu. 

For the rest of this comment I will illustrate the problem by showing what are
the effects on the user when he runs Mu in either the `de_DE.utf8`-locale or in
the `pt_BR.utf8`-locale. The Mu interface is translated for both languages but
there are some slight differences between these two when Mu is run. 

#### Related issues on github
For me this all started out with #832 (now closed) which I opened because the `Ok` and
`Cancel` Buttons on MessageBoxes where *not* translated into german but showed
the english text. In the meantime I have closed this issue because it was just
an edgecase to the broader problem. 
And there is #756 by @tmontes which refers to another edgecase: *Context Menus
are not localized* which got me started to dig a little bit deeper...

#### What parts of the interface are affected?
1. QMessageBoxes/QDialogs that ask the user for confirmation of some action. (Like
   quitting Mu while there are unsaved files) The buttons should read `OK` and
`Abbrechen` in a german locale, but the english labels are shown.
2. QFileDialog for opening and saving files. The buttons should read `Öffnen` or
   `Speichern` again the english labels are shown.
3. The (right-click) context-menu in the `editor` pane
4. The (right-click) context-menu in the `REPL` pane

#### Where do translated strings come from?
There are 3 different sources for translations that must work together to
result in a fully localized Mu interface.
1. The `po`-files created by Mu developers. These are complete for german
   and portuguese (and a few other languages).
2. Default translations provided by the Qt-Lib itself that localize the
   QMessageBoxes and the QFileDialog. They are available for german but **not**
for portuguese :-(
3. Translations for the context-menu in the editor are provided by the
   `qscintilla` project, but **only** german and spanish versions are shipped
with the library. (There are resources for 3 more languages in `qscintilla` but they are only
partly translated and miss the translations for the context-menu)

#### Localizing `QMessageBoxes` and `QDialogs` 
There are two options for this:
1. Load the standard translations from the `Qt-Lib` and let Qt do the work. You
   would have to install a `QTranslator` in the app and load the resources
shipped with `Qt`. This works for german but **not** for portuguese because the
translation for this language is incomplete!
2. Set strings for the Buttonlabels manually in the `Mu`-code. This is easily possible
   and in my opinion the better solution. Requires some changes to the code but
will work for every language `Mu` is translated into.

#### Localizing `QFileDialog` 
The same two options that apply to messageboxes and plain dialogs are available for this component:
1. Load the standard translations from the `Qt-Lib` and let Qt do the work. You
   would have to install a `QTranslator` in the app and load the resources
shipped with `Qt`. This again works for german but **not** for portuguese because the
translation for this language is incomplete!
2. Set strings for the Buttonlabels manually in the `Mu`-code. 
There is a bit more work here to be done, because `Mu` uses the static member functions of the 
`QFileDialog` class (see https://doc.qt.io/qt-5/qfiledialog.html). You can
**not** update the labels through a static function, you need to create an
instance of `QFileDialog`. A bit more work, but not too hard.
After that it will work for every language `Mu` is translated into.

#### Localizing the editors context-menu
Since the `qscintilla`-lib does not allow direct access to the labels in the
context-menu we are at the mercy of the `qscintilla` developers.
We could load the existing translations for the component with a `QTranslator`
but this will **only** work for german and spanish! All other languages
will see the english labels.
> I found some code-fragment that replaces the context-menu with a custom one
> but haven't tried that yet. (see https://stackoverflow.com/questions/53905083/qscintilla-how-to-add-a-user-context-menu-to-textedit-c)

#### Localizing the REPL context-menu
This is an easy one!
The menu-items in the context-menu where simply not marked as translatable strings in `mu/interface/panes.py`. 
```
mu/interface/panes.py:163:        menu.addAction("Copy", self.copy, copy_keys)
mu/interface/panes.py:164:        menu.addAction("Paste", self.paste, paste_keys)
mu/interface/panes.py:804:        menu.addAction("Copy", self.copy, copy_keys)
mu/interface/panes.py:805:        menu.addAction("Paste", self.paste, paste_keys)
```
Once they are marked up with `_("Copy")` and `_("Paste") they can be translated
easily.

tmontes, 2019-05-11:
@wayoda, thanks for a great summary! I've just closed #756 to avoid duplication and reduce the number of outstanding issues, hoping to simplify issue triaging and the overall project's management.

dybber, 2021-02-06:
After investigating the part about using the build in Qt support with the QTranslator object, also mentioned above, but which we haven't enabled, I just want to write down what I learned. 

It seems like the process is a bit complicated, so I'm not sure we would actually benefit from it instead of just doing work-arounds such as suggested in #983

Here's a guide: https://doc.qt.io/Qt-5/internationalization.html
And another one on their wiki: https://wiki.qt.io/Qt_Localization

**Main steps**
 - We need to bundle translations files in a binary file-format called `.qm` with the applications
 - The `.qm` files are build from XML translation-files called `.ts`, which are available in this repository: https://github.com/qt/qttranslations
  - So we have to download these translations, generate .qm files as part of the build process, or just keep the binaries in the repository
  - Then we follow the steps in guide I linked above, creating a QTranslator object, loading the .qm file and installing the translator with the QApplication object
  
**Suggested next steps**
 - Create a script that downloads .ts files and build .qm files for the languages we support
 - Commit those .qm files to the repository (for every release, we should run the script again to update .qm files)
 - Introduce a QTranslator in `app.py` to load these translation files
 - Test that it works
 - As @wayoda mentions, not all languages are completely translated, and for any issues we would then have to make translators send their corrected translations to the https://github.com/qt/qttranslations repository
  
  
 Does the above seem like a sensible plan? Or do we rather want to avoid this complexity and go for a solution like in #983? I think we should focus on getting these Qt strings translated first and wait with the `qscintilla` issues, which are probably issues we should raise with the `qscintilla` directly instead of trying to fix them in our end.

ntoll, 2021-02-14:
Hi folks... just catching up on Mu things.

> It seems like the process is a bit complicated,

(Understatement of the year @dybber :slightly_smiling_face: ) For what it's worth, I went with Python's translation method rather than Qt's in the hope that Pythonistas would be more familiar with the i18n `gettext` module built into Python than Qt's own mechanism. I'm reticent to introduce two ways to do translation mostly for Zen-of-Python reasons and to avoid introducing yet another wrinkle.

It sounds to me that the "simplest" (of a set of complicated solutions) would be to drop the Python-based translations and move whole-sale over to the Qt manner of i18n. But this feels like a LOT of work and would require specialist knowledge and skills in order to create the translations (by our wonderful localisation volunteers). It feels like `gettext` is well known and supported, but the Qt way isn't.

Perhaps something to discuss in the Monday Mu-moot..?

dybber, 2021-02-18:
I agree that ideally we should do what makes sense in Python, not what makes sense for Qt. 

There are two parts of the translations:

1. Mu specific strings
2. Strings built into Qt ("OK", "Cancel", "Close", "Unable to launch external application", "Open File", "New Folder", "Rename Folder", etc.)

In my opinion we have to work with translations in these two cases: for the 1. we should always use what is easiest for our translators (the current `gettext`-approach), for 2. we have to discuss whether we want to use Qt translation facilities or we want to do work arounds as described in #938.

carlosperate, 2021-10-06:
@ntoll we should probably figure out this one before reaching out to translators.

carlosperate, 2021-11-04:
@ntoll @tmontes @xbecas if we don't address this before finishing up the set up of the translations should we move it to the 1.2 milestone?

tmontes, 2021-11-04:
Just reviewed the issue -- yikes, great work by @dybber! :)

Inclined to go `gettext` as much as possible + inclined to defer to a future release.

More -- just for fun, right?! -- the splash screen "this may take a while" bitmap message: should be localized! :) :) :)