mu-editor/mu #835: Rename Adafruit mode to CircuitPython mode

dhalbert, 2019-05-10:
Now that CircuitPython is supported on a number of non-Adafruit devices (see [circuitpython.org](https://circuitpython.org)), and non-Adafruit boards have been added to Mu, we're thinking it would be appropriate to rename the mode to "CircuitPython". A CircuitPython device has a CIRCUITPY drive and support CircuitPython-specific modules (`board`, `busio`, etc.).

The mode description
```
Adafruit CircuitPython
     Use CircuitPython on Adafruit's line of boards.
```
would change to something like
```
CircuitPython
     Use CircuitPython on Adafruit, MeowCat, Particle, and other CircuitPython boards.
```

I'm planning to do a PR for this soon, but someone else can too. Mostly it just involves some string changes, and renaming the `adafruit.py` files in the source tree. There are also some images to update. I will also remove the specific Adafruit boards listed in `mu/modes/adafruit.py`, since they are covered by the general `(0x239A, None),    # Any Adafruit Boards` entry.

ntoll, 2019-05-28:
Should the logo be replaced too? (From the Adafruit corporate logo to Blinka [please suggest an appropriate picture of Blinka since there are several I could choose, and I'd like to reflect your [CircuitPython's] judgement on this]).

bwshockley, 2019-06-26:
I worked this a bit with PR #868 - I think I got the changes implemented, but I wasn't sure of the VID & PIDs for other boards.

ntoll, 2019-07-04:
Fixed in #869.

closed 2019-07-04 by ntoll