mu-editor/mu #839: ESP-mode REPL: Off by one in error-message line numbers.

dybber, 2019-05-24:
Error messages reported when executing a file using the "Run" button in ESP-mode, are reported on line N+1, when error is on line N.

E.g. with an error on line 14, the following error is reported:
```Traceback (most recent call last):
  File "<stdin>", line 15, in <module>```

dybber, 2019-05-28:
@ntoll: Do you remember why this newline was added:
https://github.com/mu-editor/mu/blob/master/mu/interface/panes.py#L309

ntoll, 2019-05-28:
No. I believe it may be an un-wanted part of the "recipe" for putting the device into raw mode, copied from the microfs command. I'm up to my eyes in creating a new web mode for Mu, but once that's ready to be a PR I can tackle the "low hanging fruit" bugs in preparation for the alpha.2 release.

dybber, 2019-05-28:
Okay, I think we should use one of the existing implementations for entering/exiting raw REPL (e.g. call the one in microfs.py), they include very specific timings and checks that all the right phases have passed, and I don't believe a newline here is necessary. I can look at it one of the next days.

dybber, 2019-05-28:
Okay, the code in microfs is harder to reuse than I anticipated as microfs uses pyserial, and Mu uses QSerialPort which are incompatible.

dybber, 2019-06-09:
I can see the reason for using QSerialPort to use Qt signals on incoming messages. Should I try porting the enter/exit raw REPL code from microfs.py or pyboard.py to use QSerialPort?

carlosperate, 2020-09-29:
@dybber did #839 fix this issue?

dybber, 2020-09-30:
Yes, this was fixed by #1017

closed 2020-09-30 by dybber