mu-editor/mu #841: How do a "save as" in mu-editor ?

BEZIAT11, 2019-05-24:
Hello, can you tell me how to do a "save as" in mu-editor ?
If it's not possible currently, could anyone add an icon "save as" ?
Thanks you.

tmontes, 2019-05-24:
@BEZIAT11, you can double-click the "named edtiro tab" to save its contents under a different filename. Not sure if there are other options.

Does this help?

BEZIAT11, 2019-05-24:
Thanks but i try to find a simpler solution like others programs for my students.

tmontes, 2019-05-26:
@BEZIAT11, as far as I can tell, there is no simpler solution today.

I admit it may not be seen as *obvious* given that there is no "Save as" button in the UI, much like, I suppose, the "other programs" you refer to. However, keep in mind that Mu is targeted at beginners, and that removing friction (thus, options and UI complexity) from the whole experience is paramount: it could be argued that a more explicit "Save as" option / button would be distractive, lead to mistakes and, most importantly, go widely unused by the target audience.

The current solution *still lets* users rename their files in the event, say, that they mistyped their name when saving.

Having said this, mostly to share my perspective on how and why Mu behaves the way it does, I wonder what you would consider a "simpler solution" for your students: maybe it can be incorporated or, who knows, you may be able to code it yourself with our guidance.

Feel free to share your thoughts. :)

BEZIAT11, 2019-05-27:
Thank you for sharing your point of view. I specify that I ask my students to make different versions of their program. If they forget to make a copy of the file of the previous version, after modification they can not "save as" to create a new version. They must then copy, close Mu, open Mu, paste, then save with more risk of losing the advancement for a beginner or previous version. This seems to me more complicated.
By allers I am unable to code this change and I remain interested in outside help.

Thanks for your interest.

ntoll, 2019-05-28:
@BEZIAT11 hi..!

This is covered in the very first tutorial on the Mu website:

https://codewith.mu/en/tutorials/1.0/start

To rename a file or to save a file as another name, just double click on the filename in the tab. The old file will still exist, so no need to copy the contents.

That's it..! :-)

I'll close this issue since we already do a "save as" and the current behaviour reflects the behaviour we were told teachers wanted (as described by the Raspberry Pi Foundation).

closed 2019-05-28 by ntoll