mu-editor/mu #842: ctrl-c in repl does not respond

V2man, 2019-05-24:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

V2man, 2019-05-24:
I am trying to interact with CPX, Adafruit circuit playground express board using ctrl-c in REPL. 
In order to do so I have to save my code.py file to the CIRCUITPY drive before REPL will acknowledge ctrl-c.  I should not have to waste a write cycle to the EEPROM of ATmega32U4 on cpx board to do this.
MU version 1.02 on Windows 10.

dhalbert, 2019-05-24:
This is described in https://forums.adafruit.com/viewtopic.php?f=60&t=151826. Because one user has also seen this issue with Tera Term connecting to the REPL, not just Mu, I don't believe this is a Mu issue, but instead is some CircuitPython-on-Windows USB anomaly, which we are attempting to reproduce.

ntoll, 2019-05-24:
@dhalbert thank you for responding to this. You're a :star:.

@V2man I'm going to close this issue as it's not related to Mu directly (i.e. it appears to be something to do with CircuitPython-with-Windows).

Thanks for taking the time to feed this back.

closed 2019-05-24 by ntoll