mu-editor/mu #843: File close button on tabbar to the right hand side?

rolandvs, 2019-05-27:
It seems to me that there is a little inconsistency with mu-editor. Many editors have the close "button" of a file on a tabbar at the right hand side of the tab.

ntoll, 2019-05-28:
Hi @rolandvs,

This is platform dependent. On my Linux machine (and on Windows) the close "X" button on each tab is, indeed, on the right hand side. On OSX I believe the convention for closing tabs, is that the close "X" button is on the left. Mu uses Qt5 as a GUI frameworks, and Qt tries to do what is appropriate for the host OS.

I hope this clarifies what's happening.

Many thanks!

N.

closed 2019-05-28 by ntoll

ZanderBrown, 2019-07-20:
Further information for anyone finding this:

I 'themed' Mu, yep the ugly colour scheme is my fault! It was my intention that Mu would have close on the right as on Windows/Linux, and other tabbed things such as firefox/chrome in macOS, and for a while the development version on Mu did have this

However a Qt bug meant we couldn't ship Mu like this (see #594 for all the gory details) which eventually led to #619 while we waited for Qt to fix this (as far as I know they still haven't) and so out temporary solution has stayed

But this may change! Another issue relating to tab styling prompted me to take another look at this resulting in #899 which (if merged) will finally give us consistent tabs