mu-editor/mu #846: Scroll wheel doesn't zoom text

senkyr, 2019-05-31:
Without the ability to zoom in and out with mouse scroll wheel this otherwise promising editor is of no use for me. I am a teacher and I frequently adjust zoom level when explaining various parts of code to students.

ntoll, 2019-05-31:
Hi,

Thank you for your feedback! :+1:

We have two zoom adjustment buttons already built into the UI (see the tutorial here: https://codewith.mu/en/tutorials/1.0/start).

Mouse scroll wheel activity allows users to scroll up and down through their text, just like all other text editors. Changing this would be a significant disruption to users' expectations and make Mu behave in a way that was completely unlike all other editors. We endeavour to make Mu as "conventional" as possible so that beginner coders are able to graduate to other development environments without having to re-learn behaviours and UI/UX principles.

Since we already have zoom built in and your suggestion would break existing behavioural expectations, I'll close this issue. 

However, it's quite likely that I've missed something here (for example, this could be an accessibility issue), so please feel free to re-open this issue; but then please also explain **exactly** what your UI behaviour expectations are, and address my comments above about there being an existing solution for zooming and how scroll-wheel re-assignment is a significant disruption to expected behaviour. If it's helpful, our approach to user interface and user experience development is detailed here: https://mu.readthedocs.io/en/latest/user-experience.html

Once again, thank you for your feedback! :slightly_smiling_face:

senkyr, 2019-05-31:
Thank you for your quick reply.

I wasn't entirely exact about my issue, I'll admit that much. What I was talking about was the ability to zoom in and out with scroll wheel **while holding down CTRL key**. That is the behavior I am used to from all known code development software (namely Notepad++, Visual Studio, Visual Studio Code, Atom, etc.)

The option to zoom in and out via dedicated keyboard shortcuts (`CTRL`+`+`, `CTRL` + `-`) or buttons in GUI is good but too cumbersome when the need to adjust zoom level comes often. Me personally, I just feel disabled without `CTRL` + scroll wheel combo.

I hope I have explained my point clearly this time.

Best regards!

ntoll, 2019-05-31:
@senkyr AHA..! I see now... :smile:  

I didn't even know this behaviour existed (I don't really use a mouse while editing, preferring to use keyboard shortcuts).

OK... let me re-open this and see if we can make this work. :+1:

tjguk, 2019-05-31:
I was just about to comment: I've just checked and (on Windows, at least) we *do* support ctrl-scroll for zooming.

tjguk, 2019-05-31:
(I haven't checked, but I suspect we've just inherited functionality built into the QScintilla control...)

abixadamj, 2019-05-31:
Works PERFECTLY on Linux mu-editor 1.02 ;-) CTRL+wheel mouse.

senkyr, 2019-05-31:
I am in Windows and I have tried the latest version just before opening this issue. No zooming happened. What am I doing wrong?

tmontes, 2019-05-31:
Confirmed working on macOS 10.12 Sierra: both on 1.0.2 and current master.

ntoll, 2019-05-31:
Hmmm.... so @senkyr it looks like Mu does already do this out of the box on Windows, but there's something about your machine or setup which stops it working for you. Can you please let us know more details about your system..? Also, can you please confirm that the text area is in focus when you try to use CTRL+scroll..? (Just checking all the obvious problems here).

senkyr, 2019-05-31:
I am lost here.

I have downloaded and deployed the portable version 1.0.2 here on Windows 10 1703. I have launched `Launch Mu.bat`, got the window, left the untitled document with a `# Write your code here :-)` template open.

I have pressed down `CTRL` key, hold it there. Pressed the `+` key, the text zoomed in. Pressed the `-` key, the text zoomed out. I have released `CTRL`.

I have pressed down `CTRL` key again, hold it there. Scrolled the mouse wheel one way, the other way. _Nothing happened._

senkyr, 2019-05-31:
It works this way on all Windows machines I was trying Mu on in the last year or so. At least four different machines. All Windows 10, various builds.

Yes, the text area is positively in focus. The cursor is blinking right under the text.

ntoll, 2019-05-31:
I just tried on Linux, and it works for me. :-/

Not really sure what's going on.

The problem is, if we can't reproduce the bug, then there's nothing for us to fix. That it only appears to affect @senkyr suggests there's something about his particular setup which is causing the problem. I'm not really sure what to suggest in this case.

I'm really clutching at straws here, but could you try it on another Windows computer? I notice you mention you were running it via PortaMu (i.e. from a USB stick). Can you confirm the behaviour of the editor if you actually use Mu installed via the official installer onto the hard drive of the computer..?

tmontes, 2019-05-31:
Intriguing... Like @ntoll suggested, can you please try and install Mu from published the distribution at [https://codewith.mu/en/download](https://codewith.mu/en/download)?

senkyr, 2019-05-31:
I have downloaded and installed the 64-bit Windows version of Mu from https://codewith.mu/en/download

No change in behavior.

tmontes, 2019-05-31:
Thanks for your persistence, @senkyr,

Given that we have multiple confirmations of the feature working across several platforms, Windows 10 included (I just tested it on a Windows 10 1803 VM), I suspect there must be something specific to your system(s):

* Windows 10 release is different - Does not feel like a culprit candidate.
* Could your system(s) be running some kind of mouse / trackpad driver that could be affecting this behaviour with Mu? (even though it is very strange that it would work with other software)

Maybe we're hitting a bug at the intersection of QScintilla / Qt / Windows 10 here... Ideas? :)

tjguk, 2019-05-31:
The only thing I can think of is to try using spy++ or similar to track the messages arriving at the window -- to see if we're getting a WM_MOUSEWHEEL message with a Ctrl modifier. But I don't know if the @senkyr is  up for that? You mentioned VS so presumably you have access to spy++ already?

senkyr, 2019-05-31:
> Windows 10 release is different - Does not feel like a culprit candidate.
Probably not the case, since I have tried this on multiple releases.

> Could your system(s) be running some kind of mouse / trackpad driver that could be affecting this behaviour with Mu? (even though it is very strange that it would work with other software)
Not that I am aware of. I use scroll zoom without issues in all other software.

I am now trying to persuade several friends of mine to test this bug. So far I have one Fedora user zooming successfully, and one Windows 10 1803 user with the _same issue_ as mine.

senkyr, 2019-05-31:
@tjguk I am not familiar with spy++, will investigate further.

tjguk, 2019-05-31:
Thanks @senkyr. I'm frankly clutching at straws, but I'm interested in seeing what messages are getting to the editor window when the mouse wheel is scrolled with/without the ctrl-key modifier.

The complicating thing here is that this zoom functionality isn't ours: it's built into the editor control we're using (which is a nightmare to build from scratch). So it's not easy to detect the messages from within. Hence the need for Windows spy

ntoll, 2019-05-31:
So, I've just booted up my Windows 10 laptop to test things. I **can't** do a CTRL+SCROLL to zoom in or out -- so I'm able to reproduce the behaviour described by @senkyr. However, I cannot get CTRL+Scroll to work for zooming an **ANY** text edit type application such as VSCode or even Notepad.

From this, it feels to me like there's a system setting somewhere that needs to be enabled so CTRL+Scroll acts as a zoom controller. I've spent the last half hour trying to find such a setting but (given that I'm not a Windows user) I've not had much success. 

To echo @tjguk's comments... Mu should do whatever the default is for the OS. It feels like, on Windows 10 at least, the default is not to zoom when using CTRL+scroll. Tim, you mentioned it *did* work for you... any idea what settings might affect this?

This is turning into a deep rabbit hole! ;-)

tjguk, 2019-05-31:
@ntoll AFAIK the ctrl-scroll functionality is not a system functionality such; it's defined on an app-by-app basis. For example, if you're in an explorer window (including the desktop) then ctrl-scroll will resize the icons / layout. But that's because explorer implements that. 

If you're in, say, scite (which also uses Scintilla) then it zooms; but if you're in the built-in notepad, it doesn't. In all browsers I'm aware of, it does.

In other words, I'm not aware of a system setting which says "don't zoom on ctrl-scroll". And even if there were, I would expect that to disable the effect for every app, because presumably it would prevent that combination of actions from sending the WM_MOUSEWHEEL message (or it would suppress the Ctrl modifier on that message): https://docs.microsoft.com/en-us/windows/desktop/inputdev/wm-mousewheel

So this is quite a mystery to me.

senkyr, 2019-05-31:
I am glad we are at least on the same page here (almost).

Meanwhile I have got another Windows user in our organization to reproduce the same issue. But again, this behavior is not bound to our machines only (at least one friend of mine is unable to zoom either in Mu on Windows).

@ntoll Zooming with mousewheel works for me in every software other than Mu. It is not a system-wide problem, at least for me.

ntoll, 2019-05-31:
OK... so putting aside the fact that the CTRL-scroll for zooming behaviour is "flaky" on Windows and inconsistent with other platforms (for instance, it doesn't respect the zooming feature we've already created since I can zoom to minute and HUGE proportions on Linux), then I think we have a clear solution:

We should trap the top level `wheelEvent` handler for the parent widget (https://doc.qt.io/qt-5/qwidget.html#wheelEvent) and plug this into the existing zoom functionality. In this way we will (hopefully) explicitly make it work on Windows out-of-the-box, make the behaviour reflect the current zoom functionality and make the behaviour consistent across all platforms.

Does this sound like a plan..?

ntoll, 2019-05-31:
To be clear, the logic should be that if the `wheelEvent` fires and the `CTRL` key is pressed, then an appropriate zoom in / out should happen depending on the direction of the wheel (up=zoom in, down=zoom out).

tjguk, 2019-05-31:
I've just tweeted this issue in case anyone in the wider community might have an insight.

re @ntoll point above, I did realise when scanning the code that we're essentially implementing our own controlled zoom via the buttons, and bypassing that when using the built-in ctrl-scroll or Ctrl+/-. Trapping that signal ourselves to meet our existing implementation seems sensible. (Does Qscintilla emit a "zoom" signal?)

senkyr, 2019-05-31:
I was ultimately unable to get Spy++ to spy (doesn't want to protocol messages, not sure why) but it seems unnecessary by now (?)

tjguk, 2019-05-31:
@senkyr thanks for trying. If @ntoll approach to trapping the action specifically works then we should be ok. (At the last: we'll be able to debug from without our own code)

ntoll, 2019-05-31:
Folks, give me 10 mins and I think I have a solution PR in the works. :-)

senkyr, 2019-05-31:
That is seriously great! Would it work, I am gonna use Mu + Pygame in the introductory programming course after summer break.

ntoll, 2019-05-31:
Et voila... we have a fix with unit tests here: #848 

Please can Mu developers test this with Windows and OSX. Thank you..!

It'll be in the next alpha release. :-)

Kudos and many thanks to @senkyr for bringing this to our attention and following up. :cow: :heart: :mouse: :wheel_of_dharma: :+1: 

I'll close this issue when folks have tested the PR.

tjguk, 2019-05-31:
Works for me on Win10. (Or rather: doesn't stop working for me on Win10)

PR #848 (2019-07-04): Using CTRL-scroll with a mouse, controls built-in zoom functionality
Fixes #846.

This needs testing on:

 * [X] Linux
 * [ ] Windows 10
 * [ ] Windows 7
 * [ ] OSX
 * [ ] Raspberry Pi
closed 2019-07-04 by ntoll