mu-editor/mu #847: Icons doesn't change when switching color themes

senkyr, 2019-05-31:
Three brightness levels are fine but icons remain dazzling white all the time. Kind of kills the purpose of dark mode.

ntoll, 2019-05-31:
That's an aesthetic opinion... ;-)

The icon designs were created to work in high contrast mode for all themes. For instance, if you're in dark mode and have a visual impairment, then finding the right button if such a button were "dark" would be a huge impediment.

If you can come up with a **high contrast** dark version of the buttons, we'd certainly use them (I'll close this issue, but please feel free to create a PR with suggested dark high contrast button designs).

closed 2019-05-31 by ntoll

senkyr, 2019-05-31:
[image]

How about these? Just a quick mockup, I am no artist.

Not exactly of high contrast, but again, that would be contrary to the reason dark modes exist in software in the first place. Those who need high contrast (for vision impairment etc.) wouldn't choose the dark subdued theme anyway.

For three color themes there should be three icon variants. Pure whites belong to the high contrast setting only. My opinion.

FreshAlacrity, 2023-09-22:
I personally use dark mode because I find bright, high-contrast UI elements to be eye-straining; it looks like I'll need to learn how to make those changes for myself to use Mu for now, but I'd like to note that it can be an accessibility issue in the opposite direction as well. Glad the project is open source so I can learn how to do that!