mu-editor/mu #851: Connecting to ESP32 on 64-bit Windows 10

cgarmstr, 2019-06-03:
I've been trying to connect to an ESP32-WROVER-KIT from a Windows 10 machine.

I can connect ok to the REPL with TeraTerm on Port COM5, but nothing happens with Mu. Do I need to specify serial port settings somewhere?
[image]

The log file is below.

Many thanks

Cecil

PS I saw the post about an upcoming fix to ESP32 mode. Is there a way of getting informed when the next alpha is released?
---------------------------------------------------------------
2019-06-03 22:08:36,791 - root:112(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.1
2019-06-03 22:08:36,793 - root:113(run) INFO: uname_result(system='Windows', node='OPTIPLEX-7040', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-03 22:08:36,793 - root:114(run) INFO: Python path: ['C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\cecil\\AppData\\Local\\Mu', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python', 'C:\\Users\\cecil\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-06-03 22:08:36,793 - root:115(run) INFO: Language code: en_GB
2019-06-03 22:08:37,081 - mu.logic:574(__init__) INFO: Setting up editor.
2019-06-03 22:08:37,082 - mu.logic:596(__init__) INFO: Settings path: C:\Users\cecil\AppData\Local\python\mu\settings.json
2019-06-03 22:08:37,082 - mu.logic:597(__init__) INFO: Session path: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-03 22:08:37,082 - mu.logic:598(__init__) INFO: Log directory: C:\Users\cecil\AppData\Local\python\mu\Logs
2019-06-03 22:08:37,082 - mu.logic:599(__init__) INFO: Data directory: C:\Users\cecil\AppData\Local\python\mu
2019-06-03 22:08:37,100 - mu.logic:612(setup) INFO: Available modes: python, adafruit, microbit, esp, debugger, pygamezero
2019-06-03 22:08:37,942 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-03 22:08:37,947 - mu.logic:661(restore_session) INFO: Restoring session from: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-03 22:08:37,947 - mu.logic:662(restore_session) DEBUG: {'theme': 'day', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-03 22:08:37,947 - mu.logic:683(restore_session) INFO: Loaded files.
2019-06-03 22:08:37,947 - mu.logic:687(restore_session) INFO: User defined environment variables: []
2019-06-03 22:08:37,947 - mu.logic:691(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-03 22:08:38,504 - mu.logic:714(restore_session) INFO: Starting with blank file.
2019-06-03 22:08:38,520 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-03 22:09:00,226 - mu.modes.base:233(find_device) WARNING: Could not find device.
2019-06-03 22:09:00,226 - mu.modes.base:234(find_device) DEBUG: Available ports:
2019-06-03 22:09:00,226 - mu.modes.base:238(find_device) DEBUG: ['PID:0xa13d VID:0x8086 PORT:COM3', 'PID:0x0000 VID:0x0000 PORT:COM1', 'PID:0x6010 VID:0x0403 PORT:COM4', 'PID:0x6010 VID:0x0403 PORT:COM5']
2019-06-03 22:09:00,278 - mu.interface.main:740(show_message) DEBUG: Could not find an attached device.
2019-06-03 22:09:00,278 - mu.interface.main:741(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2019-06-03 22:09:14,074 - mu.modes.base:261(toggle_repl) INFO: Toggle REPL on.
2019-06-03 22:09:16,997 - mu.logic:1180(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'esp', 'debugger', 'pygamezero']
2019-06-03 22:09:26,025 - mu.logic:736(new) INFO: Added a new tab.
2019-06-03 22:09:30,108 - mu.logic:854(get_dialog_directory) INFO: Using path for file dialog: C:\Users\cecil\mu_code
2019-06-03 22:09:33,081 - mu.interface.main:256(get_load_path) DEBUG: Getting load path: 
2019-06-03 22:09:46,891 - mu.logic:1060(show_help) INFO: Showing help at 'https://codewith.mu/en/help/1.1'.
2019-06-03 22:13:42,188 - root:112(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.1
2019-06-03 22:13:42,188 - root:113(run) INFO: uname_result(system='Windows', node='OPTIPLEX-7040', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-03 22:13:42,188 - root:114(run) INFO: Python path: ['C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\cecil\\AppData\\Local\\Mu', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python', 'C:\\Users\\cecil\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-06-03 22:13:42,188 - root:115(run) INFO: Language code: en_GB
2019-06-03 22:13:42,287 - mu.logic:574(__init__) INFO: Setting up editor.
2019-06-03 22:13:42,287 - mu.logic:596(__init__) INFO: Settings path: C:\Users\cecil\AppData\Local\python\mu\settings.json
2019-06-03 22:13:42,288 - mu.logic:597(__init__) INFO: Session path: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-03 22:13:42,288 - mu.logic:598(__init__) INFO: Log directory: C:\Users\cecil\AppData\Local\python\mu\Logs
2019-06-03 22:13:42,288 - mu.logic:599(__init__) INFO: Data directory: C:\Users\cecil\AppData\Local\python\mu
2019-06-03 22:13:42,311 - mu.logic:612(setup) INFO: Available modes: python, adafruit, microbit, esp, debugger, pygamezero
2019-06-03 22:13:42,866 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-03 22:13:42,867 - mu.logic:661(restore_session) INFO: Restoring session from: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-03 22:13:42,867 - mu.logic:662(restore_session) DEBUG: {'theme': 'day', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-03 22:13:42,867 - mu.logic:683(restore_session) INFO: Loaded files.
2019-06-03 22:13:42,867 - mu.logic:687(restore_session) INFO: User defined environment variables: []
2019-06-03 22:13:42,867 - mu.logic:691(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-03 22:13:43,358 - mu.logic:714(restore_session) INFO: Starting with blank file.
2019-06-03 22:13:43,371 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-03 22:13:51,675 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\cecil\AppData\Local\python\mu\Logs\mu.log
2019-06-03 22:13:51,675 - mu.logic:157(installed_packages) INFO: Packages found: []
2019-06-03 22:16:18,702 - mu.logic:1151(show_admin) INFO: No admin settings changed.
2019-06-03 22:16:20,772 - mu.logic:1060(show_help) INFO: Showing help at 'https://codewith.mu/en/help/1.1'.
2019-06-03 22:23:32,264 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\cecil\AppData\Local\python\mu\Logs\mu.log
2019-06-03 22:23:32,264 - mu.logic:157(installed_packages) INFO: Packages found: []

If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2019-06-03:
Hi @cgarmstr,

Thanks for the bug report.

The alpha can be downloaded from the official download page: https://codewith.mu/en/download
Or the GH release: https://github.com/mu-editor/mu/releases/tag/1.1.0-alpha.1

There should be an ESP mode in that version that allows you to connect to the ESP32 board.

If you try that and still doesn't work, could you provide exact steps (including code running on the board) on how to reproduce the problem, what happens, and what do you expect to happen?

dybber, 2019-06-09:
My best guess will be that the particular ESP-board is not detected by the esp-mode. The various ESP-devices are identified based on their vendor ID's (VID) and product ID's (PID). Could you try to lookup the VID/PID of the esp board?

Here is a description of how to do it on Windows: https://superuser.com/a/1106248

cgarmstr, 2019-06-09:
Comparing the list of devices with the board turned off and on, the board supplies devices:

USB Composite Device, Hardware IDs 
USB\VID_0403&PID_6010&REV_0700
USB\VID_0403&PID_6010

USB Serial Converter A
USB\VID_0403&PID_6010&REV_0700&MI_00
USB\VID_0403&PID_6010&MI_00

USB Serial Converter B
USB\VID_0403&PID_6010&REV_0700&MI_01
USB\VID_0403&PID_6010&MI_01

Hope this helps

Cecil

carlosperate, 2019-06-09:
That is possibly the FTDI VID, but not the one included in the ESP mode (https://github.com/mu-editor/mu/pull/790/files#r266196766).
What board have you got? and if you know, which UART-to-USB chip?

cgarmstr, 2019-06-09:
Ah, now you're going outside my knowledge and competence. 

It is an ESP-WROVER-Kit, see https://docs.espressif.com/projects/esp-idf/en/latest/get-started/get-started-wrover-kit.html 

The docs seem to suggest that the USB driver is FTDI, see  https://docs.espressif.com/projects/esp-idf/en/latest/get-started/establish-serial-connection.html

4 of the ESP32 boards on there use the CP210x driver, which I think you already have covered, and 3 use the FTDI USB Driver.

Is this useful? Let me know if you want me to do any other testing.

Thanks!

carlosperate, 2019-06-09:
That is very useful, that's exactly what we needed, thanks!

And this document from FTDI confirms the values:
https://www.ftdichip.com/Support/Documents/TechnicalNotes/TN_100_USB_VID-PID_Guidelines.pdf

carlosperate, 2019-06-10:
From https://github.com/mu-editor/mu/pull/855#issuecomment-500379017:

> I followed the guidelines for Developer Setup https://mu.readthedocs.io/en/latest/setup.html but after doing python run.py got
ModuleNotFoundError: No module named 'appdirs'
pip install appdirs
fixed that but then got
from PyQt5.Qsci import QsciScintilla, QsciLexerPython, QsciAPIs
ModuleNotFoundError: No module named 'PyQt5.Qsci'

It sounds like `pip install -e .[dev]` didn't work, could you open a new issue showing all the steps you followed and the console output?

From https://github.com/mu-editor/mu/pull/855#issuecomment-500379017:

> PS I cheated and put the extra line
(0x0403, 0x6010), # FT2232C/D/L/HL/Q (ESP-WROVER-KIT, ESP32DemoBoard)
in esp.py in my existing mu-editor_1.1.0-alpha.1_win64.exe installation.
Mu appears doesn't give a no device connected error, but there is no response to the REPL.

Can you see any output when connection with tera term?
When the REPL panel opens, can you press enter and see anything appears?
Could you send us a screenshot of what it looks like when the REPL panel is opened?

carlosperate, 2019-06-10:
And also, could you open the REPL and send us the log without doing any other actions? (So that the REPL action is one of the last things in the log).

cgarmstr, 2019-06-10:
[image]
[image]
[image]

No response when hitting return after REPL screen comes up. I think Teraterm tries all the Com ports until it finds one that responds.

Log file looks like Mu is trying to open the REPL on COM4, rather than COM5.

Note that this log file is with the extra line in esp.py i.e.
    valid_boards = [
        # VID  , PID
        (0x1A86, 0x7523),  # HL-340
        (0x10C4, 0xEA60),  # CP210x
        (0x0403, 0x6015),   # Sparkfun ESP32 VID, PID
        (0x0403, 0x6010),  # FT2232C/D/L/HL/Q (ESP-WROVER-KIT, ESP32DemoBoard)
    ]

The log file
--------------------------------------------------------------------------
2019-06-10 11:43:48,008 - root:112(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.1
2019-06-10 11:43:48,009 - root:113(run) INFO: uname_result(system='Windows', node='OPTIPLEX-7040', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-10 11:43:48,010 - root:114(run) INFO: Python path: ['C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\cecil\\AppData\\Local\\Mu', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python', 'C:\\Users\\cecil\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-06-10 11:43:48,010 - root:115(run) INFO: Language code: en_GB
2019-06-10 11:43:48,288 - mu.logic:574(__init__) INFO: Setting up editor.
2019-06-10 11:43:48,288 - mu.logic:596(__init__) INFO: Settings path: C:\Users\cecil\AppData\Local\python\mu\settings.json
2019-06-10 11:43:48,288 - mu.logic:597(__init__) INFO: Session path: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-10 11:43:48,288 - mu.logic:598(__init__) INFO: Log directory: C:\Users\cecil\AppData\Local\python\mu\Logs
2019-06-10 11:43:48,288 - mu.logic:599(__init__) INFO: Data directory: C:\Users\cecil\AppData\Local\python\mu
2019-06-10 11:43:48,307 - mu.logic:612(setup) INFO: Available modes: python, adafruit, microbit, esp, debugger, pygamezero
2019-06-10 11:43:48,905 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-10 11:43:48,910 - mu.logic:661(restore_session) INFO: Restoring session from: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-10 11:43:48,910 - mu.logic:662(restore_session) DEBUG: {'theme': 'day', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-10 11:43:48,910 - mu.logic:683(restore_session) INFO: Loaded files.
2019-06-10 11:43:48,910 - mu.logic:687(restore_session) INFO: User defined environment variables: []
2019-06-10 11:43:48,910 - mu.logic:691(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-10 11:43:49,490 - mu.logic:714(restore_session) INFO: Starting with blank file.
2019-06-10 11:43:49,510 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-10 11:51:14,076 - root:112(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.1
2019-06-10 11:51:14,077 - root:113(run) INFO: uname_result(system='Windows', node='OPTIPLEX-7040', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-10 11:51:14,077 - root:114(run) INFO: Python path: ['C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\cecil\\AppData\\Local\\Mu', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\Python', 'C:\\Users\\cecil\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\cecil\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-06-10 11:51:14,077 - root:115(run) INFO: Language code: en_GB
2019-06-10 11:51:14,175 - mu.logic:574(__init__) INFO: Setting up editor.
2019-06-10 11:51:14,175 - mu.logic:596(__init__) INFO: Settings path: C:\Users\cecil\AppData\Local\python\mu\settings.json
2019-06-10 11:51:14,175 - mu.logic:597(__init__) INFO: Session path: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-10 11:51:14,175 - mu.logic:598(__init__) INFO: Log directory: C:\Users\cecil\AppData\Local\python\mu\Logs
2019-06-10 11:51:14,175 - mu.logic:599(__init__) INFO: Data directory: C:\Users\cecil\AppData\Local\python\mu
2019-06-10 11:51:14,194 - mu.logic:612(setup) INFO: Available modes: python, adafruit, microbit, esp, debugger, pygamezero
2019-06-10 11:51:14,730 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
2019-06-10 11:51:14,730 - mu.logic:661(restore_session) INFO: Restoring session from: C:\Users\cecil\AppData\Local\python\mu\session.json
2019-06-10 11:51:14,730 - mu.logic:662(restore_session) DEBUG: {'theme': 'day', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-10 11:51:14,730 - mu.logic:683(restore_session) INFO: Loaded files.
2019-06-10 11:51:14,730 - mu.logic:687(restore_session) INFO: User defined environment variables: []
2019-06-10 11:51:14,730 - mu.logic:691(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-10 11:51:15,286 - mu.logic:714(restore_session) INFO: Starting with blank file.
2019-06-10 11:51:15,300 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\cecil\mu_code
[... cut ...]
2019-06-10 20:30:53,251 - mu.modes.base:229(find_device) INFO: Found device on port: COM4
2019-06-10 20:30:53,251 - mu.modes.base:230(find_device) INFO: Serial number: 5&16BB5AE6&0&10&1
2019-06-10 20:30:53,273 - mu.modes.base:280(add_repl) INFO: Started REPL on port: COM4
2019-06-10 20:30:53,273 - mu.modes.base:261(toggle_repl) INFO: Toggle REPL on.
2019-06-10 20:31:07,435 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\cecil\AppData\Local\python\mu\Logs\mu.log
2019-06-10 20:31:07,436 - mu.logic:157(installed_packages) INFO: Packages found: []

cgarmstr, 2019-06-10:
From https://docs.espressif.com/projects/esp-idf/en/latest/get-started/get-started-wrover-kit.html

Key Component | Description
-- | --
FT2232 | The FT2232 chip serves as a multi-protocol USB-to-serial bridge which can be programmed and controlled via USB to provide communication with ESP32. FT2232 also features USB-to-JTAG interface which is available on channel A of the chip, while USB-to-serial is on channel B. The FT2232 chip enhances user-friendliness in terms of application development and debugging. See ESP-WROVER-KIT V4.1 schematic.

The ESP32-WROVER-Kit aappears to have two serial ports, which are shown as COM4 and COM5 below. Are these the channel A and channel B referred to above?
[image]

cgarmstr, 2019-06-10:
Yes, COM4 is serial converter A and COM5 is serial converter B.
[image]

carlosperate, 2019-06-10:
Yes, I think you found the issue. If there are several ports that match the VID/PID it will try to connect to the first one, which in this case would be COM4 with the USB-to-JTAG on channel A.

The easiest work-around at the moment would be for you to disable COM4 from the device manager, and that should let Mu connect to COM5.

But yeah, that's just a workaround, not 100% about the best way to fix this in Mu. Maybe if it finds several matches for the current mode it should try all until one of them passes? Not sure, we'd have to think about this one.

cgarmstr, 2019-06-11:
USB Composite Device, Hardware IDs
USB\VID_0403&PID_6010&REV_0700
USB\VID_0403&PID_6010

USB Serial Converter A
USB\VID_0403&PID_6010&REV_0700&MI_00
USB\VID_0403&PID_6010&MI_00

USB Serial Converter B
USB\VID_0403&PID_6010&REV_0700&MI_01
USB\VID_0403&PID_6010&MI_01

I would gues you can use the &MI_01 to distinguish channel B. It would be nice to have confirmation with somebody with a different board but same chip.

carlosperate, 2019-06-11:
Thanks for the suggestion @cgarmstr.
Getting some of this additional USB info has proven problematic in the past (different versions of PySerial capturing data differently, and having different values in different operating systems), so ideally I'd prefer not to go down that route as it will require a lot of testing and could easily end-up being a rabbit hole.

cgarmstr, 2019-06-13:
BTW I disabled COM4 in the Device Manager but Mu still attempts of access it. See end of log:

2019-06-13 10:00:00,689 - mu.modes.base:229(find_device) INFO: Found device on port: COM4
2019-06-13 10:00:00,689 - mu.modes.base:230(find_device) INFO: Serial number: 5&16BB5AE6&0&10&1
2019-06-13 10:00:00,710 - mu.modes.base:280(add_repl) INFO: Started REPL on port: COM4
2019-06-13 10:00:00,710 - mu.modes.base:261(toggle_repl) INFO: Toggle REPL on.
2019-06-13 10:00:08,395 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\cecil\AppData\Local\python\mu\Logs\mu.log
2019-06-13 10:00:08,395 - mu.logic:157(installed_packages) INFO: Packages found: []

carlosperate, 2019-06-13:
I was hoping disabling the device would make the port "disappear" from the list that PySerial returns to Mu when queering what serial devices are available.

Alternatively, in windows you could change the COM Port number. I don't know if this will work, but if the USB-to-JTAG on channel A port is changed to a higher number (like COM 9), maybe the other one will be listed first? Don't know if that will work, but might be worth a try.

Apologies for only being able to provide possible workarounds at the moment, we are all volunteers, and I personally haven't got too much free time this week.

dybber, 2019-06-13:
The problem seems to be that the ESP-WROVER kit doesn't have it's own VID/PID but exposes the VID/PID from the generic FTDI serial-to-USB chip. As the PDF @carlosperate linked to mentions, with this setup "OEM-specific identification of hardware is only available through the manufacturers’ string", thus we cannot identify it through VID/PID and the manufacturer string is not reliable (the same manufacturer produces many different USB-devices).

I think a good long-term solution would be some way to manually select a device from the Mu Administration-window, where a list of detected serial-devices could be displayed in a dropdown or similar you can select from. Here VID/PID/Manufacturer string could also be displayed, which could be useful in future bug reports.

ntoll, 2019-07-04:
@dybber +1 on that suggestion.

carlosperate, 2019-09-11:
> I think a good long-term solution would be some way to manually select a device from the Mu Administration-window, where a list of detected serial-devices could be displayed in a dropdown or similar [...]

Yeah, I agree that a way for the user to select the COM port (possibly from a list of ports with the  current mode VID/PIDs) would be a much better solution. But I think it should possibly be a window that pops up when more than a single option for the mode is found, this is because things in the settings should be more "long term" configuration, and the connected devices available can change several times during the same session.

dybber, 2020-09-05:
This should be fixed with #1017 and #1094, and I'm thus closing this issue. Please feel free to reopen or create a new issue, if it still doesn't work when the next version of Mu is released.

closed 2020-09-05 by dybber

cgarmstr, 2020-09-06:
Excellent Martin.

How do I find out when your fixes have been incorporated in the alpha release?

Thanks

Cecil

dybber, 2020-09-06:
We don't have a mailing list I'm afraid, so you will probably have to just check-in on the website and see if a new version has been released, or perhaps follow @ntoll on Twitter - I guess he will post there when the new version is out. I can't promise to remember to write here when it's released, I'm sorry.

cgarmstr, 2020-09-06:
No probs. Thanks.
Cecil

carlosperate, 2020-09-06:
You can also watch the repository for releases and that should give you a notification when a new GH release is created.