mu-editor/mu #852: pip install -e .[dev] error in mac zsh

carlosperate, 2019-06-04:
When running `pip install -e .[dev]` in Z shell you get the following error:

```
$ pip install -e .[dev]
zsh: no matches found: .[dev]
```

This is because ([source](https://stackoverflow.com/questions/30539798/zsh-no-matches-found-requestssecurity/30539963)) :
> zsh uses [square brackets for globbing / pattern matching](http://zsh.sourceforge.net/Guide/zshguide05.html#l137).

So we need to run this instead:

```
pip install -e .'[dev]'
```

Normally I would have consider this a well-known, or easy to find, difference in a custom shell, that most developers that use it would be able to sold on their own. However with yesterday's announcement that macOS Catalina will use zsh **by default** (https://support.apple.com/en-ca/HT208050) we should probably add a note to the dev docs.

tjguk, 2019-06-04:
(Sigh) Just when we think we've got a consistent, universal way of installing a dev environment...

FWIW it works in Windows if you use double quotes:

`pip install -e ."[dev]"`

Does that fly in macOS as well?

carlosperate, 2019-06-04:
Yep, testing zsh 5.3 and bash 3.2 in macOS Mojave, `pip install -e ."[dev]"` works, and `pip install -e ".[dev]"` works too.

tjguk, 2019-06-04:
Ok; so if someone can confirm from the Linux / Bash perspective then I think we can tweak the docs. @ntoll ?

ukBaz, 2019-06-04:
I have just tested `pip install -e ".[dev]"` with csh and bash. It worked with both.

carlosperate, 2019-06-04:
Thanks Barry!
So it looks like `pip install -e ".[dev]"` works universally.

DarkSuniuM, 2019-06-04:
`pip install -e .\[dev\]` works as well
"[", "]" and "!" are special characters in Zsh

ntoll, 2019-06-04:
Hah... just got in from a bunch of meetings to find this thread. Well done everyone for finding, fixing and testing..! :+1:

I have to admit having a big *sigh* at the same point in the conversation as @tjguk.

ntoll, 2019-06-04:
@darksunium good to know..!

tmontes, 2019-06-07:
Good catch @carlosperate.

I'm just wondering how / whether one could address "all possible CLI shell" nuances such as the one you point out. Of course we want to remove all friction from beginners, and having the documentation say `pip install -e ".[dev]"` will be better that `pip install -e .[dev]`, in that regard.

I went looking at how other projects document it:
* [Twisted](https://twistedmatrix.com/trac/) says `pip install -U -e '.[dev]'` [here](https://twistedmatrix.com/trac/wiki/TwistedDevelopment#WithoutTox).
* [attrs](https://www.attrs.org/en/stable/index.html) says `pip install -e '.[dev]'` [here](https://www.attrs.org/en/stable/contributing.html#local-development-environment).

Interestingly these are using single-quotes `'` instead of double-quotes `"`. Being very familiar with UNIX like systems, I'd prefer to go with single-quotes too (easier to type on most keyboard layouts and no variable interpolation at play)...

Then I tested Windows, and even though `PowerShell` likes the single-quotes approach, `cmd.exe` fails miserably; tested the suggested double-quotes approach and it worked in both Windows CLI shells. So I guess that's it: `pip install -e ".[dev]"`

+1

DarkSuniuM, 2019-06-09:
@tmontes That's because double quotes are for string formatting, like back ticks (`) in JavaScript or f-Strings in Python

```bash
export x=10
echo "${x}"  # Output: 10
echo '${x}'  # Output: ${x}
```

carlosperate, 2019-07-03:
PR raised: https://github.com/mu-editor/mu/pull/878

tmontes, 2019-07-04:
I suppose that given the fact that #878 has been merged, this issue can be closed? :)

carlosperate, 2019-07-04:
Yes, indeed. Thanks everybody for looking into all the multiple options!

closed 2019-07-04 by carlosperate