mu-editor/mu #853: canno start mu-portable 32 on win 10

abixadamj, 2019-06-07:
I have only problems ;-( What a life...
I want to help my friend who teach in one school and don't have access to administration of Windows, so cannot install any soft.
I think - ok, lets try mu-portable copied to student's Desktop directory . So I downloaded it and test in my clean fresh win 10 in Virtualbox.
[image]

Error! The same, as it was when I had Python 3.7 and tried to run mu installed standard way. 
But, now I have 3.6 and normal mu-editor works....
[image]

I cant figure out, what is it now... and I am afraid, that it will not works if there will be 3.7 Python installed on those computers in school...

LOG is:

C:\Users\Uczeñ\AppData\Local\python\mu\Logs>type mu.log
2019-06-07 13:11:59,994 - root:112(run) INFO:

-----------------

Starting Mu 1.0.2
2019-06-07 13:11:59,994 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-FMFAHVA', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-07 13:11:59,994 - root:114(run) INFO: Python path: ['pkgs', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\Python\\python36.zip', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\Python', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\pkgs', 'C:\\Users\\Ucze+ä\\AppData\\Roaming\\Python\\Python36\\site-packages', 'pkgs\\IPython\\extensions']
2019-06-07 13:11:59,994 - root:115(run) INFO: Language code: pl_PL
2019-06-07 13:12:16,512 - root:112(run) INFO:

-----------------

Starting Mu 1.0.2
2019-06-07 13:12:16,512 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-FMFAHVA', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-07 13:12:16,512 - root:114(run) INFO: Python path: ['C:\\Users\\Ucze+ä\\Desktop\\mu\\pkgs', 'C:\\Users\\Ucze+ä\\Desktop\\mu', 'C:\\Users\\Ucze+ä\\AppData\\Local\\Programs\\Python\\Python36-32\\python36.zip', 'C:\\Users\\Ucze+ä\\AppData\\Local\\Programs\\Python\\Python36-32\\DLLs', 'C:\\Users\\Ucze+ä\\AppData\\Local\\Programs\\Python\\Python36-32\\lib', 'C:\\Users\\Ucze+ä\\AppData\\Local\\Programs\\Python\\Python36-32', 'C:\\Users\\Ucze+ä\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Users\\Ucze+ä\\AppData\\Local\\Programs\\Python\\Python36-32\\lib\\site-packages', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\pkgs\\IPython\\extensions']
2019-06-07 13:12:16,512 - root:115(run) INFO: Language code: pl_PL
2019-06-07 13:14:37,334 - root:112(run) INFO:

-----------------

Starting Mu 1.0.2
2019-06-07 13:14:37,334 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-FMFAHVA', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-06-07 13:14:37,334 - root:114(run) INFO: Python path: ['pkgs', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\Python\\python36.zip', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\Python', 'C:\\Users\\Ucze+ä\\Desktop\\mu\\pkgs', 'C:\\Users\\Ucze+ä\\AppData\\Roaming\\Python\\Python36\\site-packages', 'pkgs\\IPython\\extensions']
2019-06-07 13:14:37,334 - root:115(run) INFO: Language code: pl_PL

C:\Users\Uczeñ\AppData\Local\python\mu\Logs>

tmontes, 2019-06-07:
> I have only problems ;-( What a life...
> I want to help my friend who teach in one school and don't have access to administration of Windows, so cannot install any soft.
> I think - ok, lets try [...]

I'm not familiar with the Mu portable distribution, but the standard [Mu installer for Windows](https://codewith.mu/en/download) does not require admin privileges: it will complete a user-level installation, AFAIC. Did you give that a try?

PS: See "Step 5 - Choose Users"  in the [Installation guide](https://codewith.mu/en/howto/1.0/install_windows).

abixadamj, 2019-06-07:
Hmm.... @tmontes - I did not know about it. I will try...

abixadamj, 2019-06-07:
ok. I have tested, and:

mu-portable works ONLY, when there is NO Python at all in windows
mu-standard CAN be installed without admin priviliges.

tmontes, 2019-06-07:
> mu-standard works when:
> 
> 1. there is no python al all installed
> 2. there is python 3.6 installed
> 
> mu-standard does not work, when there is Python 3.7 installed.

Interesting. My experience is different on a Windows 10 system:

* Python 3.6 installed (user level).
* Python 3.7 installed (user level).
* Mu 1.0.2 installed (user level).

All "works" in the sense that Mu:
* Lanches from the Windows menu.
* Runs a Python program.
* Debugs a Python program.
* REPL works.

Installation notes: Python 3.6 is not in the PATH, but 3.7 is in the PATH: typing `python` in a `cmd.exe` CLI brings up the Python 3.7 REPL.

What do you mean by "**mu-standard does not work**, when there is Python 3.7 installed"?
* Does it launch from the Windows start menu?
* If not, what happens if you launch it from the `cmd.exe` CLI with:

```
C:\Users\...> cd AppData\Local\Mu
C:\Users\...\AppData\Local\mu> Python\pythonw.exe Mu.launch.pyw
```

* What if you use `python.exe` instead of `pythonw.exe` in the command above?

ntoll, 2019-06-07:
> I have only problems ;-( What a life...

@abixadamj when we next meet in real life, I'll buy you a beer or something... Believe it or not, you reporting all these problems is REALLY USEFUL, so please keep it up and keep in mind that beer I'll eventually buy you..!

:cow: :heart: :beers:

abixadamj, 2019-06-07:
I did not tried Python 3.6 and 3.7 both the same time. I tried only mu-standard 1.02 with python 3.6 - everything works perfect in the sense like you.
Strange, now I have installed 3.7 in no user/local path, I have installed it into c:\Python-37 (added to PATH), and mu-standard works with it!. This is strange, because @ntoll should remember I had problems with it. (I will remeber about beer ;-) - at some PyCon... next year.

At the end, I have hacked the system ;-) and run Pygame Zero from IDLE ;-)
[image]

I am doing this to help my son get A++ grade from Computer Science ;-)

tmontes, 2019-06-07:
> I did not tried Python 3.6 and 3.7 both the same time. I tried only mu-standard 1.02 with python 3.6 - everything works perfect in the sense like you.

I was describing my environment for the purposes of comparison. Having zero, one, or more versions of Python installed, user- or system-wide, should not affect a standard Mu installation. If it does, then there's something that needs fixing. :)

That's why I also went for more a more specific description of what "does not work" meant to you, while sharing possible next steps -- including running the installed Mu from the CLI -- to help understand the possible (but undesirable) interference of existing Python installations with the Mu installation, that your description suggested might be going on.

> Strange, now I have installed 3.7 in no user/local path, I have installed it into c:\Python-37 (added to PATH), and mu-standard works with it!. This is strange, because @ntoll should remember I had [...]

...so if Mu works, why are you going with IDLE? :)

abixadamj, 2019-06-07:
Because I do not have 100% sure if mu will work on computers in classrom in school. That is all.

ScottEP, 2019-09-27:
For whomever, in case it helps:

I had PortaMu 1.02 x64 and Python 3.6 x64 (user install) installed on Windows 10 Home x64 v1903 -- Mu worked.

I uninstalled Python 3.6 and installed Python 3.7 -- Mu broke:
-- Click the icon -- nothing happens.
-- Move the Python 3.6 paths before the Python 3.7 paths and try again -- nothing happens.
-- Remove the Python 3.7 paths and try again -- nothing happens.
-- Run "C:\Users\Me\OneDrive\Portable apps\PortaMu (portable Mu)\Launch Mu.bat" -- nothing happens.
-- Open a command prompt and cd to "C:\Users\Me\OneDrive\Portable apps\PortaMu (portable Mu)" and:
-- -- Run "C:\Users\Me\OneDrive\Portable apps\PortaMu (portable Mu)\Launch Mu.bat" -- nothing happens.
-- -- Run start Python\pythonw.exe "Mu.launch.pyw" /min -- nothing happens.
-- -- Run Python\pythonw.exe "Mu.launch.pyw" /min -- nothing happens.

*** But then: ***
-- -- Run Mu.launch.pyw -- *** success ***

Bonus f@ck-me:
-- Debug the *** broken *** command with WinDbg -- *** it works ***
-- -- Executable: C:\Users\Me\OneDrive\Portable apps\PortaMu (portable Mu)\Python\pythonw.exe
-- -- Arguments: "Mu.launch.pyw" /min
-- -- Start directory: C:\Users\Me\OneDrive\Portable apps\PortaMu (portable Mu)

ScottEP, 2019-09-27:
Forget to mention: I re-installed Python 3.6 right after I noticed that Mu was broken.