mu-editor/mu #857: Fail to make package on Windows 10: ImportError: Could not find 'mu'

bigeyex, 2019-06-11:
I tried to package Mu on Windows 10, by make win32 and make win64. 

The same error message appears:

INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\qtconsole\4.4.3\qtconsole-4.4.3-py2.py3-none-any.whl
INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\semver\2.8.1\semver-2.8.1-py2.py3-none-any.whl
INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\setuptools\28.8.0\setuptools-28.8.0-py2.py3-none-any.whl
INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\six\1.12.0\six-1.12.0-py2.py3-none-any.whl
INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\toml\0.10.0\toml-0.10.0-py2.py3-none-any.whl
INFO:nsist.wheels:Downloading wheel: https://files.pythonhosted.org/packages/9d/76/8a52c49e0a7e2638442f379d34f55232c773d4df7f69b297e451c980f99d/tornado-6.0.2-cp36-cp36m-win_amd64.whl

INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\traitlets\4.3.2\traitlets-4.3.2-py2.py3-none-any.whl
INFO:nsist.wheels:Using cached wheel: C:\Users\makeblock\AppData\Local\pynsist\pypi\wcwidth\0.1.7\wcwidth-0.1.7-py2.py3-none-any.whl
Traceback (most recent call last):
  File "C:\Users\makeblock\workspace\mu\package\mu_nsist.py", line 192, in <module>
    main()
  File "C:\Users\makeblock\workspace\mu\package\mu_nsist.py", line 182, in main
    ec = MuInstallerBuilder(**args).run(makensis=(not options.no_makensis))
  File "C:\Users\MAKEBL~1\AppData\Local\Temp\mu-pynsist-xjqxmf_j\mu-packaging-venv\lib\site-packages\nsist\__init__.py", line 471, in run
    self.prepare_packages()
  File "C:\Users\MAKEBL~1\AppData\Local\Temp\mu-pynsist-xjqxmf_j\mu-packaging-venv\lib\site-packages\nsist\__init__.py", line 348, in prepare_packages
    py_version=self.py_version, exclude=self.exclude)
  File "C:\Users\MAKEBL~1\AppData\Local\Temp\mu-pynsist-xjqxmf_j\mu-packaging-venv\lib\site-packages\nsist\copymodules.py", line 165, in copy_modules
    mc.copy(modname, target, exclude)
  File "C:\Users\MAKEBL~1\AppData\Local\Temp\mu-pynsist-xjqxmf_j\mu-packaging-venv\lib\site-packages\nsist\copymodules.py", line 115, in copy
    raise ImportError('Could not find %r' % modname)
ImportError: Could not find 'mu'
Copying installer file to C:\Users\makeblock\workspace\mu\dist
Traceback (most recent call last):
  File "win_installer.py", line 294, in <module>
    run(bitness, repo_root)
  File "win_installer.py", line 279, in run
    destination_dir,
  File "C:\apps\Python3.6\lib\shutil.py", line 241, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\apps\Python3.6\lib\shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\MAKEBL~1\\AppData\\Local\\Temp\\mu-pynsist-xjqxmf_j\\build\\nsis\\mu-editor_64bit.exe'

What could be the reason for this? Thanks.

carlosperate, 2019-06-11:
Did you `pip install .[dev]` before running make?
Maybe it need Mu to be installed before it can package it.

bigeyex, 2019-06-11:
> Did you `pip install .[dev]` before running make?
> Maybe it need Mu to be installed before it can package it.

After trying this, the output is the same.

carlosperate, 2019-06-11:
If you run this does Mu open?
```
$ python -m mu
```

In the python shell, can you import mu?

```
$ python
Python 3.7.3 (default, Apr  8 2019, 01:44:44) 
[Clang 10.0.1 (clang-1001.0.46.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mu
>>>
```

bigeyex, 2019-06-11:
> If you run this does Mu open?
> 
> ```
> $ python -m mu
> ```
> 
> In the python shell, can you import mu?
> 
> ```
> $ python
> Python 3.7.3 (default, Apr  8 2019, 01:44:44) 
> [Clang 10.0.1 (clang-1001.0.46.3)] on darwin
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import mu
> >>>
> ```

Both run well. Seemingly the packaging happens in a separate environment?

carlosperate, 2019-06-11:
Yes, it creates a venv and installs all the packages from the current environment into the new venv.
It then installs mu, and it looks like that part might not be working correctly:
https://github.com/mu-editor/mu/blob/2f5f0982defa67e227c7de14cfa9db615736b767/win_installer.py#L254

Could you copy the entire output from make?

bigeyex, 2019-06-11:
> Yes, it creates a venv and installs all the packages from the current environment into the new venv.
> It then installs mu, and it looks like that part might not be working correctly: [...]

Of course. Hope this will help.
[mu-make-win32.log](https://github.com/mu-editor/mu/files/3275668/mu-make-win32.log)

tjguk, 2019-06-11:
@tmontes feel free to pitch in on this. I'll have a look later when I have a little time, assuming someone else hasn't got there first...

tmontes, 2019-06-11:
Trying to help out, but without much time at the moment (will happily get back to the issue at hand, when I have more of that available!):

* Clearly, [pynsist](https://pypi.org/project/pynsist/) is failing to do its thing.
* The first failure is in `...\nsist\copymodules.py` while trying to copy the `mu` package.
* @carlosperate's note on installing things beforehand made sense.
* It is not 100% clear to me what's going on with @bigeyex's environment.

Before running `python make.py win32`, I wonder what the output of the following would be:
* `python -m pip freeze --all`
* `python -c 'import sys; print(sys.executable)'`
* `python -c 'import sys; print(sys.path)'`

More:
* What is the current working directory of the `python make.py win32` command? It should be the root of Mu's repository, IIRC.
* Is the `python make.py win32` running under an activated virtual environment where Mu was installed with `pip install -e ".[dev]"`? How was the virtual environment created?

Something that should work:
1. Create a virtual environment with `<full-path-to-python-executable> -m venv venv`.
2. Activate it with `.\venv\Scripts\activate`.
3. Change working directory to Mu's repository root.
4. Install with `pip install -e ".[dev]"`.
5. Package with `python make.py win32`.

...is this similar to what you've done, @bigeyex?

PS: I'm not 100% sure that step 3 above is really needed but, like I commented, I'm a bit tight on availability at the moment. :(

More PS:
* Interestingly, this log line does not indicate that Mu was installed:
```
Successfully installed PyQt5-5.12.1 PyQt5-sip-4.19.17 PyQtChart-5.12 QScintilla-2.11.1 appdirs-1.4.3 attrs-19.1.0 backcall-0.1.0 black-19.3b0 click-7.0 colorama-0.4.1 colorzero-1.1 decorator-4.4.0 gpiozero-1.5.0 guizero-0.6.4 ipykernel-5.1.1 ipython-7.5.0 ipython-genutils-0.2.0 jedi-0.13.3 jupyter-client-5.2.4 jupyter-core-4.4.0 nudatus-0.0.3 numpy-1.16.4 parso-0.4.0 pgzero-1.2 pickleshare-0.7.5 pigpio-1.42 prompt-toolkit-2.0.9 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pygments-2.4.2 pyserial-3.4 python-dateutil-2.8.0 pyzmq-18.0.1 qtconsole-4.4.3 semver-2.8.1 six-1.12.0 toml-0.10.0 tornado-6.0.2 traitlets-4.3.2 wcwidth-0.1.7
```
* Instead, the first logged message in that "sequence" is:
```
Creating the packaging virtual environment.
Installing mu with C:\Users\MAKEBL~1\AppData\Local\Temp\mu-pynsist-y33_mirv\mu-packaging-venv\Scripts\python.exe
Processing c:\users\makeblock\workspace\mu
  Requirement already satisfied (use --upgrade to upgrade): mu-editor==1.1.0a1 from file:///C:/Users/makeblock/workspace/mu in c:\users\makeblock\workspace\mu
```
* Both pretty much inline with `pynsist` complaining about not finding the `mu` package! :)

bigeyex, 2019-06-17:
Hi, @tmontes , I tried the instructions above with no luck. However, I have built Mu successfully with another computer, and that's pretty strange.

Anyway, I'll post my log and executable/path information below. Hope that may provide some insight. I'm glad to try more if it can help the project. And feel free to close this issue if needed.

Log File:
[mu-make-win32-venv.log](https://github.com/mu-editor/mu/files/3295186/mu-make-win32-venv.log)

Executable/path info:
```
>>> import sys
>>> print(sys.executable)
C:\apps\Python3.6\python.exe
>>> print(sys.path)
['', 'C:\\apps\\Python3.6\\python36.zip', 'C:\\apps\\Python3.6\\DLLs', 'C:\\apps\\Python3.6\\lib', 'C:\\apps\\Python3.6', 'C:\\Users\\makeblock\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\apps\\Python3.6\\lib\\site-packages', 'c:\\users\\makeblock\\workspace\\mu']
```

tmontes, 2019-06-17:
@bigeyex, thanks for your patience and for posting further details.

Like you said, something strange seems to be going on, given that you have successfully built Mu with another system (like myself, and others!).

## Log analysis

The log file you attached indicates a few interesting things:
* You setup and activate a new virtual environment in lines 1-5.
* You then `pip install -e ".[dev]"` under `C:\Users\makeblock\workspace\mu`, which I suppose is your git repository clone.
* Lines 8-199 are the output of the installation.
* In particular, line 197 clearly indicates that the packaged `mu-editor` was installed (albeit in development mode, as requested). Good.
* It is interesting, as noted in line 198, that you are running a somewhat oldish `pip` version.
* Lines 214 onwards are produced by the `win_installer.py` script:
  * Line 214 is produced by the `print` call in `win_installer.py:248`.
  * Line 215 is produced by the `print` call in `win_installer.py:250`.
  * Line 216 is produced by the `print` call in `win_installer.py:253`.
  * Lines 217-306 are produced by the subprocess call to `pip install mu` in `win_installer.py:254` (yes it is supposed to do that!).
  * These lines are **very interesting**:
    * Line 218 states that mu is already installed (which it isn't in the sub-virtual-environment created by `win_installer.py`, logged in line 215).
    * Line 304 confirms that mu was not installed.
    * Lines 305-306 tell us about the availability of a newer `pip` version.
* Later in the log, line 464, is produced by `win_installer.py:271` when it is about to run `pynsist` to create the windows installer package (after having prepared everything up until that point).
* Lines 465-524 are the output of that:
  * Lines 511-524 are a traceback from `pynsist` complaining that it did not find the `mu` package (it wasn't installed, remember?).
* Lines 525-* are a subsequent failure where `win_installer.py` tries to copy the installable windows executable but fails because it was never produced.

To summarize: the underlying issue seems to be due to the fact that the `mu` installation within the "sub-virtual-environment" that `win_installer.py` needs to create to prepare the packaging process is skipped (lines 218 and 304 in the log file).

I wonder if it could be due to some distinct behaviour from the older `pip` version.

PS: Not that is is really relevant now, but what I was looking for was for the output of those commands after creating and activating the virtual environment (lines 1-5) and before running the packaging process (before line 202!).

## Things you can try

The first thing you could check is the `pip` version for the system where the packaging worked for you. I suppose it will be more recent that the `9.0.1` indicated in your log for the system that fails to package Mu.

The other thing (hack) you could try is:
* Edit `win_installer.py` and locate lines 253-254 with:
```
print('Installing mu with', venv_python)
subprocess.run([venv_python, '-m', 'pip', 'install', repo_root])
```

* Add a line between those two, updating `pip` itself, such that those two lines become three:
```
print('Installing mu with', venv_python)
subprocess.run([venv_python, '-m', 'pip', 'install', '--upgrade', 'pip'])
subprocess.run([venv_python, '-m', 'pip', 'install', repo_root])
```

Now try the `make win32` after this *fix attempt*. Does it work now?
If it does, we can then work together in creating a pull request to improve Mu's windows packaging. :)

Thanks again for your patience.
Let us know of your progress!

bigeyex, 2019-06-25:
Hi, sorry for the delay. I tried your suggestion (updating pip in venv), guess what? It worked! Will work on a pr for that :)

tmontes, 2019-06-25:
@bigeyex, those are great news. :)

Let me know if you need any assistance in the PR, will you?
PS: Maybe the docs could be updated to recommend updating `pip` too, what do you say?

tmontes, 2019-07-04:
...ok, the fix has been merged in PR #879. Shall we update the docs, accordingly? :)

bigeyex, 2019-07-08:
Hmm, I figured out where the doc is and issued a pull request.
There is still question about whether we should recommend updating local pip. For example, my machine that fails to package says that pip is already up-to-date (which is strange). But updating the local pip is harmless whatever.

ntoll, 2019-07-08:
Heh... So there's this phenomenon called Schroedinger's `pip` (which I've just invented) - your OS packaged version of `pip` may/will/forces pip to report it's up to date (thus NOT breaking the OS's packaged version of pip). But then the "official" `pip` from Python will report something else. Hard to tell which is which until you check... and even then, are you using the `pip` you think you're using (aka `pip` vs `pip3` vs `its-pip3-but-called-pip`, `pip-in-the-virtualenv-isnt-the-pip-you-think-it-is` etc....).

Here's a happy Packaging Platypus to cheer you up... ;-)

[image]

Of course... there's this from PyCon 2019 too...

[image]

carlosperate, 2019-07-08:
Let's not forget that when pip was updated to v10 it broke a bunch of package installations that were (incorrectly) using pip internals in setup.py, or other parts of the codebase.
So I would vote to leave the local pip version as it is, and only update the internal pip to the Mu installation/virtualenv, where we know it won't break anything else.

ntoll, 2019-07-08:
Heh... the `pip` devs EXPLICITLY tell folks NOT to use the pip internals and codebase. I really feel for those folks (there's apparently only 5 of them). They're not financially supported by the PSF for their work (IMHO they should be) and they do an amazing job in difficult circumstances. Such is the world in which we live. :-/

tmontes, 2019-07-08:
(I'll share a quick comment here, given that PR #884 has been merged already)

Short ideas:
* IIUC, the PR recommends that whichever is "the base pip" is updated. For the reasons stated above, I think this is bound to both fail and be useless in the context of the virtual environment.
* What it should instead recommend is create virtual env, then update pip, then install Mu.
* Additionally, on Windows, the command `pip install --upgrade pip` is bound to fail due to the way Windows handles open file renames/deletes. Instead, the more generic alternative should be used `python -m pip install --upgrade pip`.

My 2c from not-so-sunny, but warm Basel. :)

carlosperate, 2020-11-08:
Can this issue be closed with the merge of https://github.com/mu-editor/mu/pull/884?

carlosperate, 2021-03-24:
We can assume this is resolved and this issue can be opened again if anybody finds this is still an issue.

closed 2021-03-24 by carlosperate