mu-editor/mu #858: Problem with esp32 and mu-editor on windows 7

sbelin, 2019-06-13:
Hello,
I use esp32 (M5stack) device with mu_editor version 1.1.0. alpha.1 on macos.
It works very well.
When I use the same device with the same version under windows 7, I can not access the files.
I do not understand.
Thank you in advance,

Firmware on the M5stack : UIFlow-v1.2.3 (MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on M5Stack with ESP32)

Here is the contents of the log file:
2019-06-13 15: 31: 54,112 - mu.logic: 1162 (sync_package_state) INFO: Synchronize package states ...
2019-06-13 15: 31: 54,112 - mu.logic: 1163 (sync_package_state) INFO: Old: set ()
2019-06-13 15: 31: 54,112 - mu.logic: 1164 (sync_package_state) INFO: New: set ()
2019-06-13 15: 31: 55,516 - mu.modes.esp: 184 (toggle_files) INFO: Toggle filesystem off.
2019-06-13 15: 31: 56,079 - mu.modes.base: 229 (find_device) INFO: Found on COM3
2019-06-13 15: 31: 56,079 - mu.modes.base: 230 (find_device) INFO: Serial number: 013EFE32
2019-06-13 15: 31: 56,086 - mu.modes.esp: 180 (toggle_files) INFO: Toggle filesystem on.
2019-06-13 15: 31: 57,155 - mu.modes.base: 403 (ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
File "C: \ Users \ Desktop \ AppData \ Local \ Mu \ pkgs \ mu \ modes \ base.py", line 400, in ls
result = tuple (microfs.ls (self.serial))
File "C: \ Users \ Desktop \ AppData \ Local \ Mu \ pkgs \ mu \ contrib \ microfs.py", line 181, in ls
], serial)
File "C: \ Users \ Desktop \ AppData \ Local \ Mu \ pkgs \ mu \ contrib \ microfs.py", line 132, in execute
raw_on (serial)
File "C: \ Users \ Desktop \ AppData \ Local \ Mu \ pkgs \ mu \ contrib \ microfs.py", line 82, in raw_on
raise IOError ('Could not enter raw REPL.')
OSError: Could not enter raw REPL.
2019-06-13 15: 31: 57,156 - mu.interface.main: 740 (show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging / plugging-in your device and / or restarting Mu.
2019-06-13 15: 31: 57,157 - mu.interface.main: 741 (show_message) DEBUG: None
2019-06-13 15: 32: 00,286 - mu.logic: 1120 (show_admin) INFO: Showing admin with logs from C: \ Users \ Desktop \ AppData \ Local \ python \ mu \ Logs \ mu.log
2019-06-13 15: 32: 00,287 - mu.logic: 157 (installed_packages) INFO: Packages found: []

carlosperate, 2019-06-13:
It looks like Mu cannot enter the "raw REPL" mode.
This can sometimes be caused if the program running on the MicroPython device is in a tight loop and sending the "Ctrl+C" command doesn't take, or the program sends too much data via serial.

Can you delete any programs you might have running on the device and try again?

WayneKeenan, 2020-01-17:
This is occurring for me because there is code in `boot.py` that is taking a few seconds to complete before presenting the REPL prompt.  
Is there a way to configure the timeout before Mu gives up?

dybber, 2020-09-29:
This should be fixed if #1114 gets merged

closed 2020-10-01 by dybber