mu-editor/mu #859: What’s the best way to share a Code With Mu project?

arfrankq, 2019-06-16:
Is there a recommended way to share a Code With Mu project, by which I mean everything needed for a project to run - not just the code, but all the assets (such as images, sounds etc.) too? I am particularly interested in easy ways to share projects within groups of young coders (e.g. a jam or other coding club, or a school class etc).

I guess this is really two questions:

1. Can a Code With Mu project be "exported" in some way so that someone else can "import" it and all the code and other assets will end up in the right place so the code will run and work straight away?

2. Is there a preferred or recommended way to share projects amongst a specific group of people? Github requires you to have (or open) an account and it's an open platform - so the project is essentially shared with the whole world. I wonder if there are options or alternatives which would work for a slightly smaller group than that! Some people may not feel ready to share their code with the world and/or may not be willing or able to sign up to the the Terms and Conditions - e.g. they normally include at least a minimum age limit (13 for github) so aren’t available to younger coders.

I've mentioned this to @ntoll elsewhere (I'm opening this issue at his suggestion) and I know he has some ideas, as do others in the Mu community, so am really looking forward to hearing your thoughts and suggestions.

arthomnix, 2019-06-16:
Maybe in the next Mu release, you could add a zip-based format to export projects (put the Python and assets in a zip file with a custom extension). This is basically what is used by Office Open XML (Microsoft Office) and OpenDocument (LibreOffice/OpenOffice) files (just try renaming a Word document to a .zip and unzipping) and Java's jar files, so I wouldn't really consider it a hack. 

As for sharing projects, maybe there should be an official Mu Git hosting service, like a self-hosted GitLab, Bitbucket or Gitea. Note that they have to be self hosted as the non-self-hosted versions have age restrictions as @arfrankq mentioned.

arthomnix, 2019-10-05:
However, would there be a legal obligation to have an age restriction even if it was self-hosted? It's hard to make an account anywhere without collecting information, assuming that an email address counts as information. Probably easiest to get parents/carers to upload to GitHub on young coders' behalf.

ZanderBrown, 2019-10-06:
email definitely counts as private info, and even if it didn't assuming children have an email account isn't great (though probably true....)

arthomnix, 2019-10-06:
What actually is the law around age restrictions? For example, schools need to collect private information about pupils, but nobody's going to ban school until age 13. Also, you are allowed to collect data from children under 13 **with parental consent**, but you have to make "reasonable efforts" to prove that the person providing consent is actually a parent/carer, which sounds difficult (maybe why social networks don't do this - they'd love the extra users).

ZanderBrown, 2019-10-06:
Sure schools get around it because

A) They are schools
B) Parents sign a bit of paper to say it's okay

A doesn't apply to us and I'm not sure B is particularly practical :-)

Of course the simple solution (as used by various education-related services) is to either not require an email address or make it optional

But that somewhat discounts a git based solution and I'm not sure anyone has the time/money to maintain a custom service

devdanzin, 2020-12-26:
Creating "projects" should probably better be discussed on #659.

Regarding sharing, a GitHub for Kids would be lovely, but there are very serious reasons one doesn't exist. However, some big tech players that allow featureful children accounts.

By far, IME, Google and Microsoft children accounts give you the best control/flexibility. Each child account is administered by the parent(s), who can control (the many, oh so many) services, review usage, etc. The kids get monitored access to e.g. Google Drive and other content creation/sharing services. 

It's even possible that Google kid account could get access to Colab and run Python notebooks in the cloud ([alas](https://github.com/googlecolab/colabtools/issues/1328#issue-637301980)).

Sure, I'd not like to be the one to tell a parent "hey, why don't you create a Google account for your kid?". Whole can of worms, some people might be rightfully offended by that. But you can ask whether anyone has such an account. And you can still use your Drive to make your content available (and versioned) and even post what the kids (or their parents) send you by, e.g., email. Depending on your target audience, you could reduce your workflows to "kids with accounts" and "parents I send links to".

If you want bespoke project sharing, maybe the way to achieve that while benefiting from children accounts would be to write an app that uses either Google's or Microsoft's platforms without storing anything itself.

closed 2021-03-24 by carlosperate