mu-editor/mu #861: How to disable automatic recording in mu-editor ?

BEZIAT11, 2019-06-17:
Hello, could you tell me how to disable automatic recording in Mu editor. My goal is to be able to make different versions of the same program.

Thank you for you help.

Guillaume

tmontes, 2019-07-15:
@BEZIAT11, thanks for using Mu and for your question (also thanks for your patience, we're all volunteers here and sometimes we don't have as much time as we'd like to help out with Mu!).

Maybe you mean "auto-saving"? If that is the case, the way you can create different versions of the same program is as follows, which is compatible with the existing "auto-save" feature of Mu (if that's not what you mean, please could you try to rephrase it?)

1. Open Mu.
2. Close all editor tabs by clicking the "x" in them.
3. Create a new editor tab.
4. Edit at will and then save it: call it "program_1.py".
5. Double click the editor tab: call it "program_2.py".
6. Edit at will and save (this will be your second version).
7. Double click the editor tab: call it "program_3.py".
8 Edit at will and save (this will be your third version).

At any time during the process, you are free to click Open to open any other program, including other versions of the same program which you might have saved before.

Is this helpful to your use case?

tmontes, 2019-10-05:
Hey @BEZIAT11, we haven't heard from you in quite a while... :( Did you manage to have different versions of the same program, with the hints I shared? If so, could you please close this issue? Thanks!