mu-editor/mu #866: Not Detects ""Clones"" adafruit

tcpipchip, 2019-06-24:
If your board is not the original adafruit board,
[image]
Not detects the Board!
For example, i ported CircuitPython to u-blox module (NRF52840), but the mu editor not detects it (only the Mass storage), not the serial
[image]

ZanderBrown, 2019-07-22:
Mu whitelists devices for the Adafruit/CircuitPython mode

https://github.com/mu-editor/mu/blob/acf9cb00893d776b5fe52d1c8c4fb9ab04c48503/mu/modes/circuitpython.py#L38-L51

I suspect your device isn't in that list, if it is compatible I suggest opening an PR

carlosperate, 2020-09-29:
@tcpipchip as @ZanderBrown the board VID and PID are included in the Mu source code to identify the compatible boards. If you have any board you'd like to add you can create a PR to add that.
Thanks!

closed 2020-09-29 by carlosperate

tcpipchip, 2020-09-29:
I use the Adafruits! Thanks!