mu-editor/mu #872: Mu copies junk to device on startup

inventhouse, 2019-06-30:
If my device is connected when I start Mu, it automatically and without prompting creates a number of directories and copies assets I don't want, apparently for a board I'm not using, to my device.  These include `fonts/`, `images/`, `music/`, and `sounds/`.

It seems I hadn't updated Mu in quite some time, as the offending code has been in there for about a year: https://github.com/mu-editor/mu/blob/3393a2ca7ee793df6a2c9d655ca926dd86074e4a/mu/logic.py#L626

* On any of these devices 50-60KB of unnecessary assets is an awful lot, on my Trinket M0 based project it exhausts "disk" space
* If there isn't sufficient space on the device, Mu crashes on launch (which I'll file as a separate issue)
* I'm not using a pygamezero, why is code for that running on every startup of Mu anyway?
* _Always *always* ask before copying things to the user's board_

I'm running Mu 1.0.2 on macOS Mojave 10.14.5 (18F132) and have confirmed the problem with Adafruit Trinket M0, ItsyBitsy M0, and ItsyBitsy M4; log is pasted below.

# Mu log:

2019-06-30 07:34:07,460 - mu.logic:1053(quit) DEBUG: Session: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-30 07:34:07,504 - mu.logic:1054(quit) DEBUG: Saving session to: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:34:07,505 - mu.logic:1068(quit) INFO: Quitting.

2019-06-30 07:35:56,797 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-06-30 07:35:56,797 - root:112(run) INFO: uname_result(system='Darwin', node='Inara.local', release='18.6.0', version='Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-06-30 07:35:56,797 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-06-30 07:35:56,797 - root:114(run) INFO: Language code: en_US
2019-06-30 07:35:57,186 - mu.logic:543(__init__) INFO: Setting up editor.
2019-06-30 07:35:57,186 - mu.logic:562(__init__) INFO: Settings path: /Users/bjh/Library/Application Support/mu/settings.json
2019-06-30 07:35:57,186 - mu.logic:563(__init__) INFO: Session path: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:35:57,186 - mu.logic:564(__init__) INFO: Log directory: /Users/bjh/Library/Logs/mu
2019-06-30 07:35:57,186 - mu.logic:565(__init__) INFO: Data directory: /Users/bjh/Library/Application Support/mu
2019-06-30 07:35:57,198 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-06-30 07:35:57,199 - mu.logic:590(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/images
2019-06-30 07:35:57,237 - mu.logic:597(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/fonts
2019-06-30 07:35:57,238 - mu.logic:600(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/sounds
2019-06-30 07:35:57,240 - mu.logic:605(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/music
2019-06-30 07:35:57,394 - mu.logic:1150(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-06-30 07:35:57,399 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:35:57,399 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-30 07:35:57,399 - mu.logic:649(restore_session) INFO: Loaded files.
2019-06-30 07:35:57,399 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-06-30 07:35:57,399 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-30 07:35:57,488 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-06-30 07:35:57,521 - mu.logic:1150(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-06-30 07:40:36,424 - mu.logic:1053(quit) DEBUG: Session: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-30 07:40:36,424 - mu.logic:1054(quit) DEBUG: Saving session to: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:40:36,425 - mu.logic:1068(quit) INFO: Quitting.

2019-06-30 07:40:40,513 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-06-30 07:40:40,513 - root:112(run) INFO: uname_result(system='Darwin', node='Inara.local', release='18.6.0', version='Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-06-30 07:40:40,513 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-06-30 07:40:40,513 - root:114(run) INFO: Language code: en_US
2019-06-30 07:40:40,869 - mu.logic:543(__init__) INFO: Setting up editor.
2019-06-30 07:40:40,869 - mu.logic:562(__init__) INFO: Settings path: /Users/bjh/Library/Application Support/mu/settings.json
2019-06-30 07:40:40,869 - mu.logic:563(__init__) INFO: Session path: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:40:40,869 - mu.logic:564(__init__) INFO: Log directory: /Users/bjh/Library/Logs/mu
2019-06-30 07:40:40,869 - mu.logic:565(__init__) INFO: Data directory: /Users/bjh/Library/Application Support/mu
2019-06-30 07:40:40,880 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-06-30 07:40:40,881 - mu.logic:590(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/images
2019-06-30 07:40:40,904 - mu.logic:597(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/fonts
2019-06-30 07:40:40,905 - mu.logic:600(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/sounds
2019-06-30 07:40:40,906 - mu.logic:605(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/music
2019-06-30 07:40:41,470 - mu.logic:1150(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-06-30 07:40:41,478 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/bjh/Library/Application Support/mu/session.json
2019-06-30 07:40:41,478 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-06-30 07:40:41,478 - mu.logic:649(restore_session) INFO: Loaded files.
2019-06-30 07:40:41,479 - mu.logic:653(restore_session) INFO: User defined environment variables: []
[... 9 lines cut ...]
-----------------

Starting Mu 1.0.2
2019-06-30 07:51:51,986 - root:112(run) INFO: uname_result(system='Darwin', node='Inara.local', release='18.6.0', version='Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-06-30 07:51:51,986 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-06-30 07:51:51,986 - root:114(run) INFO: Language code: en_US
2019-06-30 07:51:52,285 - mu.logic:543(__init__) INFO: Setting up editor.
[... cut ...]
2019-06-30 07:51:52,455 - mu.logic:649(restore_session) INFO: Loaded files.
2019-06-30 07:51:52,455 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-06-30 07:51:52,455 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-06-30 07:51:52,526 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-06-30 07:51:52,551 - mu.logic:1150(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-06-30 07:51:58,406 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/bjh/Library/Logs/mu/mu.log

ntoll, 2019-07-01:
@inventhouse Hi, thanks for reporting this and #873. I think this may have been fixed in 1.1.alpha.1, but if not I'll make sure it is fixed in the next alpha (due out this week).

dglaude, 2020-01-08:
I had this problem on OSX.
So I made a full re-installation of the Alpha version of Mu for OSX: https://github.com/mu-editor/mu/releases/download/1.1.0-alpha.2/mu-editor_1.1.0-alpha.2_osx.dmg

Then I tried plugging and unplugging alternatively a PyGamer (that already contain that junk) and a PyRuler (that does not have enough space for this).

As long as Mu was running I could plug and unplug PyRuler and PyGamer without side effect.

But if PyRuler is plugged (with CIRCUITPY volume visible) and then I start Mu, it start fill the file system until it is full.

I will try this one more time a full uninstall/reinstall trying to find the shortest reproduction path. This time I want to do it without using my PyGamer, but only with my PyRuler...

dglaude, 2020-01-09:
@ntoll I have a "nice" reproduction sequence and it does not involve a PyGamer (it might fail with any board that expose a folder CIRCUITPY)

1. Uninstall the application
2. Connect a PyGamer (make sure there is no image folder and space left, empty the trash)
3. Download the DMG
4. Open the DMG
5. Drag and drop Mu to Application
6. Wait for the end of the installation
7. Close the installer and the DMG
8. Start Mu (have to say you trust the internet)
9. Watch LIVE in Finder the folder "images" being created, files "alien_hurt.png", "alien.png", "cat1.png" and you file system is full

The only thing I am not totally sure of is if my uninstall procedure is sufficient... right now I drag and drop Mu in the Trash Bin and empty it. I just hope there is no left over from one installation to another.

Once your CIRCUITPY is full, here are the option:

A) If you don't touch your drive, stop Mu, restart Mu ...

At that point, Mu is stuck, likely because CIRCUITPY is full and it tries to put even more junk.

B) You stop Mu, remove the image folder, empty the trash (not needed it seems), restart Mu ... 

At that point Mu start, create that image folder, put as many file as it can and then you can use Mu (but you file system is full already).

I really hope it help.

ntoll, 2020-01-10:
OK... this is definitely a bug and many many thanks for the steps to reproduce this problem, that's really appreciated. 

Basically, the problem is that when Mu starts for the first time it sets itself up with various configuration files and default assets used in demo applications. That's the junk you're seeing copied onto your device. My guess is the path is referenced by the wrong thing, or being set to the wrong value as a result of the steps you outline above. I'll make sure this is fixed in the next release.

Thanks once again for the heads up.

devdanzin, 2020-12-08:
As a quick fix, we could check for free disk space and avoid copying the files if it's under a given threshold:

```python
        import shutil
        MIN_FREE_SPACE = 1024 * 1024 * 10   # 10MB
        du = shutil.disk_usage(wd)[2]
        if du > MIN_FREE_SPACE:
            # Ensure PyGameZero assets are copied over.
            ...
```

Mu seems to work fine without the assets in place, and will even create the right empty folders from clicking the resource buttons. But it would be a new failure mode we'd need to tell users about ("I installed Mu but there's no alien image").

carlosperate, 2020-12-09:
Just to clarify, the files copied are for the pygame zero mode, which should go into the `mu_code` directory.
The bug in this case is that the files are somehow ending up in the CIRCUITPY drive instead of the `mu_code` folder.

This is probably because on the first start-up Mu finds the device and if it switches into CircuitPython mode it changes the working directory to the CIRCUITPY drive, and then it copies the pygame zero files there instead of `mu_code` (the startup code probably expects that to be the working directory).

ntoll, 2021-07-10:
I believe this has been fixed. When creating the `mu_code` workspace directory, Mu always uses the path provided by the "default" `python` mode (which always refers to the path on the local filesystem, rather than that on any attached USB device).

closed 2021-07-10 by ntoll