mu-editor/mu #873: Mu crashes on launch if device is connected and has little free space

inventhouse, 2019-06-30:
If a device with little free "disk" space is connected when Mu is started, it silently crashes without surfacing any error message in a way that a beginner might see.

In `setup`, Mu attempts to copy a number of assets to the device (#872), when that fails there isn't sufficient error handling to catch and surface the problem to the user, especially a novice.

As a workaround, I can "eject" my device, start Mu, then re-connect my board.

I'm running Mu 1.0.2 on macOS Mojave 10.14.5 (18F132) and have confirmed the problem with Adafruit Trinket M0; I don't know where the startup log is stored, but I diagnosed this issue by starting it directly on the command-line:

$ /Applications/mu-editor.app/Contents/MacOS/mu-editor
Overriding built-in editor font due to Issue #552
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 144, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 598, in setup
    os.makedirs(fonts_path)
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/os.py", line 220, in makedirs
OSError: [Errno 28] No space left on device: '/Volumes/CIRCUITPY/fonts'

carlosperate, 2020-11-08:
Thanks for the repor @inventhouse.

> If a device with little free "disk" space is connected when Mu is started, it silently crashes without surfacing any error message in a way that a beginner might see.

Does this mean the CircuitPython device crashes or Mu crashes?

TonyLHansen, 2022-04-27:
I just ran into this issue as well. It still exists with mu version 1.1.1.

- I can confirm that it is **mu** that crashes and not CirciutPython.
- The current version 1.1.1 takes you to the crash page via a browser. So the user is not TOTALLY in the dark. However, there is nothing to indicate WHY mu is trying to do what it is doing. So the user doesn't have much of a chance.
- The code is erroneously trying to create copies of pygamezero resources on the device: images, fonts, sounds and music.
- There is no reason whatsoever that `mu` should be creating these directories or copying these resources to the CircuitPython device.
- The bad code is in logic.py Editor::setup().
- I've attached a log file showing the crash. [mu.log](https://github.com/mu-editor/mu/files/8572308/mu.log)
- My crash was on `music_path` instead of `font_path` (as reported above) because I DO have a `font_path` directory on the device. 

My workaround was the same: disconnect the device, start up mu, then reconnect the device.

TonyLHansen, 2022-04-27:
Bottom line, I consider there to be two bugs here:

- `get_default_workspace()` should never return a directory path on the device. (Perhaps there could be a check in `get_default_workspace()` to return the default directory if it finds that the path has the device path at its beginning.)
- Whatever code it was that SET "workspace" in `settings.json` to the device path ("/Volumes/CIRCUITPY") is probably incorrect.