mu-editor/mu #874: Feature request: configurable keyboard shortcuts

inventhouse, 2019-06-30:
I'm used to `Cmd-/` to toggle commenting selected lines, but Mu hard-codes the keyboard shortcut for that instead of letting me set it: https://github.com/mu-editor/mu/blob/4344548bc25df03ffb89b9535aec7dd3b033518b/mu/app.py#L154

I would like to be able to configure my keyboard shortcuts, either in a preferences editor or in a configuration file of some kind.

ntoll, 2019-07-01:
@inventhouse Hi, thank you for this feedback.

Just to let you know, we have a "rule of three" when it comes to features. If three people have independently asked for it, then we look to implement it. This is usually done via a "spike" PR around which discussion and input can form before we finalise the implementation. This is the first time anyone has asked for customisation of key bindings.

Just so you have some context:

* The shortcuts are based upon the most common we know of in other graphical editors.
* Since Mu is beginner focused we're trying to get folks used to keyboard shortcuts they'll probably use when the graduate away from Mu (which is the point of Mu, it helps you to graduate away from it, to something more serious -- we liken it to the toddling stage between crawling and walking.
* As a result, it's *intentionally* not supposed to be that configurable... if you're asking to configure it, you're probably of an advanced enough level to not really have to use Mu, if you see what I mean. ;-)

However, to counter-act that third point, we have the rule of three. ;-)

I hope this makes sense... and please don't think I'm saying "nope". I'm just giving a bit of context so you know we're not pushing back... just trying to keep Mu simple and beginner focused (but within a framework for making features happen, when folks let us know they need 'em).

inventhouse, 2019-07-04:
Thanks for the detailed explanation; I agree that keeping a "beginner" focus is good.  As for points two and three: you're right, I don't use Mu as my primary editor (hence being used to other shortcuts 😉 ) but there are some things is really darn convenient for - serial console, plotter, and making small edits right on the device like adding or _commenting out_ log lines and such.

I tried to hack my copy to change the shortcut but I'm really unfamiliar with Qt and wasn't able to get `Ctrl+/` to work; do you know what string it needs to make that work for me?  Thanks!

ntoll, 2019-07-04:
Perhaps you need to escape the "/" as `Ctrl+\/`..? (I'm just guessing here)

inventhouse, 2019-07-04:
Nope, `Ctrl+\/` didn't work either 🤷‍♂   Anyway, maybe a couple more requests will filter in, but it's ok that it's not a super-high priority.

tmontes, 2019-07-04:
Investigated a bit.

It seems that the editor component, QScintilla, is handling CTRL-/ by itself. My observation is that hitting that key combination moves the cursor to the previous word (or something similar).

Here's a minimal code sample that illustrates the behaviour:
```
#!/usr/bin/env python3

import sys
from PyQt5.QtWidgets import QApplication
from PyQt5.Qsci import QsciScintilla

if __name__ == '__main__':
    app = QApplication(sys.argv)
    editor = QsciScintilla()
    editor.show()
    sys.exit(app.exec_())
```

Unfortunately, the docs for QScintilla weren't helpful enough in guiding me through the possibility of disabling what I imagine is an out-of-the-box key configuration.

PS: Tested on macOS, but I presume behaviour will be equivalent on other platforms.

inventhouse, 2019-07-04:
Interesting!  I'm also on macOS and, yes, I see the back-by-word behavior, but didn't know why setting a specific key combo didn't override the built-in one, so I assumed I wasn't doing it right 🤷‍♂ Thanks!

closed 2021-03-24 by carlosperate