mu-editor/mu #875: ModuleNotFoundError: No module named 'PyQt5.sip'

tim1mw, 2019-07-01:
I've installed Mu using pip3 on Mageia 7, I'm getting the following error:

Traceback (most recent call last):
  File "/usr/local/bin/mu-editor", line 6, in <module>
    from mu.app import run
  File "/usr/local/lib/python3.7/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ModuleNotFoundError: No module named 'PyQt5.sip'

I've got the python3-pyqt5-sip distro package installed (4.19.16), but that doesn't contain PyQt5.sip and a search of the package repository suggests this file is no longer available in any package. Digging around the web there seem to be quite a few bug reports against various packages suggesting that this file is no longer packaged in the latest distros. 

The answer appears to be in this post:

https://raspberrypi.stackexchange.com/questions/87125/installing-pyqt5-on-raspberry-unable-to-imoprt-pyqt5-sip

It seems that PyQt5.sip is no longer generated by the default settings in the sip build process, it has to be explicitly configured and presumably my linux disto and quite a few others aren't doing this. Using:

pip3 install --user -I PyQt5-sip

To overwrite the distro packaged version with 4.19.17 doesn't help. I'm guessing if I go and make my own build of PyQt5.sip that will solve the problem but that will be a task for another day.

tim1mw, 2019-07-05:
I had a go a few days ago at generating the necessary PyQt5.sip module by rebuilding from source using the official instructions here:

https://www.riverbankcomputing.com/static/Docs/PyQt5/installation.html#building-and-installing-from-source

However, despite following these instructions the module was never built. This isn't the place to solve that particular problem, I'm posting this for reference since this is going to hit increasing numbers of Linux users as people upgrade to distros that lack this vital file and not everybody has the skills to do source builds. However, if Mu can be patched to remove the dependency on PyQt5.sip that would be great or alternatively an AppImage that avoids the problem would also be useful.

tmontes, 2019-07-06:
> However, if Mu can be patched to remove the dependency on PyQt5.sip that would be great or alternatively an AppImage that avoids the problem would also be useful

@tim1mw, I'm not really sure if that is possible or even makes sense, being unfamiliar with `PyQt5` implementation details.

Mu, by itself, does not depend directly on `PyQt5.sip`. As your traceback details show, the `PyQt5.sip` import failure results from Mu's `from PyQt5.QtCore import QTimer, Qt`. So importing `PyQt5.QtCore` is triggering the failure. I presume this should work on a "regular" PyQt5 installation.

Can you isolate and reproduce that failure in your system? I mean:

* Create an isolated virtual environment (or not, if using system Python + distribution packaged PyQt5).
* Install PyQt5 into it (or use the distribution packaged version, I guess).
* Create a single line program with:
```
from PyQt5.QtCore import QTimer, Qt
```
* Does it run successfully or does it fail?

If it fails, my initial feeling is that whoever packaged/distributed PyQt5 forgot something... Or maybe I'm mistaken (there's that serious possibility!) and something should/must be approached in a different way.

My 2c.

carlosperate, 2019-07-08:
How exactly did you install Mu and it's dependencies, and could we get the terminal output? The more detail in all the steps taken the more useful it will be for us to locate the problem.

If you used `pip install mu-editor`, then it should have retrieved all the dependencies from PyPI, independently from your distro package manager.

tim1mw, 2019-07-11:
@tmontes I get this from the test:

Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from PyQt5.QtCore import QTimer, Qt
ModuleNotFoundError: No module named 'PyQt5.QtCore'

So this suggests this is caused by the change to the build defaults in sip mentioned in my first post, with a knock on effect to the distro packages.. It isn't just Mageia 7 and Mu there's quite a lot of reports around the web from other python packages with the same problem after people upgraded to a distro version with the latest sip and PyQT5 packages. What worries me is that manually rebuilding sip and PyQt5 from source with the correct build options still doesn't generate the missing file. So while this isn't necessarily Mu's problem any help and thoughts from here are appreciated. I'll certainly see if I can feed back a bug report to Mageia, though it seems the problem may be upstream if the issue is that the new build option to generate this file doesn't actually work.

@carlosperate , I uninstalled the version I had and then did a user install in an entirely new user account to try and keep things clean. I might have time to set up a VM at the weekend and do an even cleaner version, but no promises on that.

[tim@flossie ~]$ pip3 install mu-editor --user
Collecting mu-editor
  Using cached https://files.pythonhosted.org/packages/fa/08/833e6ca83c8540733ca10b912cb21ad8d3b2b278567d34ae7651871f391a/mu_editor-1.0.2-py3-none-any.whl
Requirement already satisfied: semver>=2.8.0 in /usr/local/lib/python3.7/site-packages (from mu-editor) (2.8.1)
Requirement already satisfied: nudatus>=0.0.3 in /usr/local/lib/python3.7/site-packages (from mu-editor) (0.0.3)
Requirement already satisfied: gpiozero>=1.4.1 in /usr/local/lib/python3.7/site-packages (from mu-editor) (1.5.1)
Requirement already satisfied: guizero>=0.5.2 in /usr/local/lib/python3.7/site-packages (from mu-editor) (0.6.4)
Requirement already satisfied: pycodestyle==2.4.0 in /usr/local/lib/python3.7/site-packages (from mu-editor) (2.4.0)
Requirement already satisfied: pigpio>=1.40.post1 in /usr/local/lib/python3.7/site-packages (from mu-editor) (1.42)
Requirement already satisfied: appdirs>=1.4.3 in /usr/local/lib/python3.7/site-packages (from mu-editor) (1.4.3)
Requirement already satisfied: requests>=2.19.1 in /usr/lib/python3.7/site-packages (from mu-editor) (2.21.0)
Requirement already satisfied: qtconsole==4.3.1 in /usr/local/lib/python3.7/site-packages (from mu-editor) (4.3.1)
Requirement already satisfied: pyserial==3.4 in /usr/local/lib/python3.7/site-packages (from mu-editor) (3.4)
Requirement already satisfied: pgzero==1.2 in /usr/local/lib/python3.7/site-packages (from mu-editor) (1.2)
Requirement already satisfied: pyflakes==2.0.0 in /usr/local/lib/python3.7/site-packages (from mu-editor) (2.0.0)
Requirement already satisfied: qscintilla==2.10.8 in /usr/local/lib64/python3.7/site-packages (from mu-editor) (2.10.8)
Requirement already satisfied: Pillow>=5.2.0 in /usr/lib64/python3.7/site-packages (from mu-editor) (5.4.1)
Requirement already satisfied: PyQtChart==5.11.3 in /usr/local/lib64/python3.7/site-packages (from mu-editor) (5.11.3)
Requirement already satisfied: pyqt5==5.11.3 in /usr/local/lib64/python3.7/site-packages (from mu-editor) (5.11.3)
Requirement already satisfied: matplotlib==2.2.2 in /usr/lib64/python3.7/site-packages (from mu-editor) (2.2.2)
Requirement already satisfied: colorzero in /usr/local/lib/python3.7/site-packages (from gpiozero>=1.4.1->mu-editor) (1.1)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/lib/python3.7/site-packages (from requests>=2.19.1->mu-editor) (3.0.4)
Requirement already satisfied: idna<2.9,>=2.5 in /usr/lib/python3.7/site-packages (from requests>=2.19.1->mu-editor) (2.7)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /usr/lib/python3.7/site-packages (from requests>=2.19.1->mu-editor) (1.24.2)
Requirement already satisfied: ipykernel>=4.1 in /usr/local/lib/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (5.1.1)
Requirement already satisfied: jupyter-client>=4.1 in /usr/local/lib/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (5.2.4)
Requirement already satisfied: ipython-genutils in /usr/local/lib/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (0.2.0)
Requirement already satisfied: pygments in /usr/local/lib64/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (2.4.2)
Requirement already satisfied: traitlets in /usr/local/lib/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (4.3.2)
Requirement already satisfied: jupyter-core in /usr/local/lib/python3.7/site-packages (from qtconsole==4.3.1->mu-editor) (4.5.0)
Requirement already satisfied: numpy in /usr/lib64/python3.7/site-packages (from pgzero==1.2->mu-editor) (1.16.2)
Requirement already satisfied: pygame<2.0,>=1.9.2 in /usr/local/lib64/python3.7/site-packages (from pgzero==1.2->mu-editor) (1.9.6)
Requirement already satisfied: PyQt5_sip<4.20,>=4.19.11 in /usr/lib64/python3.7/site-packages (from pyqt5==5.11.3->mu-editor) (4.19.17)
Requirement already satisfied: cycler>=0.10 in /usr/lib/python3.7/site-packages (from matplotlib==2.2.2->mu-editor) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/lib/python3.7/site-packages (from matplotlib==2.2.2->mu-editor) (2.2.0)
Requirement already satisfied: python-dateutil>=2.1 in /usr/lib/python3.7/site-packages (from matplotlib==2.2.2->mu-editor) (2.8.0)
[... cut ...]
Installing collected packages: mu-editor
Successfully installed mu-editor-1.0.2
[tim@flossie ~]$ mu
mu-editor           multiarch-dispatch  multiarch-platform  multiblend.flt      multinit            munch
[tim@flossie ~]$ mu-editor 
Traceback (most recent call last):
  File "/home/tim/.local/bin/mu-editor", line 6, in <module>
    from mu.app import run
  File "/home/tim/.local/lib/python3.7/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ModuleNotFoundError: No module named 'PyQt5.sip'

tmontes, 2019-07-13:
> @tmontes I get this from the test:
> 
> Traceback (most recent call last):
> File "test.py", line 1, in 
> from PyQt5.QtCore import QTimer, Qt
> ModuleNotFoundError: No module named 'PyQt5.QtCore'

Interesting, that's different from the exception you originally observed: `ModuleNotFoundError: No module named 'PyQt5.sip'`. I suppose something in the environment might have changed since then, can you identify what?

> So this suggests this is caused by the change to the build defaults in sip mentioned in my first post, with a knock on effect to the distro packages.. It isn't just Mageia 7 and Mu there's quite a [...]

I understand your point of view. My linux testing tends to be very light, but as of the most recent stable/long-term-support versions, and in my observation, Mu happily installs and runs on Debian, Ubuntu, and Raspbian. Are you wondering whether these distributions will "bring us" troubles too, in the future?

Thanks! :)

tim1mw, 2019-07-13:
@tmontes basically yes, this is what I'm wondering.

For the distos that work for you which version of PyQT5 do they have installed by default? The problematic change to the build script was introduced in v5.11.

Some of the posts I'm seeing suggest that removing the distro supplied version of PyQT5 and the letting pip install it instead may help, but that hasn't worked for me so far (and has knock on issues because of system dependencies). However, this isn't very user friendly, ideally it would be nice to see a Mu appimage for Linux that avoids these issues.

I'll give this a go with a virtualenv next when I have a moment, perhaps that will solve the problem.

This is particularly frustrating since we are doing a Micro:Bit/Mu session at the local Linux user group in a few days time. Right now, I'm only able to run Mu on windows.

tmontes, 2019-07-14:
> For the distos that work for you which version of PyQT5 do they have installed by default? The problematic change to the build script was introduced in v5.11.
>
> Some of the posts I'm seeing suggest [...]

My experience is quite the opposite, actually, and very seamless (but not 100% beginner friendly!). I just checked an Ubuntu 18.04 VM I often use:
* Distribution supplied Python 3 installed.
* No distribution Qt packages installed.
* Installing Mu from source / PyPI within a virtual environment brings in the required PyQt5 packages as wheels, including the necessary already compiled Qt C++ libraries.
* This won't work on 32 bit systems because, AFAICT, PyQt5 isn't distributing 32 bit Linux wheels.

I wonder if something like this would work for you under Mageia (which I've never tried):
```
$ python3 -m venv mu-venv                 # Create a virtual environment for Mu
$ source mu-venv/scripts/activate        # activate it
(mu-venv) $ pip install mu-editor         # Or pip install <path-to-mu-repository-root>
```

tim1mw, 2019-07-14:
Taking out QT as a whole kills the entire KDE desktop, not really an option for me as a KDE user. It also knocks out VLC, Libreoffice, VirtualBox and a load of other common apps, so even people on a no-QT desktop would have issues doing that. The HP print manager tool also has a dependency on PyQT5 so pulling just that isn't really an option if you have an HP printer.

However, I gave this a go with the distro QT installed, the procedure needed a slight change to account for differing file paths:

python3 -m venv mu-venv
source mu-venv/bin/activate
pip3 install mu-editor

And it seemed to work, so that's good news, hopefully anybody else who encounters this problem will find this issue and use that work around. Thanks for the advice!

tmontes, 2019-07-14:
I really didn't mean "go ahead an uninstall Qt". Instead I meant "regardless of system-wide Qt, pip installing Mu will bring in an independent Qt implementation" (except on the Raspberry Pi where the system Qt is accounted for).

Maybe we can improve the installation guide here -- https://codewith.mu/en/howto/1.0/install_with_python. What is your feeling on that? Or maybe you didn't know about it, and we need to find better ways of helping people find it. :)

PS: It's good that it worked for you! :)

tmontes, 2019-10-03:
Hi @tim1mw, I suppose after having installed Mu per the linked guide in my previous message things have been running smoothly for you. If that is the case, could you please close this issue? If not, please let us know what is failing or should be improved, in your perspective. Thanks.

tim1mw, 2019-10-04:
@tmontes  Sorry I've not updated this further, I've had a very intense few months at work. The virtual environment has done the trick as mentioned in my earlier message. However that doesn't really feel like a long term solution. It may still be that this is really a packaging problem, but it seems to be one that may become more common. I think this can close for now, if I find out more I may post again or open a new issue.

closed 2019-10-04 by tim1mw

tmontes, 2019-10-05:
@tim1mw, thanks for sharing your feedback.

I do agree that installing this way (virtual env + pip install + etc) is not "the solution" for Mu's target audience of beginners; it is very important, though, and mostly directed at Mu contributors / developers. And it *does work* (nearly everywhere, by the way), as you can attest. :)

Software distribution is more complex than many would wish for. In an ideal world, Mu would be available for single click installation on each platform's native "application catalog" -- that's something that would serve beginners well. While the Windows and macOS targets are somehow confined, the Linux universe is much richer and having Mu in each and every distribution's "native" software repositories will never happen -- there has been some work in having it integrated into the Debian and Fedora repos, but I'm not 100% up to speed on that status / progress.

Having said all of this, if you or someone else is willing to contribute (and maybe figure out how, along the way!) to having Mu integrated into Mageia's repositories, I'll be glad to walk you through the existing Mu's packaging process. :)

Again thanks.

tim1mw, 2019-10-07:
@tmontes  I believe the work on getting Mu into Debian is finished, this was being done by a friend of mine, he gave a talk on the process to our local LUG a few months ago. As for packaging for Mageia, very very maybe, the work needed to do the Debian build was considerable, I think it took the best part of a year (admittedly fitted in between other commitments) because of the missing and very complex dependencies. That was partly caused by the Debian requirement for everything to be built from source, so the he had to sort out an entire tool chain to build the micropython interpreter binary. Mageia may not be as complex since I suspect they won't insist on that.

tmontes, 2019-10-07:
@tim1mw, you are correct: Mu 1.0.2 is available for GUI based installation on both Debian 10 and Fedora 30, at least. The package is named `mu-editor` @Debian, and just `mu` @Fedora, much like the notes in #543. Again, thanks for your details. :)

meatnordrink, 2022-01-02:
Coming to this two years later, and hitting the same issue (and others; Mu installation via pip apparently doesn't work on distros based on Ubuntu 21.10 (where Python is upgraded to 3.9; doesn't work at all on KDE), I think it would be really helpful to mention that Mu is available via `apt` in these instructions: 

https://codewith.mu/en/howto/1.0/install_with_python

As it is, I tried for two hours on three different computers, all with no success. I realize I could have made it work with `virtualenv` or anaconda, but, as mentioned above, that seems inappropriate for something meant for beginners (and I'm trying to teach a class to HS students using this). The `apt` installation worked like a charm, once I found it; but it's weird that it's not mentioned anywhere on that page. It would have saved me a lot of time and frustration.

dsssssssss9, 2022-01-02:
Hi

Another Mu user here..

What version did you install via apt?
On my Ubuntu system it installs  an old version 1.0.2
The current version is 1.1.0 beta7
I have had success using the new ( at least to me ) pipx installer...

https://github.com/mu-editor/mu/issues/1963#issuecomment-1002690292

meatnordrink, 2022-01-04:
Oh, that's probably true; I didn't really care if I installed the newest version.