mu-editor/mu #876: 引脚中断不断触发

tutuwin, 2019-07-03:
设置一个引脚位下降沿中断，触发一次后就会不停的触发，无法停止，但是引脚电压测量已经恢复高电平，使用的引脚位io16。

dhalbert, 2019-07-03:
Title: _Pin interrupts are constantly triggered_
Text: _Set a pin bit falling edge interrupt, it will trigger continuously after one trigger, can't stop, but the pin voltage measurement has returned to high level, using pin io16._

This doesn't appear to be related to the Mu editor. @tutuwin: I think you posted in the wrong place.

ntoll, 2019-07-03:
@tutuwin 我希望你不介意我用谷歌翻譯回复。 我認為問題不在於“Mu”，而在於您正在使用的設備或Python版本。 請嘗試在這些項目的網站上尋求支持。

向來自英國的中國朋友致以問候! :cn: :handshake: :gb: :cow: :slightly_smiling_face:

ntoll, 2019-07-03:
@dhalbert thank you..! I had no idea you spoke Chinese... :stuck_out_tongue:

closed 2019-07-03 by ntoll

dhalbert, 2019-07-03:
Neither did I :)