mu-editor/mu #880: Toggle comments de-comments embedded comments causing syntax errors

inventhouse, 2019-07-05:
In every editor I've ever used, commenting out-and-back-in a line restores the line to exactly the way it was; Mu's "toggle comments" (Cmd-K) strips comment markers beyond the first, de-commenting previously existing comments which causes syntax errors.

Starting with this line:
`print("testing")  # test that this is called`

Cmd-K comments it out as expected:
`# print("testing")  # test that this is called`

But Cmd-K again to put it back yields this:
`print("testing")  test that this is called`

It should restore the original, leaving the trailing comment intact.

I'm running Mu 1.0.2 on macOS 10.14.5 (18F132)

### Mu log:
-----------------

Starting Mu 1.0.2
2019-07-05 08:19:31,618 - root:112(run) INFO: uname_result(system='Darwin', node='Inara.local', release='18.6.0', version='Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-07-05 08:19:31,618 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-07-05 08:19:31,618 - root:114(run) INFO: Language code: en_US
2019-07-05 08:19:31,952 - mu.logic:543(__init__) INFO: Setting up editor.
2019-07-05 08:19:31,952 - mu.logic:562(__init__) INFO: Settings path: /Users/bjh/Library/Application Support/mu/settings.json
2019-07-05 08:19:31,952 - mu.logic:563(__init__) INFO: Session path: /Users/bjh/Library/Application Support/mu/session.json
2019-07-05 08:19:31,952 - mu.logic:564(__init__) INFO: Log directory: /Users/bjh/Library/Logs/mu
2019-07-05 08:19:31,952 - mu.logic:565(__init__) INFO: Data directory: /Users/bjh/Library/Application Support/mu
2019-07-05 08:19:31,963 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-07-05 08:19:31,964 - mu.logic:589(setup) DEBUG: BJH: Not creating junk dirs & assets (issue #872)
2019-07-05 08:19:32,120 - mu.logic:1151(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-07-05 08:19:32,125 - mu.logic:628(restore_session) INFO: Restoring session from: /Users/bjh/Library/Application Support/mu/session.json
2019-07-05 08:19:32,125 - mu.logic:629(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-07-05 08:19:32,125 - mu.logic:650(restore_session) INFO: Loaded files.
2019-07-05 08:19:32,125 - mu.logic:654(restore_session) INFO: User defined environment variables: []
2019-07-05 08:19:32,125 - mu.logic:658(restore_session) INFO: Minify scripts on micro:bit? False
2019-07-05 08:19:32,203 - mu.logic:681(restore_session) INFO: Starting with blank file.
2019-07-05 08:19:32,237 - mu.logic:1151(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-07-05 08:19:54,118 - mu.interface.editor:520(toggle_comments) INFO: Toggling line 1
2019-07-05 08:19:55,270 - mu.interface.editor:496(toggle_comments) INFO: Toggling comments
2019-07-05 08:19:59,531 - mu.logic:1078(show_admin) INFO: Showing logs from /Users/bjh/Library/Logs/mu/mu.log

ntoll, 2019-07-05:
You're right..! Thank you for pointing this out. Obviously, it should only remove the first matching `#`. Care to submit a pull request..? ;-)

inventhouse, 2019-07-05:
I'll look at the code in detail later today 👍

ntoll, 2019-07-05:
This is the function you should be changing... (to save you 20mins of wading through code):

https://github.com/mu-editor/mu/blob/master/mu/interface/editor.py#L537

inventhouse, 2019-07-05:
I often use multiple comment markers to label "sections" of code, like this:
`###  Config  ###`

 I'd like it to leave those alone; what would you think about making it ignore lines starting with `##`?

inventhouse, 2019-07-05:
> I often use multiple comment markers to label "sections" of code, like this:
> `### Config ###`
> 
> I'd like it to leave those alone; what would you think about making it ignore lines starting with `##`?

So I went ahead and wrote it like this; let me know if you really hate the idea in my PR 😉

tmontes, 2019-07-05:
Possible idea, striving for simple and consistent.

Given a set of selected lines, classify each one into:
* Non comment
* Standard comment: zero/more leading whitespace, the `#`, one ` ` (space), followed by anything.
* Non standard comment: zero/more leading whitespace, the `#`, not followed by a ` ` (space).

When all selected lines are "Standard comments" proceed to uncommenting them:
* Remove the `# ` after the optional leading whitespace.

Otherwise, proceed to commenting them:
* Add a `# ` after the leading whitespace.

At first sight this would cope with your use case by toggling it between...
```
### Config ###
```
...and...
```
# ### Config
```

It would also support all other kinds of funny (non-standard!) commenting approaches, mostly leaving them untouched, even if "double commented" at times.

Could this be useful?

inventhouse, 2019-07-05:
@tmontes Consistent, inteligible, behavior is _really_ important, for sure!  One use-case I rather like with the "toggle" behavior is the "comment swap"; if I have two lines like this:
`MODE = "normal"`
`# MODE = "debug"`
I can select them and hit Cmd-K and it comments the first and uncomments the second in one go.

What I have implemented is as follows:
- Lines containing only whitespace or the first non-whitespace characters are `##` are left untouched
- Other lines where the first non-whitespace characters are `#` or `# ` the first comment mark is removed
- All other lines have `# ` added to the beginning

I feel that this pretty straightforward and not such a significant behavior change from how it has been working.

inventhouse, 2019-07-06:
Ok, PR is ready for review / merge 😊

tmontes, 2019-07-06:
@inventhouse, I never thought about the "toggle between two lines of code" approach. That's very cool and bound to be useful. Nice! :)

inventhouse, 2019-07-06:
@tmontes Not just two lines, you can toggle whole implementations back-and-forth 😄

carlosperate, 2019-07-08:
> Lines containing only whitespace or the first non-whitespace characters are ## are left untouched

Do any other editors do this?
To be honest this feels a bit unintuitive to me. I would expect the shortcut to add and remove a single character comment character to all lines independently of its non-standard content.

I don't use this feature often, but when I do (like in VSCode) I normally expect it to be simple:
- If all lines start with a`#` (ignoring whitespaces before the comment character) remove a single `#` from all lines
- If not: add a single `#` to all the lines

Of course, Mu can always deviate from standard conventions, but there should be good reasons for Mu's target audience.

inventhouse, 2019-07-08:
@carlosperate I use it all the time in other editors, and they usually add comment markers if any of the lines are not comments, similar to what @tmontes described above; I've not seen them do the toggle / comment-swap behavior in the first place.

As for the `##` trick, it's something I noticed in testing - the "toggle" was eating comments I didn't want it to, and it was easy to add, so I did; if the consensus during review is that it's too weird, it's easy to remove.

I really like the added control it gives, though, to have comments that _stay_ comments; is there a better way to offer that control?

tjguk, 2019-08-04:
I've had a look at the changes, both the code/test changes and the effect on the editor. The code change looks fine, and thanks for doing the tests (those are always a pain to add later!)

I don't think there's a perfect answer here. Having slightly misread your description, I tried it against a pattern I often use which is multiline comments with a *single* leading hash:

```
#
# This is a comment
#
```

Only to have my top and bottom comment markers swallowed when attempting to toggle! However toggling a comment block is not something I'd deliberately do, and I think your proposal of special-casing double markers is a sensible compromise. 

I do have a concern -- no matter what we do here -- that the rules behind the behaviour are somewhat hidden. Eg we don't really have any way of communicating to users that a double-marker comment block will be excluded from a comment toggle. And it's probably not something most people will think about ("what if I doubled up the comment markers...?")

Nonetheless, even if they don't know, they can still generate the same comment effect albeit less efficiently, ie by selecting each block and sidestepping the comment blocks. And the comment-toggling participates in the Undo-Redo buffer (I checked!). So even if they do make a mistake by over-selecting a block to comment, Undo is there for you.

I'm just sufficiently concerned about the issues above that I'm going to wait until Tuesday evening before doing the merge. And that's because I'm seeing @ntoll on Tuesday lunchtime (about something else) so, even if he doesn't chip in before then, I'll ask him then to make a call on this.

Thanks for the PR @inventhouse (and for the ping -- stuff just gets missed sometimes).

inventhouse, 2019-08-06:
@tjguk Thanks for having a look!  It would also be trivial to drop the double-comment trick, either way I'm pleased with my fix.

tjguk, 2019-08-10:
Fixed by #882 -- thanks very much @inventhouse for reporting and for the PR

closed 2019-08-10 by tjguk