mu-editor/mu #881: 1.1.0a2 | PGZero | Crash opening assets (images, sounds, etc) with fresh editor tab.

tmontes, 2019-07-05:
## Observed behaviour

When in Pygame Zero mode, with a blank/fresh/untitled editor tab, clicking any of the Images, Fonts, Sounds, or Music buttons crashes Mu.

Traceback is:
```
(mu-venv) tiago.montes (~/Work/Python/github.com/mu): mu-editor 
Traceback (most recent call last):
  File "/Users/tiago.montes/Work/Python/github.com/mu/mu/modes/pygamezero.py", line 192, in show_music
    os.path.dirname(self.view.current_tab.path), 'music')
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/posixpath.py", line 156, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
(mu-venv) tiago.montes (~/Work/Python/github.com/mu): mu-editor 
Traceback (most recent call last):
  File "/Users/tiago.montes/Work/Python/github.com/mu/mu/modes/pygamezero.py", line 159, in show_images
    os.path.dirname(self.view.current_tab.path), 'images')
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/posixpath.py", line 156, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

## Expected behaviour

Not really sure, given that the asset paths are relative to the current editor tab's file path. Maybe these buttons could be disabled if the editor tab hasn't been saved yet?

Prompting thought and discussion, here... :)

## Steps to reproduce

* Launch Mu 1.1.0.alpha.2.
* Change mode to "Pygame Zero".
* Create a new editor tab.
* Click the "Images" button.
* Mu crashes.

## Platform notes

* Tested on macOS 10.12.6.
* Running from source, with the `mu-editor` command, within a Python 3.6.8 virtual environment, after a `pip install -e .[dev]`.
* Given the nature of the traceback I expect such failure to occur in all platforms both in packaged and non-packaged variations of Mu.

amos1969, 2019-07-05:
Just tried on Linux Mint using Python3.7 in a Virtual Environment created using a previous version's requirements.txt with pip. 

I get exactly the same behaviour:

`(myvenv) amos@amos-ThinkPad-T430 ~/mu $ python mu/run.py    
Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use:    
   QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors.    
   QT_SCREEN_SCALE_FACTORS to set per-screen factors.    
   QT_SCALE_FACTOR to set the application global scale factor.    
Traceback (most recent call last):    
  File "/home/amos/mu/mu/mu/modes/pygamezero.py", line 159, in show_images    
    os.path.dirname(self.view.current_tab.path), 'images')    
  File "/usr/local/lib/python3.7/posixpath.py", line 156, in dirname    
    p = os.fspath(p)    
TypeError: expected str, bytes or os.PathLike object, not NoneType    
`    

Naming the file before clicking on images does not trigger this bug.

tmontes, 2019-07-05:
Thanks for the confirmation @amos1969.

I have just reproduced the crash on Windows 10, after installing the 64 bit packaged version.

tjguk, 2019-07-05:
That'll almost certainly be down to the change I made in #862 which @ntoll just merged. I'll try to have a look this evening. If someone gets there first, feel free..

tmontes, 2019-07-06:
Other than the somewhat confusing "which 1.1.0.alpha2 is which 1.1.0.alpha2" (!!!), I confirm the just merged PR #883 and re-released 1.1.0.alpha2 Windows 64 bit version at https://codewith.mu/ no longer crash, opening the respective assets folder under "~/mu_code".

Closing.

Thanks @tjguk and @ntoll. :)

closed 2019-07-06 by tmontes