mu-editor/mu #886: help() should show the top of the help documentation

illume, 2019-07-10:
* What you were trying to do,
help(turtle)

* What you expected to happen,

To see the top of the help.

* What actually happened,

A very long time to print the help text in the non python repl. In the ipython repl it showed the bottom of the text.

* Why this difference is problematic (it may not be a bug!),

Instead of seeing the top of the help you see the bottom of it. Because you need to scroll to begin reading the help.

ZanderBrown, 2019-07-20:
So your creating a script consisting of

```py
import turtle

help(turtle)
```

... and running it?

illume, 2019-07-21:
> So your creating a script consisting of
>
> import turtle
> help(turtle)
>
> ... and running it?

No. In the repl.

ZanderBrown, 2019-07-21:
> in the non python repl

> In the repl

I'm confused? Perhaps a screenshot?

illume, 2019-07-21:
Go into either of the repl in Mu (there are two different ones in Mu, one
using ipython).

import turtle.
help(turtle)

Now compare the behavior to a different python repl. The standard python in
a terminal(cmd prompt) shows the top of the help, and uses a pager (less).
Mu shows the bottom of the help.

Hope that helps. That’s all I got. Have a nice day.

dybber, 2020-09-27:
I will update the title of this issue to include the mention of a pager, I guess that will clear up some confusion. 

For anyone who hasn't tried it, running `help(turtle)` prints 8000 lines of documentation. Hardly useful as it is.

closed 2021-04-09 by carlosperate