mu-editor/mu #889: EP2019 BD | Webmode hard crash on macOS

tmontes, 2019-07-13:
### Context

* Observed during EP2019 Beginners' Day sessions.
* User running macOS 10.14.x with UI set to German.
* Mu 1.1.0.alpha2 (probably commit 4678e35c309849f6ec0e14422af34e00ba362e81) installed from the DMG image distributed at https://codewith.mu.

### Behaviour

* Mu launched by double clicking its icon in the Finder.
* Mu set to web mode.
* The it either crashed when (sorry, can't remember fully):
  * Starting the web server.
  * Having it respond to the first request (inclined to believe this was the actual case).

### More info

* I have collected the full log which Michael (the user) was kind enough to email me.
* Won't paste it in full here, because of its size (but keeping it, in case anyone wants to dig deeper).
* A few suspect lines show up repeatedly:

```
2019-07-09 11:04:33,196 - mu.modes.web:182(stop_server) DEBUG: Stopping Flask app.
2019-07-09 11:04:33,196 - root:96(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 122, in run_toggle
self.stop_server()
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 186, in stop_server
os.kill(pid, signal.SIGINT)
KeyboardInterrupt
2019-07-09 11:04:37,056 - root:113(run) INFO:
...
2019-07-09 11:05:16,080 - mu.modes.web:175(start_server) DEBUG: Starting Flask app.
2019-07-09 11:05:52,312 - mu.modes.web:182(stop_server) DEBUG: Stopping Flask app.
2019-07-09 11:05:52,312 - root:96(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 122, in run_toggle
self.stop_server()
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 186, in stop_server
os.kill(pid, signal.SIGINT)
KeyboardInterrupt
2019-07-09 11:05:56,421 - root:113(run) INFO:
...
2019-07-09 11:06:25,540 - mu.modes.web:175(start_server) DEBUG: Starting Flask app.
2019-07-09 11:06:37,240 - mu.modes.web:182(stop_server) DEBUG: Stopping Flask app.
2019-07-09 11:06:37,240 - root:96(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 122, in run_toggle
self.stop_server()
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 186, in stop_server
os.kill(pid, signal.SIGINT)
KeyboardInterrupt
2019-07-09 11:06:47,712 - root:113(run) INFO:
...
2019-07-09 11:07:33,636 - mu.modes.web:175(start_server) DEBUG: Starting Flask app.
2019-07-09 11:08:15,523 - mu.modes.web:182(stop_server) DEBUG: Stopping Flask app.
2019-07-09 11:08:15,527 - root:96(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 122, in run_toggle
self.stop_server()
File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/web.py", line 186, in stop_server
os.kill(pid, signal.SIGINT)
KeyboardInterrupt
2019-07-09 11:09:02,824 - root:113(run) INFO:
```

I'm not familiar with the code, and haven't looked at it but find it interesting that the apparent crash (at least we get no more entries after these lines other than the first message Mu logs when starting up) is after an `os.kill(pid, ...)`: is Mu committing suicide? :)

### More

* I tried playing around with my system's configuration, including changing UI languages, building a "packaged macOS app", running "from source", etc., but completely failed to reproduce the crash.
* I'll try to see if Michael has a github user account and if we wants to follow up on the issue and help us identify the underlying cause.

Teagum, 2019-07-16:
The error happens after the following steps:

1. start Mu via Finder
2. set Mu to web mode
3. click on Start
4. (Save dialog pops up ) enter file name, click "OK"
5. RuntimeError
6. click "Stop" -> Mu crashes

### Full traceback:

> Traceback (most recent call last):
>   File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 193, in _run_module_as_main
>   File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 85, in _run_code
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/__main__.py", line 14, in <module>
>     main(as_module=True)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/cli.py", line 894, in main
>     cli.main(args=args, prog_name=name)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/cli.py", line 557, in main
>     return super(FlaskGroup, self).main(*args, **kwargs)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 696, in main
>     _verify_python3_env()
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/_unicodefun.py", line 124, in _verify_python3_env
>     ' mitigation steps.' + extra
> RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/en/7.x/python3/ for 
> 
> ---------- FINISHED ----------
> exit code: 1 status: 0mitigation steps.

### My settings

- macOS 10.14.3

[image]

### Observations
There is **no error** if I I change the config so that German is the primary language.

dybber, 2020-12-10:
Trying to reason myself towards the source of this error:
 - Web mode uses Flask
 - Flask depends on Click
 - Click doesn't work with ASCII locales
 - Some locales on Mac OS are still using ASCII

This means that the problem is not in Mu (per se), but is to be fixed in either Click or Mac OS. However, after reading through https://click.palletsprojects.com/en/7.x/python3/ it doesn't seem likely that Click will support ASCII locales.

However, we might be able to set a locale we know works, before launching the Flask server. For example, setting these environment variables:
```
LC_ALL=en_GB.UTF-8
LANG=en_GB.UTF-8
```

Which means the following might just be enough:
```diff
diff --git a/mu/modes/web.py b/mu/modes/web.py
index 69c3858..dc61523 100644
--- a/mu/modes/web.py
+++ b/mu/modes/web.py
@@ -170,6 +170,8 @@ class WebMode(BaseMode):
             envars = self.editor.envars
             envars.append(("FLASK_APP", os.path.basename(tab.path)))
             envars.append(("FLASK_ENV", "development"))
+            envars.append(("LC_ALL", "en_GB.UTF8"))
+            envars.append(("LANG", "en_GB.UTF8"))
             args = ["-m", "flask", "run"]
             cwd = os.path.dirname(tab.path)
             self.runner = self.view.add_python3_runner(
```

I'm don't know that much about Flask. Will it be a problem not to use the users own locale?

dybber, 2021-01-26:
Was fixed by #1174.

closed 2021-01-26 by dybber