mu-editor/mu #890: EP2019 BD | Webmode fails when file saved elsewhere

tmontes, 2019-07-13:
### Observed behaviour

* Jinja2 exception rendered in the browser.

### Expected behaviour

* The "hello" page should be displayed in the browser.

### Steps to reproduce

* Start Mu.
* Change to web mode.
* Create a new file and immediately save it under, say, `~/Desktop`.
* Click the Run button.
* Click the Browser button.

A ** jinja2.exceptions.TemplateNotFound** page is displayed.

### Notes

Verified both on macOS and Windows 10, running from source.

PS: Might be related to a very odd behaviour we observed with André at the Beginner's Day. I have asked him to run a few tests to help out diagnose his particular case and invited him to join the discussion here.

seantibor, 2019-09-28:
I think I just duplicated this issue with #923. Seems like it's because the template files are only copied over when the mode changes, not when the working directory path changes.

dybber, 2020-12-10:
I just created a file in my `mu_code` with the default content when creating a file in web-mode, and I get the same error. I don't have an `index.html` in the `mu_code/templates` directory for some reason.

ntoll, 2021-07-10:
PR with fix in #1695

PR #1695 (2021-07-19): Web mode checks existence of templates directory before starting local server
Fixes #890.

If there is no `templates` directory in the same location as the Python file containing the web app, then Flask will only ever respond with a non-beginner-friendly stack trace. This change checks for the `templates` directory in the expected location and displays a warning message with details of the problem so the user can try to fix it.
closed 2021-07-19 by ntoll