mu-editor/mu #891: EP2019 | Improved repository layout

tmontes, 2019-07-13:
### Context

This is an "umbrella" issue that follows from a brief discussion I had with @ntoll at EuroPython yesterday, about many Mu related, somewhat high-level, project orientation topics.

One of those is about improving the repository layout such that it becomes, in a word, more readable. That should simplify everybody's lives, especially those who are new to the Mu's codebase.

I'll list the few ideas we discussed as individual "checkboxes" to allow for discussion. From there, we may either relate each such task to a dedicated issue, or have it "closed" by means of one or more PRs.

### Ideas

* [ ] Move to an `src/` based layout in such a way that what is now the `mu/` directory/package at the root, is moved to `src/mu/`. There are multiple benefits to this, in particular related to test and packaging reliability (see, for example, the first part of the article at https://hynek.me/articles/testing-packaging/). There might be a tradeoff in that tracking history changes with git becomes less easy, but that is something that, in time, will no longer be an issue.

* [ ] Re-arranging files in the root:
  * [ ] Move `ROADMAP.rst` to docs?
  * [ ] Remove `force-rechecks.txt`?
  * [ ] Throw away `run.py`, there's no longer a good motive to keep it, AFAICT. As a matter of fact, it leads to confusion in that there are now at least two ways to run Mu, when developing.
  * [ ] Review `todo.txt` and create issues if needed. Then drop the file.
  * [ ] Move `win_installer.py` to the `package/` directory.

* Re-arrange the files in the `package/` directory.
  * [ ] Create a `mac`, `lin`, `win` sub directory (or more, on per linux dist?).
  * [ ] Move the files to the "proper" place.
  * [ ] `package/README.rst` looks outdated: integrate what's relevant in the docs and delete it.
  * [ ] Remove the `__init__.py` from there: it is not a package.
  * [ ] Move the `debian` directory in the root here.

* [ ] Not sure what the `conf/` directory is, but seems packaging related. @ntoll suggested pinging Kushal Das, any pointers? :)

* General code layout:
  * I feel that some modules could be broken down into smaller modules. Maybe that could make reading/understanding the code a bit easier/less frightening.
  * [ ] `mu/logic.py`: Editor class is big, but there are many other bits and pieces.
  * [ ] `mu/modes/base.py`: move MicroPythonMode class to its own module?
  * [ ] `mu/interface/dialogs.py`: break apart with one module per dialog?
  * [ ] `mu/interface/main.py`: break apart with one module per component?
  * [ ] `mu/interface/panes.py`: break apart with one module per pane?
  * [ ] `mu/interface/themes.py`: break apart with one module per theme?

* [ ] Could we delete most (all?) branches from the repository? It certainly increases confusion for newcomers, wouldn't you agree?

### Parting Thoughts

This is it, AFAICT. Would love to hear feedback before getting to some of this action. :)

Thanks!

PS: Will create more issues to address other topics I reviewed with @ntoll.

tmontes, 2019-07-13:
@kushaldas, after a short discussion with @ntoll, about Mu's repository layout, he suggested I ping you about the files you helped create under the `./conf/` directory. I understand that that was a looooong time ago, but, if possible, could you help understand the following:
* Are these files related to Fedora packaging?
* If so, could they be moved to, say, `./package/linux/fedora/`, for example?

Thanks a lot in advance for any input you may provide.

One file there, `./conf/mu.appdata.xml`, has been created by @ZanderBrown: could you please chime in with some feedback too. Thanks a lot.

ZanderBrown, 2019-07-16:
@tmontes `./conf/mu.appdata.xml` is our [AppStream](https://www.freedesktop.org/software/appstream/docs/) metadata, an important part of modern Linux applications

On GNOME this is used by gnome-software to provide a detailed description of Mu, KDE store-thing does similar

[image]

(sorry I've not been around so much lately, been distracted by other things which shall not be named...)

tmontes, 2019-07-16:
@ZanderBrown, thanks for the info.

Could one say, thus, that it is somehow "packaging" related? If so, would there be any negative implications in moving it under, say, `./package/linux/appstream/`, or some more explicitly named directory other than `./conf`? :)

ZanderBrown, 2019-07-16:
@tmontes well given the other proposals downstream packing is already broken so we may as well

Of course the traditional location would be /data

Not sure the benefit in giving a single file it's own directory?

tmontes, 2019-07-17:
> @tmontes well given the other proposals downstream packing is already broken so we may as well

@ZanderBrown, I'm not sure I understand your words perfectly. What do you mean by "other proposals" and "already broken"? Are you referring to the general proposal of layout change I'm describing here? Something else?

My purpose is to try and make it easier for someone with little/no prior knowledge of Mu and its repo layout to grasp what's going on with as little effort as possible (and I think that some moving around/renaming helps with that). But that should not come at a (big) cost to others, in particular others that work on packaging Mu (itself quite a challenge, given the wide variety of platforms Mu wants to be distributed over). I'm trying to find out a good balance, here. :)

> Of course the traditional location would be /data
> Not sure the benefit in giving a single file it's own directory?

In my mind, having all packaging-related files (non-source, non-docs, non-tests) under a the already existing `./package/`, and going a bit further from there, with one subdirectory per "platform" / "target" would make understanding what-is-what a bit better: thus `./package/windows`, `./package/macos`, and I suppose, `./package/linux/debian`, `./package/linux/fedora`, etc. Eventually, if the `mu.appdata.xml` file is common to multiple Linux packaging solutions, it could be moved to `./package/linux/`.

I myself don't find it offending at all if a given directory contains a single file if that directory's name/path helps understanding what that file might be about. But I'm happy to hear other perspectives. :)

Thanks for your input.

ZanderBrown, 2019-07-17:
Sorry if i was unclear

> Are you referring to the general proposal of layout change I'm describing here?

This, but with new modes + deps packagers where going to have work to do anyway

> subdirectory per "platform" / "target"

Makes sense

> /package/linux/fedora

Do we currently have any fedora specific files?

> /package/linux/debian

I assume this means moving /debian? Unfortunately that gets back to "should not come at a (big) cost to others" in that moving /debian doesn't work

But the alternative is ditch it, ideally this would be maintained somestream and considering the last changelog entry is `1.0.0~beta8` from 2017 (was it really that long ago?) yet [buster](https://packages.debian.org/buster/mu-editor) seems to have entry for `1.0.2+dfsg-3` it seems this is already the case

Similarly Raspbian history seems to follow that of debian although has spit to [`1.0.2+dfsg~uflash-1~bpo9`](http://archive.raspberrypi.org/debian/pool/main/m/mu-editor/) for whatever reason

tmontes, 2019-07-17:
*Preliminary note: while I'm mostly at home with Python packaging, I know absolutely zero about the packaging requirements / complexities / processes / tools regarding Linux distribution specific packaging. I'm just trying to figure out a way of organising the repository in such a way that what-is-what and where-is-what becomes as clear and as obvious as possible: giving good names to directories and moving a few files around can contribute to that, hopefully!* :)

> > Are you referring to the general proposal of layout change I'm describing here?
> 
> This, but with new modes + deps packagers where going to have work to do anyway

Still not 100% sure, but assuming that moving the current `./conf/mu.appdata.xml` to `./package/linux` will imply some work by someone (wondering if that someone is you and, regardless of that, if such work would be acceptable for the sake of having a more organised repo).

PS: Read the AppStream docs you linked to (thanks!) and couldn't identify any specifics on that.

> > subdirectory per "platform" / "target"
> 
> Makes sense

Glad we agree.

> > /package/linux/fedora
> 
> Do we currently have any fedora specific files?

All I know is what @ntoll told me when I asked him "what are these files under `./conf/`": he pointed me to Fedora and @kushaldas and I found you via `git blame`. After reading the AppStream docs, I now believe that at least `mu.codewith.editor.desktop` is not Fedora-specific and thus, a good candidate to be side-by-side with `mu.appdata.xml` under `./package/linux`. Or am I understanding something wrong?

As for the other two files there, an PNG icon and what looks like a `udev` rules file, maybe they're Fedora specific, but I don't know yet.

> > /package/linux/debian
> 
> I assume this means moving /debian? Unfortunately that gets back to "should not come at a (big) cost to others" in that moving /debian doesn't work

Oh, is that a fact? :( Does the Debian project require source repos to have a `./debian` directory? :(

> But the alternative is ditch it, ideally this would be maintained somestream and considering the last changelog entry is `1.0.0~beta8` from 2017 (was it really that long ago?) yet [...]

Let me see if I get your thesis right: Debian packagers have a clone of the repo with a branch that holds the equivalent of the current `./debian` here, which they maintain: is that the possibility you are establishing?

> Similarly Raspbian history seems to follow that of debian although has spit to [`1.0.2+dfsg~uflash-1~bpo9`](http://archive.raspberrypi.org/debian/pool/main/m/mu-editor/) for whatever reason

Hmmm. Would you have any pointers of who to contact to validate these hypothesis? :)

PS: Thanks a lot for sharing all these details!

ZanderBrown, 2019-07-17:
> wondering if that someone is you

No, not yet anyware

*Zander tries to look busy*

>  such work would be acceptable for the sake of having a more organised repo

Yes :-)

> mu.codewith.editor.desktop is not Fedora-specific

It's the launcher as used by everything following [this spec](https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html)

> a PNG icon

The icon used by the .desktop, ideally this would be an SVG however

> udev rules file, maybe they're Fedora specific

I imagine debian needs them as well? Not a udev person though

> Oh, is that a fact?

Yep, tried to find a definitive source that puts this in black 'n white but alas I failed

> Debian packagers have a clone of the repo

Yes/No, Debian have there own repo with the packaging information that references this repo (or at least a tarball of it)

You can see Fedora's such repo on Pagure [here](https://src.fedoraproject.org/rpms/mu/tree/master) (note RPM packaging uses a single `.spec` which can be located anywhere rather than a whole directory....)

... and after checking through half of [packages.debian.org](https://packages.debian.org/) we find the [repo on Salsa](https://salsa.debian.org/python-team/applications/mu-editor) for Debian (notice the git history is quite different to ours)

For Raspbian you would have to as the Foundation as they are maintaing patches on top of Rasbian itself but as you would expect it's based on Debians repo (as evident by the changelog)

> pointers of who to contact to validate these hypothesis

See above :-)

ZanderBrown, 2019-07-17:
(a good tool for finding info about a package is [repology](https://repology.org/project/mu-editor/packages), might be of some interest to you)

tmontes, 2019-07-17:
@ZanderBrown, thanks a lot: that's very useful information. I'll try and reach out to distribution specific packagers and see moving a few files around and even throwing others away might be a good fit for them... Let's see how that goes.

tmontes, 2019-07-18:
FYI, I just emailed the Mu package maintainer for a bunch of Debian-like distributions (found in the link @ZanderBrown shared above):

> Dear Knowledge Junkie,
> 
> 
> My name is Tiago, and I've been recently contributing to the Mu editor project: a beginner's friendly Python editor.
> 
> After a recent discussion with Nicholas Tollervey, the original author, I created an issue to publicly discuss a few changes to the current source repository layout which, in our opinion, would make life simpler to both current developers/contributors but also and especially to new ones.
> 
> The issue is linked here -- https://github.com/mu-editor/mu/issues/891 -- and covers various aspects. Some are related, I suspect, to downstream Linux distribution specific packaging of Mu.
> 
> I found your contact as the package maintainer for several Linux distributions at https://repology.org/project/mu-editor/packages and that's the motive of my contact.
> 
> If possible, could you chime in on the topics I identify there? We have some files spread through non-obvious directories (e.g.: /conf and /debian) which I would like to either move to more aptly named places, or remove, if no longer deemed necessary or useful. But I don't want to do it "at any cost", especially not at the cost of downstream packaging efforts which we wouldn't want to compromise. :)
> 
> If you do not have a GitHub account, I'm happy to keep this discussion going over email and "mirror it" there, if that's ok with you.
> 
> 
> Thanks a lot for your efforts in packaging Mu downstream and, in advance, for any feedback you may provide.
> 
> Kind regards,
> Tiago

Let's see how that goes. :)

ZanderBrown, 2019-07-18:
Nick is here: @knowledgejunkie , see #58 for related discussion

tmontes, 2019-07-18:
> Nick is here: @knowledgejunkie , see #58 for related discussion

Oh, I had no idea! Feeling somewhat silly. :) Thanks. That should simplify a more direct interaction.

tmontes, 2019-07-18:
Sharing a valuable Debian document which I plan on reading: [Debian Upstream Guide](https://wiki.debian.org/UpstreamGuide).

ZanderBrown, 2019-07-18:
Relevent extract:

> Some projects include a rough /debian directory among source files to ease bleeding-edge package compilation and installation on Debian (and derived) systems. While this is a good effort, it is better to leave it out of the final tarball as it can interfere with debian's own packaging effort. Keeping it only in your VCS repository is usually a much saner default if it lives in a specific packaging branch, which mimics what Debian package maintainers do using git-buildpackage. Though leaving the debian folder in the normal branch can also interfere if the package maintainer is using an upstream git packaging workflow (for example: git tag based git-buildpackage workflow).

tmontes, 2019-07-19:
From the extract (thanks!) it seems that downstream Debian packaging doesn't care much about whether the source repository contains or not the packaging related files: or, maybe they prefer the source repo not to include it at all (unless we want to support bleeding edge Debian packaging, for which I fail to see a big benefit given that a `pip install` from bleeding edge source does work on Debian). :)

Nevertheless, I received a friendly feedback email from @knowledgejunkie who was nice enough to help out with direct feedback in this issue.

My current perception is that moving `./debian/` to `./package/debian/` or removing it completely will be ok WRT Debian packaging. But let's wait and see. No hurries here! :)

knowledgejunkie, 2019-08-16:
Apologies for the delayed response...

With respect to "officially" packaging mu-editor for Debian, we throw away /debian completely when we ingest a new release. With mu-editor (and the associated "Debianisation" to generate a Debian Free Software Guidelines-compatible (DFSG) binary package) now available in Debian stable, Ubuntu 19.04 and many derivatives, I'm not sure how much use the existing upstream /debian/ packaging files will get from end-users. As @ZanderBrown mentioned, Debian tooling expects the "Debianisation" files to exist in /debian/, so moving them out of this path will cause breakage for end users wanting to build a deb using these files.

When ingesting a new upstream release of software, the Debian tooling can automatically exclude files and directories from the source tree if such files are deemed unnecessary (such as CI/CD configuration, or Mac/Windows packaging), are not freely licensed, or are not compatible with Debian Policy and/or DFSG for other reasons. Reorganising the upstream repository layout for e.g. packaging may simplify the number of items included in the exclusion list (which would need to be kept up to date), but does not really make a difference. Similarly, moving mu/ under src/ will require updating the exclusions list to use the new path.

The current list of excluded files can be seen at https://salsa.debian.org/python-team/applications/mu-editor/blob/debian/master/debian/copyright

We install the AppStream data, desktop file and icon from conf/. The location of these files in the upstream repo are not important - the important step is installing them into the correct system location when the package is installed. We'd just need to update references to their new location were they to move.

The Debian packaging repository is not a clone of this repo "with added sparkles", but rather a completely separate repository with each Mu release ingested as a tarball in a commit to the "clean" upstream branch. Packaging work happens in the master branch, into which the new upstream release is merged.

Building and testing of the source happens automatically - if a "python setup.py ..." works with the new repo layout it should continue to work for Debian packaging. Sometime tweaks to the testing framework are required e.g. to install test assets or to hot-replace files removed from the source at build time (e.g. the bundled font) but again this tends to be down to how things build in clean Debian chroots rather than how the upstream codebase is organised.

We carry quite a few patches to bend the source to our will when necessary. We have to remove the bundled Source Code Pro font and replace it with an alternative available in Debian. We also need to replace the bundled uflash script and micro:bit hex with references to the separately-packaged uflash script and MicroPython runtime on Debian. These issues reflect fundamental differences between the upstream and DFSG-compatible Debian packages, and will not be affected by layout changes to the repository. Anyone interested in seeing the current patchset can check the debian/patches directory in the mu-editor salsa.debian.org repository linked above.

I hope this information is of some use - if there are any follow-up questions please don't hesitate to ask.

ZanderBrown, 2019-08-16:
That's pretty much what I thought, although probably failed to communicate