mu-editor/mu #892: EP 2019 | Improve checks

tmontes, 2019-07-13:
The current Unix `make check` implementation drives a few checks.

Of those, two of them, feel hard to maintain and read. Here's a snippet from the current `Makefile`:

```
pyflakes:
        find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path
 ./utils -o -path ./venv -o -path ./package \) -type d -prune -o -name '*.py' -print0 | $(
XARGS) pyflakes

pycodestyle:
        find . \( -name _build -o -name var \) -type d -prune -o -name '*.py' -print0 | $(
XARGS) -n 1 pycodestyle --repeat --exclude=package/*,build/*,docs/*,mu/contrib*,mu/modes/a
pi/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
```

Both are based on using `find` to determine the files that should be fed to each tool, respectively. While very powerful and expressive, I believe these can be simplified.

The `pycodestyle` check can be written as a much more direct:

```
pycodestyle --repeat --ignore=E731,E402,W504 --exclude=./mu/contrib/,./mu/modes/api mu tests
```

The `pyflakes` check could be applied the same "approach" with:
```
pyflakes tests mu win_installer.py
```

*However*, for some possibly very obvious reason that I'm failing to see now, this simpler approach generates many `undefined name '_'`, all over the place, while the existing one seems to understand that `gettext` defines `_`. **What am I missing?** :-/

Additionally, the proposed approach to the `pycodestyle` check, makes it run much faster on my already system (~3.5 seconds vs ~10 seconds), which is also nice.

If we change these check on the `Makefile`, then we can probably apply the same approach to their Windows `make.py` counterparts.

Would love to hear feedback on this, in particular regarding the behavioural inconsistency I'm observing... Thanks!

tmontes, 2019-07-13:
Explored a bit more.

Here are my `pycodestyle` timing measurements on macOS, Ubuntu 18.04 VM, Windows 10 VM:

|             | current | proposed |
| ----------- | ------- | -------- |
| macOS       | ~10s    | ~3.5s    |   
| Ubuntu      | >150s   | ~3.0s    |
| Windows     | ~12s    | ~3.5s     |

PS: Faster is nice for everyone, including waiting for CI results. :)

tjguk, 2019-07-13:
I'm up for it. I coded the make.py to mimic the Makefile (for obvious reasons) but I'm quite happy going in the direction you suggest.

(I've already got the env var set!)

tjguk, 2019-09-20:
@tmontes can this be closed now that #895 is merged?

tmontes, 2019-09-24:
I was waiting for #919 as well and have been ~a week offline. I'm back now, a closing this! :)

closed 2019-09-24 by tmontes