mu-editor/mu #897: Webmode hard crash when another server is listening on localhost:5000

Teagum, 2019-07-16:
I have a Flask app running on localhost:5000, which is totally unrelated to Mu. When I start Mu and create a further Flask app in its web mode and hit "Run", an `OSError` is raised stating that the address is already in use. When I click on "Stop" Mu crashes.

### Steps
- Flask app listening on localhost:5000
- Start further app in Mu web mode
- OSError
- Mu crahses

### Traceback
> * Serving Flask app "cxc.py" (lazy loading)
>  * Environment: development
>  * Debug mode: on
> Traceback (most recent call last):
>   File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 193, in _run_module_as_main
>   File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 85, in _run_code
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/__main__.py", line 14, in <module>
>     main(as_module=True)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/cli.py", line 894, in main
>     cli.main(args=args, prog_name=name)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/cli.py", line 557, in main
>     return super(FlaskGroup, self).main(*args, **kwargs)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 717, in main
>     rv = self.invoke(ctx)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 1137, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 956, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 555, in invoke
>     return callback(*args, **kwargs)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/decorators.py", line 64, in new_func
>     return ctx.invoke(f, obj, *args, **kwargs)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/click/core.py", line 555, in invoke
>     return callback(*args, **kwargs)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/flask/cli.py", line 771, in run_command
>     threaded=with_threads, ssl_context=cert)
>   File "/Applications/mu-editor.app/Contents/Resources/app_packages/werkzeug/serving.py", line 987, in run_simple
>     s.bind(server_address)
> OSError: [Errno 48] Address already in use

### Log
This is the last part of the log from starting Mu until it crashes. Is looks same every time take the above steps.

> 2019-07-16 11:42:11,951 - mu.logic:706(restore_session) INFO: Loaded files.
> 2019-07-16 11:42:11,952 - mu.logic:710(restore_session) INFO: User defined environment variables: [['FLASK_APP', 'wwww.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/michaelblass/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/py [... cut ...]
> 2019-07-16 11:42:55,072 - mu.interface.panes:789(start_process) INFO: Runner: /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3
> 2019-07-16 11:42:55,072 - mu.interface.panes:790(start_process) INFO: Args: ['-m', 'flask', 'run']
> 2019-07-16 11:42:55,083 - mu.modes.web:175(start_server) DEBUG: Starting Flask app.
> 2019-07-16 11:42:56,775 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/michaelblass/Library/Logs/mu/mu.log
> 2019-07-16 11:42:56,776 - mu.logic:163(installed_packages) INFO: Packages found: []

tmontes, 2019-07-17:
Good catch @Teagum! Thanks for the detailed report. :)

devdanzin, 2020-12-17:
Cannot reproduce on Windows 10, even if change the `s.bind(server_address)` line into `raise OSError(48)`. Binding a server with `allow_reuse_address = False` also raises an exception, but no crashes. Can you still repro?

There's a chance it was fixed by efb7c70efa53ab5a147170a1920204d46a0cd308.

ntoll, 2021-07-10:
Cannot reproduce.

carlosperate, 2021-07-10:
I can reproduce with this steps:

- Go to any directory and run `python3 -m http.server 5000 --bind 127.0.0.1`
- Open Mu and start a flash server
- Error printed is:
```
 * Serving Flask app "flask_test.py" (lazy loading)
 * Environment: development
 * Debug mode: on
Traceback (most recent call last):
  File "/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/flask/__main__.py", line 15, in <module>
    main(as_module=True)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/flask/cli.py", line 967, in main
    cli.main(args=sys.argv[1:], prog_name="python -m flask" if as_module else None)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/flask/cli.py", line 586, in main
    return super(FlaskGroup, self).main(*args, **kwargs)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/Users/microbit-carlos/Library/Applicat

---------- FINISHED ----------
exit code: 1 status: 0ion Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/flask/cli.py", line 852, in run_command
    run_simple(
  File "/Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210708-230234/lib/python3.8/site-packages/werkzeug/serving.py", line 982, in run_simple
    s.bind(server_address)
OSError: [Errno 48] Address already in use
```
- Then click stop and it crashes, no crash report raised

ntoll, 2021-07-10:
Aha.... thank you @carlosperate. I think this may be fixed by @dybber's work on #1319.

ntoll, 2021-07-11:
I can confirm this is fixed by #1319 (which I just merged to master).

closed 2021-07-11 by ntoll