mu-editor/mu #902: cannot copy files, but repl works fine

prankousky, 2019-07-20:
Hi everybody,

* What you were trying to do,

I flashed `micropython` onto a few `nodeMCU` devices. At first, I was able to do "everything" (run local scripts, run scripts on device via `repl`, via and move files to/from local storage to/from noceMCU). 

However, no matter what I do now, I cannot access files on the devices any longer. I tried both `CP2102` and `CH34OG` chipset, but neither will work any longer. 

The really weird thing is that it **did** work before, then stopped working all the sudden. I messed with the `boot.py` on my first test device, so I assumed that might have messed it up somehow; when I re-flashed that drive (erased flash beforehand), it still didn't work, so I flashed a brand new (still packaged, not used nor flashed) device. I then **could** access file system once, but afterwards it did not work any longer. Re-flashing did do nothing.

So I tried a third device, which was also still packaged. Flashed most recent micropython (after erasing its flash first), but then I couldn't even access files at first attempt.

* What steps you took to make this happen,

1. downloaded current micropython version and flashed to device
2. installed `mu-editor` 
3. tried connecting
3.1 no success because my user was not allowed to access `/dev/ttyUSB0`/1
3.2 added my user to group `uucp` so that I could access it without running `mu-editor` as root

* What you expected to happen,

Being able to move files between my local storage and my nodeMCU

* What actually happened,

**OKAY, THIS IS WEIRD**
I *just* tried accessing filesystem again, and **now** it did work. However, over the last few days, this always produced an error (which I wanted to reproduce right now so I could copy the exact output and paste here). I did both disconnect / power off each nodeMCU as well as reboot my machine multiple times, yet this problem consisted. Now that I actually started writing this and wanted to reproduce the error, it will not happen. wtf?

* Why this difference is problematic (it may not be a bug!),

Because I wasn't able to test things on the board that required libraries to be **on** the board while I was not able to copy them onto the board. 

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

* `Mu 1.1.10 alpha.2`
* `Linux archbook 5.2.1-arch1-1-ARCH #1 SMP PREEMPT Sun Jul 14 14:52:52 UTC 2019 x86_64 GNU/Linux`
* running `mu-editor` as regular user (not root), *but* added this user to the group that is able to access `/dev/ttyUSB*`

I have not attached the log because the error did not pop up this time. I purposely quit and restarted mu multiple times, unplugged and reinserted the usb cable with my nodeMCU, but now it works. However, I have no idea **why** it suddenly works, nor what I can do to fix this the next time it might happen. So I decided I'd still post this issue and hopefully be set for the future without not being able to work with my boards for several days.

If this is a stupid question please consider that I am still a beginner with no professional IT background and just try to learn by doing. I tried looking things up before asking here but did not find anything helpful.

Thanks in advance for your support :)

dhalbert, 2019-07-20:
Hi - your problems are very unlikely to be related to anything having to do to Mu. Mu is careful to fully write files to the filesystem. The filesystem in MicroPython (and CircuitPython, of which I am a developer) is a FAT12 filesystem that is relatively easily corrupted.

If you are copying files to the boards from a shell, make sure you also run `sync` after doing do. That will force all data to the drive immediately. Otherwise Linux may take 10-20 seconds to finish writing to the drive. If you reset or unplug the board before that, corruption may occur.

Corruption can also occur if your own program writes to the filesystem, and then the hos does too. MicroPython does not prevent the host computer and Python from simultaneously writing to the filesystem, and they can interfere with each other if both are writing. In particular, the host computer has no idea that your program may have changed the filesystem: it caches information about the filesystem that may go out of date if there is local writing by Python.

If you recently updated Arch Linux but did not reboot after doing so, make sure you reboot. I have seen an Arch user with problems that went away after rebooting.

If you have further issues, https://forum.micropython.org is probably the best place to ask.

CaptainJackey, 2019-07-25:
Because of the serial port flow not close in Mu for ESP, this conflic with the NodeMCU's uart auto download circuit。