mu-editor/mu #906: Closeed the Serial port flow control with ESP8266/ESP32

CaptainJackey, 2019-07-25:
Now I use Mu for MicroPython with NODEMCU(esp8266 and esp32).
normally it has an auto downloand with USB to UART IC( CP210X)。
I found that the REPL and file for ESP sometimes not work because the uart flow control conflict with the auto download circuit(GPIO0 and RST)。

I suggest closed the flow control for the REPL and file funtion when use ESP8266/ESP32,Thank you so much.

dybber, 2020-09-05:
Hi @CaptainJackey 

Sorry for not responding earlier. I tried understanding, what issue you are experiencing, but I'm not quite sure I get it. When does the issue occur, and how might I replicate it?

dybber, 2020-09-05:
Just looking up what default setting we use for flow control, and apparently it defaults to off: 
> "The default value is NoFlowControl, i.e. no flow control."
https://doc.qt.io/qt-5/qserialport.html#flowControl-prop 

If we wanted to enable flow control, the place to do it is here: https://github.com/mu-editor/mu/blob/master/mu/modes/base.py#L98

CaptainJackey, 2020-09-08:
Thanks your reply , "The default value is NoFlowControl, i.e. no flow control." is fit.

237827161@qq.com
 
From: Martin Dybdal
Date: 2020-09-05 22:05
To: mu-editor/mu
CC: CaptainJackey; Mention
Subject: Re: [mu-editor/mu] Closeed the Serial port flow control with ESP8266/ESP32 (#906)
Just looking up what default setting we use for flow control, and apparently it defaults to off:
"The default value is NoFlowControl, i.e. no flow control."
https://doc.qt.io/qt-5/qserialport.html#flowControl-prop
If we wanted to enable flow control, the place to do it is here: https://github.com/mu-editor/mu/blob/master/mu/modes/base.py#L98

dybber, 2020-09-08:
Okay, can I close this issue?

CaptainJackey, 2020-09-08:
ok, thanks.

237827161@qq.com
 
From: Martin Dybdal
Date: 2020-09-08 14:04
To: mu-editor/mu
CC: CaptainJackey; Mention
Subject: Re: [mu-editor/mu] Closeed the Serial port flow control with ESP8266/ESP32 (#906)
Okay, can I close this issue?

closed 2020-09-08 by dybber