mu-editor/mu #909: Travis OS X builds broken due to docutils version conflict

keith-packard, 2019-07-31:
I submitted pull request #908 yesterday and the Travis OS X build failed due to a conflicting
dependency on docutils:

mu-editor → sphinx → docutils >= 0.12
        gets docutils 0.15.2

mu-editor → briefcase == 0.2.9 → boto3 >=1.4.4 → botocore <1.13.0,>=1.12.99 (1.12.199) → docutils <0.15,>=0.10
        conflicts with 0.15.2

Here's the full log:
        https://travis-ci.org/mu-editor/mu/jobs/565847853

I kludged around this in my pull request by adding an explicit dependency on docutils<0.15, but I suspect someone with more knowledge about python packaging would have a better fix.

tmontes, 2019-08-01:
@keith-packard,

I found this intriguing and investigated a bit: dependency conflicts are things we all want to avoid! :) Here are my preliminary findings and thoughts:

* The Travis test script is [installing "everything"](https://github.com/mu-editor/mu/blob/acf9cb00893d776b5fe52d1c8c4fb9ab04c48503/.travis.yml#L53)  before running tests on macOS which, clearly, is not needed. If it only installed Mu and the test dependencies (via a `pip install .[tests]`, for example), neither the "sphinx -> docutils >= 0.12" dependency would be brought in (used for doc building), nor the "briefcase == 0.2.9 -> ... -> docutils <0.15,>=0.10" would (used for macOS packaging).
* On the other hand, I found it curious that botocore was explicitly depending on "docutils < 0.15". I found [this recent discussion](https://github.com/boto/botocore/pull/1788) around the topic.

IIUC, if we did nothing, someday in the (hopefully) near future, the "briefcase" dependency path will unlock more recent docutils version and things will eventually be fine. However, I guess we could improve (and optimize) the Travis test script not to install unneeded things in its preparation to run tests.

Thoughts? :)

keith-packard, 2019-08-01:
It's a bit disturbing to find a ceiling for any dependency as that could pin packages behind security or stability fixes. Not sure how to deal with this in the context of mu-editor though.

I think it's probably good that Travis caught this now, rather than discovering this when trying to release, so I wouldn't jump to trying to work around this by reducing the work done during the Travis build.

tmontes, 2019-08-02:
> It's a bit disturbing to find a ceiling for any dependency as that could pin packages behind security or stability fixes. Not sure how to deal with this in the context of mu-editor though.

Agreed. It seems that the 0.15 docutils release broke botocore on Python 2 which, AFAICT, is a dependency for aws-cli, used by many, I suppose (?).

With that, 12 days a go, what looks like a quick fix was merged into botocore [here](https://github.com/boto/botocore/pull/1788). What I find really interesting is that the [first commit](https://github.com/boto/botocore/pull/1788/commits/a82fcf795a4cc173a5d494a7a4db8123e98a7a86) in that PR went with `docutils!=0.15` -- not setting a version ceiling like you just identified -- and later, for some unclear reason to me, the [subsequent commit](https://github.com/boto/botocore/pull/1788/commits/add5c8b6b4e208fb0eaba49302df9c3d75b604b9) went with `docutils>=0.10,<0.15` instead. Why was that the case? Not sure!

In the meantime, docutils has released 0.15.1 and 0.15.2 that I suppose address the Python 2 regression, given the [few comments on the above mentioned botocore PR](https://github.com/boto/botocore/pull/1788#issuecomment-514402088) asking for some kind of reversion there. 

> I think it's probably good that Travis caught this now, rather than discovering this when trying to release, so I wouldn't jump to trying to work around this by reducing the work done during the Travis build.

I both agree and disagree with that.

Of course it is good that Travis caught that: but it did that by chance, maybe the "proper" way to have it catch that is by having it not only run the test and coverage suite, but also, running the packaging process... :)

So a Travis improvement could go along the following steps:
* Install Mu and testing dependencies: run tests and coverage.
* Install Mu and packaging dependencies: package mu (something AppVeyor is doing, on the Windows platform).
* Install Mu and documentation dependencies: build docs.

Each running independently of the other. Of course, such setup wouldn't have caught the dependency conflicts you identified. That could lead us to add yet another Travis step:
* Install all development dependencies: setting up a full fledged dev environment works. :)

dybber, 2020-09-29:
I'm pretty sure this is fixed now, if you rebase your branch on top of master, you shouldn't get that error. I had a long endeavor in June trying to figure out what was going on (#1065).

closed 2020-09-29 by dybber