mu-editor/mu #911: REPL doesn't open on Windows 7 using Mu Portable in Python mode

oragnu, 2019-08-14:
I was using Mu for the first time and tried to access the REPL and Mu closed.

To reproduce:
* Download Mu Portable v1.0.2, and unzip to a directory
* Run "Launch Mu.bat" or open a Command Prompt, navigate to Mu directory, and type 'start Python\pythonw.exe "Mu.launch.pyw"'
* Select Python mode
* Click the REPL icon in the command bar

Result: Mu window closes
Expected: REPL
NOTE: Same result with either the 32-bit or 64-bit version of Mu Portable

Environment: Windows 7 64-bit with Python 2.7.x/3.6.x/3.7.x installed and in the System Path

The full logfile is below the line, the relevant portion of the error here:

2019-08-14 15:25:11,698 - mu.modes.python3:224(toggle_repl) INFO: Toggle REPL on.
2019-08-14 15:25:11,705 - mu.modes.python3:57(start_kernel) INFO: ['pkgs', 'c:\\apps\\portamu_1.0.2_win32\\Python\\python36.zip', 'c:\\apps\\portamu_1.0.2_win32\\Python', 'c:\\apps\\portamu_1.0.2_win32\\pkgs', 'C:\\Users\\abcdef\\AppData\\Roaming\\Python\\Python36\\site-packages', 'pkgs\\IPython\\extensions']
2019-08-14 15:25:11,705 - mu.modes.python3:62(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2019-08-14 15:25:11,708 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "pkgs\mu\modes\python3.py", line 72, in start_kernel
  File "pkgs\traitlets\traitlets.py", line 958, in \_\_new\_\_
  File "pkgs\traitlets\traitlets.py", line 986, in setup_instance
  File "pkgs\traitlets\traitlets.py", line 977, in setup_instance
  File "pkgs\traitlets\traitlets.py", line 1691, in instance_init
  File "pkgs\traitlets\traitlets.py", line 1696, in _resolve_classes
  File "pkgs\traitlets\traitlets.py", line 1507, in _resolve_string
  File "pkgs\traitlets\utils\importstring.py", line 34, in import_item
**ModuleNotFoundError: No module named 'jupyter_client.session'**

---
Starting Mu 1.0.2
2019-08-14 15:25:06,032 - root:113(run) INFO: uname_result(system='Windows', node='--------', release='7', version='6.1.7601', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2019-08-14 15:25:06,032 - root:114(run) INFO: Python path: ['pkgs', 'c:\\apps\\portamu_1.0.2_win32\\Python\\python36.zip', 'c:\\apps\\portamu_1.0.2_win32\\Python', 'c:\\apps\\portamu_1.0.2_win32\\pkgs', 'C:\\Users\\abcdef\\AppData\\Roaming\\Python\\Python36\\site-packages', 'pkgs\\IPython\\extensions']
2019-08-14 15:25:06,032 - root:115(run) INFO: Language code: en_US
2019-08-14 15:25:06,194 - mu.logic:543(__init__) INFO: Setting up editor.
2019-08-14 15:25:06,194 - mu.logic:562(__init__) INFO: Settings path: C:\Users\abcdef\AppData\Local\python\mu\settings.json
2019-08-14 15:25:06,194 - mu.logic:563(__init__) INFO: Session path: C:\Users\abcdef\AppData\Local\python\mu\session.json
2019-08-14 15:25:06,194 - mu.logic:564(__init__) INFO: Log directory: C:\Users\abcdef\AppData\Local\python\mu\Logs
2019-08-14 15:25:06,194 - mu.logic:565(__init__) INFO: Data directory: C:\Users\abcdef\AppData\Local\python\mu
2019-08-14 15:25:06,239 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-08-14 15:25:06,812 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\abcdef\mu_code
2019-08-14 15:25:06,812 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\abcdef\AppData\Local\python\mu\session.json
2019-08-14 15:25:06,812 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-08-14 15:25:06,812 - mu.logic:649(restore_session) INFO: Loaded files.
2019-08-14 15:25:06,812 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-08-14 15:25:06,812 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-08-14 15:25:07,364 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-08-14 15:25:07,397 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\abcdef\mu_code
2019-08-14 15:25:11,698 - mu.modes.python3:224(toggle_repl) INFO: Toggle REPL on.
2019-08-14 15:25:11,705 - mu.modes.python3:57(start_kernel) INFO: ['pkgs', 'c:\\apps\\portamu_1.0.2_win32\\Python\\python36.zip', 'c:\\apps\\portamu_1.0.2_win32\\Python', 'c:\\apps\\portamu_1.0.2_win32\\pkgs', 'C:\\Users\\abcdef\\AppData\\Roaming\\Python\\Python36\\site-packages', 'pkgs\\IPython\\extensions']
2019-08-14 15:25:11,705 - mu.modes.python3:62(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2019-08-14 15:25:11,708 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "pkgs\mu\modes\python3.py", line 72, in start_kernel
  File "pkgs\traitlets\traitlets.py", line 958, in \_\_new\_\_
  File "pkgs\traitlets\traitlets.py", line 986, in setup_instance
  File "pkgs\traitlets\traitlets.py", line 977, in setup_instance
  File "pkgs\traitlets\traitlets.py", line 1691, in instance_init
  File "pkgs\traitlets\traitlets.py", line 1696, in _resolve_classes
  File "pkgs\traitlets\traitlets.py", line 1507, in _resolve_string
  File "pkgs\traitlets\utils\importstring.py", line 34, in import_item
ModuleNotFoundError: No module named 'jupyter_client.session'

carlosperate, 2020-11-08:
Thanks for the report @oragnu, could you try one of the latests builds from http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/ (there are not official releases, so they might contain other bugs) and see if the problem persist in one of those versions?

oragnu, 2020-11-08:
@carlosperate , thanks for replying. Unfortunately I have moved on from my Win7 system so I won't be able to try your suggestion.

closed 2020-11-08 by oragnu