mu-editor/mu #912: pin0.digital_write() returns True when touched on Microbit

pluckerpete2, 2019-08-24:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

tmontes, 2019-09-03:
Hi @pluckerpete2,

Thanks for using Mu. I don't have my Microbit at hand to try and reproduce the behaviour you describe. Nevertheless, I checked [the Microbit's micropython docs](https://microbit-micropython.readthedocs.io/en/latest/pin.html#pin-functions) and found:
* It seems that `pin0` is analog in only, are you sure you want to "write digital" data into it?
* The closest named function I could find is `write_digital`, taking one argument - different from your `digital_write` taking no arguments.

I was wondering if you could please confirm your experience such that I can try to reproduce it -- or if what you are trying to do may need to be done differently.

PS: There is also the chance that the micropython version currently being shipped with Mu is so old that it is not API compatible with the docs I've linked -- I'd venture saying those odds are very low. But let's see, ok?

Standing by on your feedback.

tmontes, 2019-10-03:
@pluckerpete2, you haven't shared information back for quite a while... Have you solved your issue, or do you believe there is something in Mu that needs fixing? Thanks for letting us know about that.

dybber, 2020-06-14:
Hi @pluckerpete2, I just checked with my microbit. As @tmontes have previously mentioned, there's no `digital_write`, and writing to a digital pin doesn't return anything, and digital write is thus not what you need to react on touch. To react on touch on pin 0, what you would want to do is to use `pin0.is_touched()` which returns True or False depending on the resistance between pin 0 and GND. Here's a complete example:

```
from microbit import *

while True:
    if pin0.is_touched():
        display.show(Image.HAPPY)
    else:
        display.show(Image.SAD)
```

Remember that you must touch both pin 0 and GND, for the microbit to discover the change in resistance.

I will close this issue, as it's not an issue with Mu, but an issue with microbit.

closed 2020-06-14 by dybber