mu-editor/mu #913: Neopixel Attribute error

astrotutor9, 2019-09-03:
If you are reporting a bug, we would like to know:

* What you were trying to do,
This exercise was originally written using a different Windows PC with Mu 1.0.1 (I think).
The original save from that I have on a flash drive.

If the code is rewritten and flashed to the Microbit it throws an attribute error line 4. Yet, if I open the original file on the flash drive, copy and paste the offending line from that into the new tab, comment out the offending line 4 the Microbit does not complain. Re-comment back and the Line 4 error reappears.
This error shows up on multiple Raspberry Pi and now a new PC. All now running 1.0.2.

[testing_Mu.zip](https://github.com/mu-editor/mu/files/3569168/testing_Mu.zip)

* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)
Log
2019-09-03 11:52:50,020 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-09-03 11:52:50,023 - root:113(run) INFO: uname_result(system='Windows', node='LAPTOP-ACMH4H7C', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2019-09-03 11:52:50,023 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Users\\astro\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2019-09-03 11:52:50,023 - root:115(run) INFO: Language code: en_GB
2019-09-03 11:52:50,253 - mu.logic:543(__init__) INFO: Setting up editor.
2019-09-03 11:52:50,254 - mu.logic:562(__init__) INFO: Settings path: C:\Users\astro\AppData\Local\python\mu\settings.json
2019-09-03 11:52:50,254 - mu.logic:563(__init__) INFO: Session path: C:\Users\astro\AppData\Local\python\mu\session.json
2019-09-03 11:52:50,254 - mu.logic:564(__init__) INFO: Log directory: C:\Users\astro\AppData\Local\python\mu\Logs
2019-09-03 11:52:50,254 - mu.logic:565(__init__) INFO: Data directory: C:\Users\astro\AppData\Local\python\mu
2019-09-03 11:52:50,290 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-09-03 11:52:51,312 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\astro\mu_code
2019-09-03 11:52:51,314 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\astro\AppData\Local\python\mu\session.json
2019-09-03 11:52:51,314 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['C:\\Users\\astro\\Documents\\1A - Software Cornwall\\Tech Jam\\Microbit_Satellite\\satellite_station_transmitter.py', 'C:\\Users\\astro\\Documents\\1A - Software Cornwall\\Tech Jam\\Microbit_gravity\\gravity_sender.py', 'C:\\Users\\astro\\Documents\\1A - Software Cornwall\\Tech Jam\\Microbit_gravity\\gravity_receiver.py', 'C:\\Users\\astro\\Documents\\1A - Software Cornwall\\Tech Jam\\Microbit Servo 8 Ball\\Servo_8_Ball.py', 'C:\\Users\\astro\\Documents\\1A - Software Cornwall\\Tech Jam\\Microbit_Satellite\\satellite_station_origin_morse.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-09-03 11:52:51,314 - mu.logic:715(_load) INFO: Loading script from: C:\Users\astro\Documents\1A - Software Cornwall\Tech Jam\Microbit_gravity\gravity_sender.py
2019-09-03 11:52:51,315 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2019-09-03 11:52:51,316 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2019-09-03 11:52:51,316 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2019-09-03 11:52:51,316 - mu.logic:789(_load) DEBUG: from microbit import *
from math import sqrt
import radio

radio.config(channel=7)
radio.on()

while True:
    x, y, z = accelerometer.get_values()
    gravity = round(sqrt((x**2)+(y**2)+(z**2)))
    radio.send(str(gravity))
    sleep(5)
2019-09-03 11:52:51,854 - mu.logic:715(_load) INFO: Loading script from: C:\Users\astro\Documents\1A - Software Cornwall\Tech Jam\Microbit_Satellite\satellite_station_origin_morse.py
2019-09-03 11:52:51,856 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2019-09-03 11:52:51,856 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2019-09-03 11:52:51,856 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2019-09-03 11:52:51,856 - mu.logic:789(_load) DEBUG: from microbit import *
import radio

# turn on the radio using default channel 7
radio.config(channel=7)
radio.on()

dots_and_dashes = []
[... 328 lines cut ...]

import neopixel

ring = neopixel.Neopixel(pin0, 16) # This throws attribute error

# ring = neopixel.NeoPixel(pin0, 16)  # This line is good!

ring[0] = (100, 0, 0)

ring.show()
2019-09-03 11:59:52,050 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in C:/Users/astro/Documents/1A - Software Cornwall/testing_Mu.py.
2019-09-03 12:00:02,044 - mu.logic:871(save_tab_to_file) INFO: Saving script to: C:/Users/astro/Documents/1A - Software Cornwall/testing_Mu.py
2019-09-03 12:00:02,045 - mu.logic:872(save_tab_to_file) DEBUG: from microbit import *

import neopixel

ring = neopixel.Neopixel(pin0, 16) # This throws attribute error

# ring = neopixel.NeoPixel(pin0, 16) This line is good!

ring[0] = (100, 0, 0)

ring.show()
2019-09-03 12:00:02,061 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in C:/Users/astro/Documents/1A - Software Cornwall/testing_Mu.py.
2019-09-03 12:01:42,897 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\astro\AppData\Local\python\mu\Logs\mu.log

tmontes, 2019-09-03:
@astrotutor9,

I have no experience with NeoPixels at all, but looking at the code you shared:

```python
from microbit import *
import neopixel

ring = neopixel.Neopixel(pin0, 16) # This throws attribute error
# ring = neopixel.NeoPixel(pin0, 16) This line is good!

ring[0] = (100, 0, 0)

ring.show()
```

...maybe you need to review line 4's casing and use `neopixel.NeoPixel` instead of `neopixel.Neopixel`? (note upper/lower casing difference)

My 2c.

tmontes, 2019-09-03:
@astrotutor9,

I noticed you edited your original text to include:

> Forget this! Spelling error. This error has come up so many times and yet the error was not spotted.
> neopixel.NeoPixel(pin0, 16) not neopixel.Neopixel(pin0, 16)

Does that mean that this is no longer suspected to be a Mu related issue? If that is the case, can you please close the issue, to facilitate triaging? Otherwise, could you re-state/re-frame the difficulties you are finding with Mu?

Thanks. :)

tmontes, 2019-10-03:
@astrotutor9, I suppose this issue is resolved, correct? If so, can you please go ahead and close it? Thanks.

dybber, 2020-06-14:
Hi @astrotutor9, I will close this, as it seems resolved (see the previous two comments), feel free to reopen, if that's not the case.

closed 2020-06-14 by dybber