mu-editor/mu #914: Feature Request: Add UI indication of which USB (or path) a tab's code is attached to

anecdata, 2019-09-05:
In the context of editing CircuitPython on macOS Mu, I almost always have multiple tabs open with identical code, all called code.py, running on similar devices, each on its own USB port. But unless I manually track which tab is which device (write down or order the tabs in the physical order of the devices on my desk, for example), it gets really confusing.

The motivation for this is developing and testing code that works on variations of hardware. I'll test an edit on one variation in one tab, then want to see how it behaves on a different variation in a different tab.

It would be nice to have some indication in the UI of which USB corresponds to which tab (or at least the open tab).

dybber, 2020-02-15:
Hi anecdata.

Would it be a solution for you, to be able to select devices as currently discussed over here: https://github.com/mu-editor/mu/issues/988 ?

anecdata, 2020-02-15:
Does that replace the "Load" button? I think it's fine. Anything that shows which USB  the current macOS Mu editing tab is connected to is great, so it's apparent which code tab goes with which device.

ZanderBrown, 2020-02-16:
Not sure doing this per tab is particularly valuable, the current proposal I would expect to be "global"

anecdata, 2020-02-16:
If it's global, it doesn't help with this issue. What's desired for this issue is for the case of having multiple tabs open, each connected to a different device, but curently having no visual indication of which tab connects with which device.

dybber, 2020-02-16:
And just to make sure, it wouldn't work for you to have two different Mu instances open, connected to separate devices?

anecdata, 2020-02-16:
Well, it could be accomplished that way I suppose if I can run multiple instances in macOS (so I'd make an arbitrarily large number of copies of Mu.app with different names, or...?)

But tabs are so much easier. I may have 1, 3, or many, and the devices to which they connect will vary.

dybber, 2020-02-16:
The indicator could be solved (for example as in the screenshot in the issue I linked to), the difficult part would be to make Mu able to connect with multiple devices and switch between them. That will require much more engineering.

I can see your point though, on Mac it's actually pretty hard to open multiple instances of Mu (at least currently it is).

anecdata, 2020-02-16:
> the difficult part would be to make Mu able to connect with multiple devices and switch between them. That will require much more engineering.

But I do this all the time currently. I just have to manually remember which device is in which tab because there's no visual indicator (and it can get very confusing with more than 2 or 3).

anecdata, 2020-02-16:
Maybe this will help: 
[image]
Each tab is a different USB device, but unless I remember the order I loaded them, I have no way to know which is which. And this in only 4.