mu-editor/mu #915: Python-Mode console does not support getpass

r0the, 2019-09-08:
I tried to use the `getpass` module to read a password from the mu console. While entering the user name using `input` works fine, the program hangs a the `getpass` call. The prompt is not shown.

I think it's important to have a method to read passwords without displaying them, especially if you want to demonstrate a program in front of a class.

Here is my program:
```
import getpass

print("BEFORE")
a = input("User:")
b = getpass.getpass("Password:")
print("AFTER")
```

I'm using Mu 1.1.0 alpha 2 on Debian Buster. Here's the log:

```
2019-09-08 13:02:56,043 - mu.logic:951(save_tab_to_file) DEBUG: import getpass

print("BEFORE")
a = input("User:")
b = getpass.getpass("Password:")
print("AFTER")

2019-09-08 13:02:56,059 - mu.interface.panes:691(start_process) INFO: Running script: /home/stefan/mu_code/test_getpass.py
2019-09-08 13:02:56,059 - mu.interface.panes:693(start_process) INFO: Running with interactive mode.
2019-09-08 13:02:56,060 - mu.interface.panes:696(start_process) INFO: Command args: []
2019-09-08 13:02:56,061 - mu.interface.panes:755(start_process) INFO: Running with environment variables: [['FLASK_APP', 'web.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-editor:/usr/lib/python37.zip:/usr/lib/python3.7:/usr/lib/python3.7/lib-dynload:/home/stefan/.local/lib/python3.7/site-packages:/usr/local/lib/python3.7/dist-packages:/usr/lib/python3/dist-packages:/home/stefan/.local/share/mu/site-packages:/usr/lib/python3/dist-packages/IPython/extensions'], ['PYTHONPATH', '/opt/mu-edit [... cut ...]
2019-09-08 13:02:56,063 - mu.interface.panes:790(start_process) INFO: Args: ['-i', '/home/stefan/mu_code/test_getpass.py']
2019-09-08 13:03:02,221 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /home/stefan/.cache/mu/log/mu.log
2019-09-08 13:03:02,222 - mu.logic:163(installed_packages) INFO: Packages found: ['/home/stefan/.local/share/mu/site-packages/et_xmlfile-1.0.1.dist-info', '/home/stefan/.local/share/mu/site-packages/openpyxl-2.6.3.dist-info', '/home/stefan/.local/share/mu/site-packages/jdcal-1.4.1.dist-info']

```

r0the, 2019-09-08:
I get the same problem with Mu 1.1.0 alpha 2 on Windows 10.

tmontes, 2019-09-09:
@r0the,

Thanks for reporting this. You are hitting a very non-trivial systems related issue here. I suppose that, if instead of running such program from Mu, you type it in in the REPL, you'll get a different behaviour: you'll get the prompt, a warning telling you about the impossibility of "hiding away" the typed input, and, other than that (important detail) it does work.

What's going on?

* `getpass.getpass` is clearly directed at terminal I/O based programs.
* Terminals (and GUI terminal emulations) are non-trivial things: without going into all the details, they tend to know how to handle some clever-ish output (colour, positioning, showing/hiding the cursor, etc.) and also some clever-ish input (like telling you their geometry, or not displaying the typed in text when reading passwords).
* With Mu, the "terminal thing" you get when running programs is not the same kind of "thing" you get when using the REPL: that's why you get the different behaviour I pointed out above.

More:
* Terminal concepts and APIs are very different between UNIX-like (Linux, macOS, BSD, etc.) systems and Windows: even though the user-experience appears to be comparable.
* `getpass.getpass` expects to interact with a "terminal-like thing" that Mu does not provide in full: that's why it only kind of works from within Mu. More, while `getpass.getpass` tries to be cross-platform, it does a somewhat limited effort at addressing it on Windows -- the result is that, on Windows, when running from Mu (REPL is different, recall) `getpass.getpass` will perform the I/O on the terminal you launched Mu from... (don't ask me where the I/O goes when launched from the GUI, I have no idea)

Here's a quick representation of that:

[image]

So what?
* Mu could have better implementations of the "terminal things" it uses for both the REPL and running users code:
  * They should behave identically: REPL or running should provide the same terminal I/O experience.
  * They should both be 100% cross platform.
  * Both would handle nice terminal-like features like, say, not-echoing user input as in your case, support colour output, and more.
* Implementing those on UNIX-like systems is something that could be done, with some effort. Adding Windows to the mix makes it much much harder given that the OS level abstractions are very different (even though, AFAICT, Microsoft is adding a PTY-like concept to Windows 10 which should make creating cross-platform advanced terminal based software easier).

With all of this said, I still haven't pointed you at a "solution to your problem"! :( I do hope, at least, you get a feeling of why you are getting the results you're getting, and why "fixing that" is not something easy / quick.

I'm wondering about your target audience -- to whom you say you are explaining how to read passwords "securely" -- and Mu's target audience: your case is clearly targeting terminal based interaction which might be considered somewhat advanced-ish... Not really sure.

Would love to hear other people's thoughts.

r0the, 2019-09-09:
@tmontes Thank you for your comprehensive answer. I really didn't know that console interaction was that complicated.

My target audience are senior high school students with special interest in computer science and programming.

After your explanation, I'm thinking a better approach is probably to use tkinter (with guizero) for reading user input.

tmontes, 2019-09-09:
> Thank you for your comprehensive answer. I really didn't know that console interaction was that complicated.

It is, indeed. Most people aren't aware of that, though: because it does "feel" simple, I guess. :)

> My target audience are senior high school students with special interest in computer science and programming.
> 
> After your explanation, I'm thinking a better approach is probably to use tkinter (with guizero) for reading user input.

I would say that if they are comfortable with the command line interface (the only reason to use the `getpass` module, AFAICT) you could try to explain that your sample program works, as long as it is launched from the CLI itself, with something like:

```
$ python program.py
```

Otherwise, like you suggest, maybe a GUI based solution is the way to go and, in that case, the `getpass` module is of no use. :)

ZanderBrown, 2019-09-09:
> colour, positioning, showing/hiding the cursor, etc.

Of course unlike Windows UNIX doesn't really have a terminal api, just in/out streams which are assumed to have a DEC VT100 at the other end

tmontes, 2019-09-09:
> Of course unlike Windows UNIX doesn't really have a terminal api, just in/out streams which are assumed to have a DEC VT100 at the other

*[respectfully disagreeing while trying to be informative]*

It actually does, even though most programmers never get to use them, thus commonly assuming nothing but a pair of I/O byte streams and a DEC VT100 behaviour on "the other side of the wire" -- a common misconception, which is perfectly ok!

*[and a somewhat less-clean corner on the "everything is a file" UNIX model, IMHO]*

For the sake of completeness (and whomever may read this in the future) things like [PTYs](http://man7.org/linux/man-pages/man7/pty.7.html) -- a variation of which Windows 10 is [about to implement](https://devblogs.microsoft.com/commandline/tag/pty/) --, TTYs, [termios](http://man7.org/linux/man-pages/man3/termios.3.html) and line disciplines, in kernel, and libraries like [POSIX curses](https://www.unix.com/man-page/posix/3ncurses/ncurses) in user-space, are part of the full set of standardized UNIX terminal-related APIs. Of course the super generic "read/write" bytes from/to this "file-which-is-actually-a-terminal" are also there! :)

Like I noted, AFAICT, these are rarely used/known/understood by most programmers... but:
* All/most are exposed in Python's [`os`](https://docs.python.org/3/library/os.html), [`termios`](https://docs.python.org/3/library/termios.html), and other Standard Library Modules (cool!).
* They are the foundation of very popular UNIX tools like [screen](https://www.gnu.org/software/screen/) or [tmux](https://github.com/tmux/tmux/wiki) (powerful terminal multiplexers), for example!

*[back on topic]*

Correctly *fixing* @r0the's issue would imply:
* Using a complete implementation of a virtualized terminal for both Mu's REPL output and running output.
  * The first one is based on QtConsole which does not implement the full things for general terminal oriented code like `getpass.getpass` to work (see [here](https://github.com/jupyter/qtconsole/issues/50), [here](https://github.com/astropy/astropy/issues/2297), or [here](https://github.com/jupyter/qtconsole/issues/60)).
  * The second one is a QTextEdit, supporting text, but that doesn't even attempt to be a terminal thing (thus even further away from supporting `getpass.getpass`).
* Implementing such a virtualized terminal would be non-trival but feasible if we stuck to UNIX, using the above mentioned APIs.
* I'm not sure doing that for Windows would be feasible with the legacy terminal APIs (not totally familiar with them), but let's assume that: the platform differences would imply a totally different implementation of such "virtualized terminal".
* Lastly, we would probably need to improve `getpass.getpass`'s implementation given that on the Windows platform it is using the Standard Library's [`msvcrt.putwch`](https://docs.python.org/3.6/library/msvcrt.html#msvcrt.putwch) function which does not seem to allow control of "which console to output to".

All in all, a very non-trivial topic to address by-the-book™... :-/

Respectfully. :)

aivarannamaa, 2019-10-08:
In Thonny I solved similar problem by including alternative command for running programs -- "Run in Terminal". You are welcome to reuse the code: https://github.com/thonny/thonny/blob/master/thonny/terminal.py

ntoll, 2019-10-08:
@aivarannamaa that's a very generous suggestion, thank you..! :+1:

My hunch would be to do to "lazy engineer" thing and look around at how others have worked around this problem (if at all) and learn from their approach. That's why open source is so awesome: together we move things forward for mutual benefit.