mu-editor/mu #918: flake8 complais about "F821 undefined name '_'"

tjguk, 2019-09-14:
Having adjusted `make.py` to match the changes to `Makefile` I ran `make flake8` only to have it complain about every line which contains the gettext-related underscore. I'm not in a position to test this on *nix at the moment, but the `Makefile` flake8 seems to rely on the setup.cfg which the `make.py` one does as well.

Has anyone else encountered this? Have I missed something obvious?

/cc @ntoll @tmontes

tmontes, 2019-09-15:
I hit precisely that issue when exploring the improvement in #892 -- see the over-striked text in my first note, there.

In summary, `pyflakes` needs to be told to ignore that, by defining an environment variable `PYFLAKES_BUILTINS=_`, much like it is set in the current `make.py` version [here](https://github.com/mu-editor/mu/blob/master/make.py#L144).

Let me know how that works for you. :)

tjguk, 2019-09-16:
Thanks @tmontes. I was sure you'd have hit the issue already. (And I even remember removing that line from make.py and thinking "we obviously don't need this any longer then"!)

tjguk, 2019-09-20:
Fixed by #919

closed 2019-09-20 by tjguk