mu-editor/mu #921: ui is quite small

ak-is-rolling, 2019-09-23:
i just installed MU on my surface pro 6
i was about use it as it use it at school but in school the ui seems to be alot bigger than usal.
it not too bad but its a bit annoying and it hurts after long time sqwinting at it[
[image]
](url)

many thanks 
Alex

ZanderBrown, 2019-09-23:
In school being:

A) A different device
B) The same surface, just in a different building

I'm also slightly confused, you say you have to squint (and in your screenshot it is quite small) but say the UI is "alot bigger"?

I seem to remember the Surface has a HiDPI display so it's possible we aren't scalling correctly

ak-is-rolling, 2019-09-23:
A) different device
is there a way to change it 
it is just that it looked like this
[image]

ZanderBrown, 2019-09-23:
So the Surface is a misdirect :-)

I know that GIF is from the website, I'm assuming your screenshot is from the school device.

What can you tell us about the school device?

ak-is-rolling, 2019-09-23:
the screen shot is the surface. the gif is what i usually expect at school. we have basic hp nuc's, i expect nomore than a pentium. i think the moniters are 1920 x1080 p.

ZanderBrown, 2019-09-23:
Okay no I'm really confused, it the problem at school or on the Surface

It _sounds_ like your having a problem at school but based on that screenshot there's an issue on the Surface

ak-is-rolling, 2019-09-23:
the surface

ZanderBrown, 2019-09-23:
School was the mislead this whole time?? That's a plot twist :-)

So the problem is that the buttons are _smaller_ than they should be on the Surface?

If so I return to my earlier hypothesis that we aren't handling HiDPI properly

ak-is-rolling, 2019-09-23:
im sorry about that. im amsuming that HiDPi means somehting to do wiht the high pixel denisty of the screen. would this be something looked into for next release or not?

ZanderBrown, 2019-09-23:
Well given how increasingly common HiDPI is it would be something to prioritise really

Yes, HiDPI or "high DPI" is where a display has an especially high pixel density, as happens when you have a high resolution display in a small size (such as on the Surface)

ntoll, 2019-09-23:
Hi folks... I love the epic thread you two are having... :-)

@ak-is-rolling we hope to have a way to configure icon sizes and other UI-ish things in the next alpha. It currently exists as a "draft" solution on my hard drive. I need to clean it up and let folks like @ZanderBrown test it and suggest changes.

It's coming.

ak-is-rolling, 2019-09-23:
thank you so much@ZanderBrown @ntoll for both helping with querrie. its good to knwo there is a good community out there. i will be using this to carry on my education on coding as it is so easy and usefull. i will wait patiently for the update to come.
thanks once again 
Alex

codescrawl, 2019-09-24:
I can chime in on this one as I have a surface pro 6 as well and just happened to install Mu myself today. The font size for the code editor is spot on for me on a default installation. Zooming in and out work just fine too. The problem only relates to the icons and the font underneath each at the top of the UI that don't appear to be scaled correctly. 
[image]

s-oldman, 2019-12-20:
Seconding this issue. If the latest alpha doesn't fully implement DPI scaling, that'd explain things.

Here's a screenshot from one of the new XPS 13 laptops [resolution is 3200x1800, DPI scaling set to 225%, OS is latest Win10 Pro x64 as of 2019-12-20 (v1909, build 18363.535)]:

[image]

That's on a screen size of 13". Can confirm that the label text does look quite small in person :)

jesseh, 2020-01-15:
I had this issues and it resolved when I upgraded to the newest version of Windows 10 (OS build 19541).

Saywa94, 2020-03-17:
Just installed Mu-editor on a surface pro 4. And the problem still persists. Any update on when a solution will be implemented?

Anyway, thank you for developing such an amazing app. It's actually pretty great for beginners.

ronturcotte, 2020-09-02:
I ran into this same problem on a new laptop today and found this thread. It didn't give me a solution, but it did lead me to research HiDPI. Here's what I finally found that worked for me:
1. Right click on the mu icon and go to properties. Note: it took me a while to find this because it depends on where you're right-clicking. In my case, I ultimately put the icon in my task bar, right clicked on it there, then right clicked again on Mu in that pop up menu - then I finally found properties.
2. Click on the Compatibility tab.
3. Scroll down and click on the "Change high DPI Settings" button.
4. Here are the settings that worked for me 
- check the box under Program DPI (use these settings instead of system settings).
- under "Use the DPI that's set for my main display when," choose "I open this program"
- check the box under High DPI Scaling override and choose "System (Enhanced)" in the drop down menu.

[image]

I hope this helps.

carlosperate, 2020-09-03:
Thanks for the tip @ronturcotte! Having this workaround will be definitely be useful for people 👍

dybber, 2020-11-23:
I've attempted to fix this through changing a few settings. Is anyone in this thread who are experiencing the problem, able to try out the pull request at #1150 and check if it works?

dybber, 2020-12-10:
@ak-is-rolling @codescrawl @Saywa94 @ronturcotte, could any of you test this development version of Mu, where I've attempted to fix the issue? https://github.com/mu-editor/mu/actions/runs/413062770 

I don't own a Surface Pro myself, so I need someone to confirm that it's actually fixed.

ak-is-rolling, 2020-12-10:
> @ak-is-rolling @codescrawl @Saywa94 @ronturcotte, could any of you test this development version of Mu, where I've attempted to fix the issue? https://github.com/mu-editor/mu/actions/runs/413062770
> 
> I don't own a Surface Pro myself, so I need someone to confirm that it's actually fixed.

ok i will install the new version of mu now and will come back to you

ak-is-rolling, 2020-12-10:
> @ak-is-rolling @codescrawl @Saywa94 @ronturcotte, could any of you test this development version of Mu, where I've attempted to fix the issue? https://github.com/mu-editor/mu/actions/runs/413062770
> 
> I don't own a Surface Pro myself, so I need someone to confirm that it's actually fixed.

thank you so much it seems to be resolved now. the gui is now opening at the size expected. i appreciate the help.

closed 2020-12-10 by ak-is-rolling

dybber, 2020-12-10:
Great, thanks for reporting in the first place and testing it now. The fix will be included in the next beta-release of Mu.