mu-editor/mu #923: Flask mode does not copy over example files when path changes.

seantibor, 2019-09-28:
I'm trying to make a set of instructions for students to begin using web mode to learn about Flask. Rather than saving flask apps to the mu_code directory, I'm encouraging students to save their Flask apps into subdirectories of mu_code so that they can find their files more easily and don't overwrite template files from their first app if they create a new Flask app.

The beginner template files are not copying over to the new subdirectory when they create their app.

Here are the setup steps I want the students to go through.

- Switch to Flask Mode
- Create a new file
- Save the file in a subdirectory of mu_code
- Open template, images, or css directory to view files.

It looks like the templates directory is being created in the working directory by the base.py's assets_dir method, but none of the files are being copied over. This is happening separately by the logic.py file in the setup method there.

I'm expecting that when I create a new flask app and save it in a directory that the resulting templates folder that is created will also contain the default templates provided in the examples. This works fine when first switching to Web mode in the mu_code working directory, but not when in a separate subdirectory or when a new file is saved in a new path.

I think this is as simple as moving some of the template setup code from logic.py file to the web.py file such that the templates, images, and css directories are created and populated with examples in the same step.

Happy to submit a PR on this but would like to make sure it's heading in the right direction first.

**Technical Info:**
Mac OSX High Sierra 10.14.6
Mu 1.10.alpha.2 installed from Web

Log files below.

--- 

### Log File

2019-09-28 15:16:06,746 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2019-09-28 15:16:06,748 - root:114(run) INFO: uname_result(system='Darwin', node='ff-tibo-002.localdomain', release='18.7.0', version='Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-09-28 15:16:06,748 - root:115(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Users/sean.tibor/Library/Application Support/mu/site-packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-09-28 15:16:06,748 - root:116(run) INFO: Language code: en_US
2019-09-28 15:16:07,120 - mu.logic:587(__init__) INFO: Setting up editor.
2019-09-28 15:16:07,120 - mu.logic:609(__init__) INFO: Settings path: /Users/sean.tibor/Library/Application Support/mu/settings.json
2019-09-28 15:16:07,121 - mu.logic:610(__init__) INFO: Session path: /Users/sean.tibor/Library/Application Support/mu/session.json
2019-09-28 15:16:07,121 - mu.logic:611(__init__) INFO: Log directory: /Users/sean.tibor/Library/Logs/mu
2019-09-28 15:16:07,121 - mu.logic:612(__init__) INFO: Data directory: /Users/sean.tibor/Library/Application Support/mu
2019-09-28 15:16:07,133 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2019-09-28 15:16:07,305 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/sean.tibor/mu_code
2019-09-28 15:16:07,306 - mu.logic:684(restore_session) INFO: Restoring session from: /Users/sean.tibor/Library/Application Support/mu/session.json
2019-09-28 15:16:07,306 - mu.logic:685(restore_session) DEBUG: {'theme': 'night', 'mode': 'circuitpython', 'paths': ['/Users/sean.tibor/mu_code/period2.py', '/Users/sean.tibor/mu_code/seconds_challenge.py'], 'envars': [['FLASK_APP', 'period1helloworld.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['FLASK_APP', 'period1helloworld.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['FLASK_APP', 'period1helloworld.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['FLASK_APP', 'period2webmode.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-packages:/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions'], ['PYTHONPATH', '/Applications/mu-editor.app/Contents/MacOS:/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload:/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages:/Applications/mu-editor.app/Contents/Resources/app:/Applications/mu-editor.app/Contents/Resources/app_packages:/Users/sean.tibor/Library/Application Support/mu/site-pa [... cut ...]
2019-09-28 15:33:16,702 - mu.modes.web:217(load_templates) INFO: /Users/sean.tibor/mu_code/flask app 1/templates
2019-09-28 15:33:16,702 - mu.logic:885(get_dialog_directory) INFO: Using path for file dialog: /Users/sean.tibor/mu_code/flask app 1/templates
2019-09-28 15:33:24,571 - mu.interface.main:273(get_load_path) DEBUG: Getting load path: 
2019-09-28 15:40:52,037 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/sean.tibor/Library/Logs/mu/mu.log
2019-09-28 15:40:52,037 - mu.logic:163(installed_packages) INFO: Packages found: []

ntoll, 2021-07-10:
Hi @seantibor :wave:

You mention you could create a PR that provides a fix for this problem - given the outline of your solution (which I agree with).

Care to do that..? (I'm all about encouraging folks to contribute and have such contributions celebrated!).

Quite understand if not, and if that's the case, please do let me know so I can create a fix instead.

Best wishes from the UK..! :+1:

seantibor, 2021-07-23:
Hi @ntoll! It's been a while since I opened this, so let me take a look at it again and see if I still remember what I had in mind. I'll see if I can get a PR going this weekend.