mu-editor/mu #925: macOS packaged Mu has hardcoded "1.0" version information

tmontes, 2019-10-01:
## The issue

As we move towards a non "1.0" release, I realised that the macOS packaged Mu contains hardcoded metadata in what concerns its version.

In macOS, Mu is distributed as an *application bundle*, something that looks like a file but is actually a directory structure.

Using the **Finder** > **File** > **Get Info** (or hitting CMD-I on the keyboard) on such *application bundles* brings up a small informational window on whichever "file" is selected. When such "files" are *application bundles*, the application version is shown, as illustrated in this image:

[image]

Note how "Version: 1.0" is displayed. It should, instead, display whichever version is set in Mu's single source of truth for that -- the `mu/__init__.py` file -- much like the window title displays when Mu is running. In this case, "1.1.0.alpha.2".

## What I did so far

A cursory glance at the packaging process, using [`briefcase`](https://pypi.org/project/briefcase/) from the [BeeWare](https://beeware.org) project, shows no indication of something that we might be "doing wrong". Moreover, the `briefcase` docs don't seem to include any information related to setting the *application bundle* version.

Another quick glance at `briefcase`'s execution log and source points to something I wasn't aware of yet: the [`Python-macOS-template`](https://github.com/beeware/Python-macOS-template) project which seems to be a [`cookiecutter`](https://pypi.org/project/cookiecutter/) template. That seems to be the culprit, given that its `Info.plist` file includes a hard-coded `1.0` version [right here](https://github.com/beeware/Python-macOS-template/blob/2.7/%7B%7B%20cookiecutter.dir_name%20%7D%7D/%7B%7B%20cookiecutter.formal_name%20%7D%7D.app/Contents/Info.plist#L20).

PS: The `Info.plist` file is a mandatory file in *application bundles* including, among other things, meta-information about the application such as the version string.

PPS: Another thing I checked was whether the Windows packaged Mu might have the same limitation. Apparently it does not: even though the file properties themselves do not seem to include a version (shows up as blank), the correct version is displayed in the installed software list/dialog.

## What next

Feels like both the `briefcase` and `Python-macOS-template` projects might need improvements to support fixing this on Mu's side.

I'll try to follow up on this, but will be happy for anyone else to chime in, too.

tmontes, 2019-10-01:
Progress:
* Found out the culprit, confirmed/fixed it locally, and [submitted a PR](https://github.com/beeware/Python-macOS-template/pull/3) to the `Python-macOS-template` BeeWare project.
* If/when merged, packagers will need to ensure that no local cache of such template exists (found at `$HOME/.cookiecutters/Python-macOS-template/` on my system) -- this might prompt a few documentation improvements.

Related thoughts:
* Not sure how the [currently distributed macOS DMG](https://github.com/mu-editor/mu/releases/download/1.1.0-alpha.2/mu-editor_1.1.0-alpha.2_osx.dmg) file is produced but I see a `./package/dmg_settings.py` file which induces [`dmgbuild`](https://pypi.org/project/dmgbuild/) is being used -- and I do seem to remember finding some references to it in the context of a few improvements I contributed too, in the past.
* Maybe the current `make macos` target could not only produce the *application bundle* but also the DMG file.
* Moreover, given that `briefcase` will build a DMG file if passed the `--build` flag, I feel we might evaluate whether just using that is enough or not -- I suppose not, given that the current DMG includes a "licence notice" which I'm not sure `briefcase`-produced DMGs include: but that might be an opportunity to improve `briefcase`, sharing that improved capability with a wider audience, and relieving Mu's codebase and CI infra from that burden.

A tangential notes:
* We're currently locking `briefcase` to version `0.2.9`, which is related to #893. 
* A newer `0.2.10` `briefcase` version seems to require the full installation of XCode which from my standpoint is a regression given that, if I comment out that check, everything still works on my system with no full install of XCode. I'll log an issue there and see if the underlying motive for such requirement is clear or if it can be reset to something lighter.

tmontes, 2019-10-01:
Progress:
* 2 PRs and 1 Issue raised.
* Let's see how they evolve and get back to fit any improvements into Mu.

tmontes, 2019-10-02:
Progress:
* 2 PRs merged.
* 1 Issue (admittedly, tangential to this topic) still outstanding.

Next:
* Update docs explaining how to ensure that the just merged PRs in other projects have effect on Mu's macOS packaging -- there's some caching going on which Mu's packagers need to be aware of.

Kind of related but maybe in a different context / issue:
* Wait progress on [outstanding issue](https://github.com/beeware/briefcase/issues/270).
* Push / hope [this `dmgbuild` PR](https://github.com/al45tair/dmgbuild/pull/14) is merged.
* Maybe issue second PR for `briefcase` improvement, simplifying Mu's macOS packaging procedure / burden even more.
* Update Mu's macOS packaging dependency versions in `setup.py`.

tmontes, 2019-10-03:
Progress:
* Confirmed that the `briefcase` caching I mentioned before is non-problematic: the fact that [this `Python-macOS-template` PR](https://github.com/beeware/Python-macOS-template/pull/3) has been merged is enough to ensure future macOS Mu application bundlers include the correct version. :)
* No need to modify Mu and/or documentation.

Related:
* Logged two issues in `briefcase`:
  * Minor inconvenience "dragons ahead warning" with [too strict requirements here](https://github.com/beeware/briefcase/issues/270).
  * Possible improvements to simplify Mu packaging efforts [here](https://github.com/beeware/briefcase/issues/273) -- maybe signing capabilities should be added, @ntoll?
* Identified two more serious "dragons ahead warning" things:
  * `briefcase` will see a revamp in the future, I will try to collaborate, but things are bound to change with regards to Mu's macOS packaging -- see [this PR here](https://github.com/beeware/briefcase/pull/266) -- I do hope this does not compromise the progress that has been made in having single sources of truth for many Mu install/develop/test/package related things.
  * `dmgbuild` used by `briefcase` and, I suppose, by @ntoll when readying a Mu release for macOS distribution is currently "broken" on Python 3 -- there is a [working PR here](https://github.com/al45tair/dmgbuild/pull/14) but having it merged is taking its time...

With all of this said, I'll now be closing this issue, while trying to keep an eye on the evolution of `briefcase` and `dmgbuild`.

tmontes, 2019-10-03:
**Afterthought**: no I won't close the issue yet. Let's have someone else confirm the issue is indeed fixed, before that. (reaching out to gitter)

tmontes, 2019-10-07:
Tangential update:
* An updated [`dmgbuild==1.3.3`](https://pypi.org/project/dmgbuild/1.3.3/) has just been released: this now let's us do the full Mu macOS packaging on Python 3. :)

carlosperate, 2020-11-08:
Sounds like https://github.com/mu-editor/mu/issues/744 is the same issue as listed in https://github.com/mu-editor/mu/issues/744. Since this one contains a lot more info I'll close https://github.com/mu-editor/mu/issues/744 and we'll have check that portable version has the fix for this included.

tmontes, 2020-11-17:
QUICK NOTE:
* I expect this issue to be solved once we use `pup` (https://github.com/mu-editor/pup) to package Mu.
* Much like #940.

dybber, 2021-02-01:
@tmontes, is this resolved, now that we've changed to Mu?

core-code, 2021-02-01:
i see that version 1.0.3 now has a proper *CFBundleShortVersionString*  thanks there! so this is much nicer for the user now

i'd still recommending also changing *CFBundleVersion* which is still at 1 and makes the release indistinguishable for macOS from other versions that also have just "1"

tmontes, 2021-02-01:
> @tmontes, is this resolved, now that we've changed to Mu?

Looks like it, `pup` propagates the distribution's version from `setup.py`:

[image]

closed 2021-02-01 by tmontes