mu-editor/mu #931: Create a "constitution" document that describes the governance of the Mu project.

ntoll, 2019-10-08:
After an email discussion with @tmontes , I guess we need a constitution for Mu so folks can see what rules (both technical and community) we follow, what we expect from contributors and so we can hold ourselves to account.

I'm very much for making things open and clear.

At the very least it should include:

* Rules of the road we (as admins) follow,
* Our expectations of contributors,
* What non-admin contributors can expect from us,
* What folks should do if there is a problem, or when things go wrong,
* How we will deal with sensitive issues (that require privacy etc),
* The process for changing our constitution.

This should be placed in a `constitution.rst` file in the `docs` folder and linked to from the front page of our docs.

As always, I hope this will be an open, collaborative and supportive process. I've assigned this to all current admins for this project. If you'd rather not take part in this process, please do let us know via a comment on this ticket.

ntoll, 2019-10-08:
I have a meeting right now, but will create an initial draft with PR relating to this, later today.

Once the PR is submitted, let's keep discussion about this subject on the PR (this issue being just a flag for admins to see).

tjg-global, 2019-10-08:
+1 on the idea. I'm not sure if this doc is the place, but I would also suggest that we indicate somewhere the likely pace of response / development. This helps would-be contributors and issue-creators to understand that, eg, they might not get an immediate response and/or what release cadence is to be expected.

This is obviously related to the fact of our being volunteers, but you can clearly have volunteers with a lot of time and energy and those with not so much.

ie I'm trying to distinguish in readers' minds between "This is my dayjob" and "I might get a couple of hours every other weekend to look at this" and everything in between

tjguk, 2019-10-08:
(Sorry; forget I was logged in under my work account; hopefully everyone can work out that was me!)

ntoll, 2019-10-08:
@tjguk -- first off, I like the omnipotence suggested by your "work" account. :wink: 

Secondly, yes... I believe the constitution is definitely the place where we can manage expectations about such things as response times, effort, availability, treating volunteers well, and all that other sort of stuff.

ntoll, 2019-10-08:
OK folks, PR open for discussion over here: #932 

All feedback, contributions, constructive critique and ideas most welcome.