mu-editor/mu #934: macOS | Crash when "clicking-away" a REPL tooltip

tmontes, 2019-10-08:
## The horrible crash

* What you were trying to do,

  * Open REPL.
  * Type some code and a tool tip shows up at some point.
  * Click somewhere outside of the tool tip, but within the REPL, to "close away" the tooltip.

* What steps you took to make this happen,

  * Open REPL.
  * Type in "import sys, subprocess"
  * Type in "subprocess.run(".
  * At this point, a very large tooltip is displayed.
  * Use the mouse cursor to click outside the tool tip, but within the REPL pane.

* What you expected to happen,

  * The tool tip to close.

* What actually happened,

  * Mu crashed.
  * Launched from the command line a stack trace is printed:
```
Traceback (most recent call last):
  File "/Users/tiago.montes/Work/Python/github.com/mu/env/lib/python3.6/site-packages/qtconsole/call_tip_widget.py", line 89, in hideEvent
    self._cursor_position_changed)
TypeError: 'method' object is not connected
```

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

  * Mu 1.1.0a2, commit b341cec, launched from command line under development venv.
  * The packaged 1.1.0a2 version at https://codewith.mu/en/download behaves the same.

## Parting words

* Came here in a hurry just to have it logged.
* Does not fail on Windows.
* Did not try on Linux.
* A kind of similar failure can be seen [here](https://github.com/spyder-ide/spyder/issues/10050), where updating `qtconsole` seems to have solved it.
* We might need to address #893, thus.

tmontes, 2019-10-09:
Progress:
* Confirmed it does not fail on Linux (Ubuntu 18.04, at least).
* Logged [this `qtconsole` issue](https://github.com/jupyter/qtconsole/issues/383).

carlosperate, 2020-11-10:
Looks like this might have been fixed in qtconsole, so the next time we update the dependencies we should check if this issue is resolved

dybber, 2021-02-02:
I can replicate the problem in the released alpha-version, but I can no longer replicate it on our current master.

I can see that the PR fixing the issue was merged on April 28th, and the version of qtconsole that we use now (qtconsole==4.7.4) was released on May 13th.

I thus think we can close this.

closed 2021-02-02 by dybber