mu-editor/mu #935: mu-editor only works in admin account. (Ubuntu 18.04)

smnbldwn, 2019-10-09:
I installed mu-editor on Ubuntu Mate 18.04 (64bit) as a snap (snap install --edge mu-editor). The install worked but the program only runs on the admin account. 

On other accounts the mu-editor icon appears in the program menu but it doesn't run. 

When I tried to run it from the terminal, I got this error message:
"**ERROR: ld.so: object 'libgtk-nocsd.do.0' from LD_PRELOAD cannot be preloaded (failed to map segment from shared object): ignored. cannot create user data directory: /home/local/usergroupx/ userx/snap/mu-editor/4: Permission denied**"

This means I can only run mu-editor from the admin account I installed it from, this is a problem as I am in a school and students access the machine on their own (non-admin) accounts.

ntoll, 2019-10-09:
cc @bennuttall 

As mentioned in the Gitter chat, the snap package was an experiment and there has been no further development of it since the initial experiment.

I (personally) don't know anything about snap as a package manager. I've cc'd Ben, who did the initial snap related work. Perhaps he can point you in the right direction..?

bennuttall, 2019-10-09:
Thanks for the report @smnbldwn. I should take another look at the snap config to fix up a few issues but haven't had a lot of spare time lately. @flexiondotorg can you spot something obvious that can be applied to resolve this particular issue?

closed 2021-06-22 by smnbldwn