mu-editor/mu #936: Text colour on OSX broken after Catalina update.

ntoll, 2019-10-09:
User @jasonjones13 reports that the text colour in Mu is always black, no matter which theme is selected (see attached screenies). To recreate this bug, use Mu 1.0.2 on the latest Catalina version of OSX. The user reports Mu worked fine *before* the update.

In addition, it appears that the REPL pane doesn't update its colour scheme as themes switch.

I wonder if this is a Qt bug..? (Google isn't turning up any obvious bugs)

ntoll, 2019-10-09:
[image]
[image]
[image]

tmontes, 2019-10-09:
Not much time now, but Mu 1.0.2 is packaged with older PyQt5 / qtconsole versions. The published alpha, packaged with more up to date versions, might show different behaviour. And, again, even more up to date versions of such packages (#893 related) might even show a different behaviour...

Can @jasonjones13 try out the alpha at https://codewith.mu/ ? :)

ntoll, 2019-10-09:
I'm in the middle of updating Mary's Macbook to Catalina, so should be able to investigate further once it's complete 9currently the screen has been saying around 18 minutes for the last half hour). ;-)

jsully22, 2019-10-09:
I have the alpha version, this bug is also an issue that I am having. Mu 1.1.0.aphla.2

jsully22, 2019-10-09:
To reiterate, I only started having this problem when I did my OS update to catalina, as ntoll stated.

ntoll, 2019-10-09:
OK... so I've tested things and it looks like we need to update to the latest version of Qt (see: #893). This will be fixed in the next upcoming alpha version of Mu.

I can confirm it works correctly with the most recent version of PyQt on Catalina. In the meantime, please use day mode for now.

jasonjones13, 2019-10-10:
> @jsully22 <https://github.com/jsully22> once #937
> <https://github.com/mu-editor/mu/pull/937> lands and we cut a release,
> it'll be fixed.

any idea when the next alpha version of Mu is due to release?

bddicken, 2019-10-11:
:+1: I just had a student of mine come by my office with this issue, and thankfully I found this issue. Hopefully it is fixed soon!

makermelissa, 2019-12-13:
Any ETA on this fix?

partsoven, 2020-01-13:
I also had Syntax highlighting disappear when I upgraded to Catalina. Anyone have a fix for this?

bddicken, 2020-01-14:
I know i've already commented, but just going to +1 that it would be great to get this fixed soon. A new semester starts tomorrow, and would really like to avoid complaints from students about the syntax highlighting. Thanks in advance.

TG-Techie, 2020-01-16:
I also have this same problem and am willing to beta test if need be. A speedy fix would be very much appreciated, or a stop-gap workaround would also be fine. Thanks for the awesome editor.

ntoll, 2020-01-16:
Folks... I'm going to cut a 1.0.3 release in the next week or so with a fix for this (along with a few other niggly bugs that have been fixed in the 1.1.0 alpha releases).

bddicken, 2020-01-16:
That's great new, thanks @ntoll.

ntoll, 2020-01-19:
OK... I spent today figuring this out, and the result can be seen below. Ironically, because we try to support as "old" a version of OSX as possible (since many educational establishments have old gear) our packing / build CI process broke because things were out-of-date. In the end I had to add a command to re-compile Ruby on the Travis image we use..!

I want to fix a few other outstanding (common) bugs in the 1.0.* version of Mu, before doing a 1.0.3 release, but I expect this to be done by the end of the coming week.

[image]

If you want to try the build I created, check out the latest zip file (containing the `.app`) in the following S3 bucket: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/ (NOTE: this is an unsigned app so you have to use the CTRL-click / open trick **twice** for it to actually open. The final release will be signed by me). 

As always, feedback most welcome.

Salemone, 2020-01-20:
Fail to run this version(mu-editor_2020-01-19_21_08_1.0-mainline_68e24a4) on macOS Catalina 10.15.2 with the CTRL-click / open trick, following message found in system.log:

`Jan 20 12:03:34 MacBook-Air-2 com.apple.xpc.launchd[1] (mu.codewith.editor.mu-editor.10884[14481]): Service exited with abnormal code: 126`
[image]

ntoll, 2020-01-20:
@Salemone thanks for the heads up. I'll look into it.

ntoll, 2020-01-20:
@Salemone out of interest, could you just try starting it again..? Alternatively, could you also try to find and attach Mu's logs to this issue. The log files will be somewhere like: `/Users/<username>/Library/Logs/mu`

Thanks..!

bbs4murai, 2020-01-21:
@ntoll I ran the `mu-editor_2020-01-19_21_08_1.0-mainline_68e24a4` version. I can confirm that the syntax highlighting is working properly. The title bar however does not conform to darkmode as the rest of the OS though. Thank you for all your work and efforts with MU. Please keep up the great work! I'm looking forward to the 1.0.3 release.

[image]

ntoll, 2020-01-21:
@bbs4murai I'm afraid the OS controls the title bar (not Mu), so this is expected behaviour.

bbs4murai, 2020-01-21:
Ok, no worries.

makermelissa, 2020-01-23:
On the latest version you have there, files that are saved consistently fail pylint with the following errors:
```
myfile.py 131: Final newline missing (missing-final-newline)
myfile.py 131: Unexpected line ending format. There is 'CRLF' while it should be 'LF'. (unexpected-line-ending-format)
```
I usually have to open in another editor and fix.

tjg-global, 2020-01-23:
@makermelissa interesting (and thanks for reporting). Insofar as you can easily find out, do you know whether this is something /new/ (ie only since this latest update) or whether it was always there?

We have code which detects the incoming line feed format and looks to preserve it. I'm fairly sure that the lack of a trailing linefeed is an advertent artefact of our file-saving-with-linefeed algorithm.

tjg-global, 2020-01-23:
Also, are you running pylint outside Mu (either by hand or as part of some automated process)? Or is this Mu's own internal [Check] functionality at work?

makermelissa, 2020-01-23:
Hi, yes it's new and yes I'm running pylint via command line so that it passes in the automated process that it's also run through.

tjguk, 2020-01-23:
That's interesting. I'll ping @ntoll to see what was included in that release because I'm not aware of any line-endy stuff having changed lately.

ntoll, 2020-01-26:
So no line-endy things have changed in this release.

It's out now (announcement here: https://madewith.mu/mu/releases/2020/01/26/mu-1-0-3.html) and it fixes the code rendering problems. Ergo I'll close this ticket. If you find bugs (be they line-ending-y or not) please make new tickets so they can be tracked properly.

Thank you, as always, for all your help in reporting this sort of stuff.

closed 2020-01-26 by ntoll

Salemone, 2020-02-05:
> @Salemone out of interest, could you just try starting it again..? Alternatively, could you also try to find and attach Mu's logs to this issue. The log files will be somewhere like: `/Users/<username>/Library/Logs/mu`
> 
> Thanks..!

@ntoll I was trying 1.0.3 on  macOS Catalina 10.15.3 (19D76), and it did start. 
Here is the log: [mu.log](https://github.com/mu-editor/mu/files/4160717/mu.log)

```
cat /Users/<username>/Library/Logs/mu/mu.log
2020-02-06 00:09:27,897 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
```

tjg-global, 2020-02-05:
It's ironic that the log file can't log its own name!

@Salemone does your username have any non-ASCII characters? (I'm assuming it does...)

Salemone, 2020-02-06:
I don't have any non-ASCII characters in my username. I did some test, and  this issue may be related to “Language&Region” setting in MacOS. 

In previous post, I only have "Chinese, Simplied" in my Preferred Language, the app can not start; but when I added English language and set English  as the Primary language, the app can start normally.

and here is the log (test in English language)[mu.log](https://github.com/mu-editor/mu/files/4163492/mu.log)

Salemone, 2020-02-06:
> I don't have any non-ASCII characters in my username. I did some test, and this issue may be related to “Language&Region” setting in MacOS.
>
> In previous post, I only have "Chinese, Simplied" in my [...]

I did more test and found something interesting:

When I drop the log file in a shell window, you can see there is not any non-ASCII characters in file path:
[image]

When I check the property of the log file in Finder, you can find some difference:
The folder name  "Users" and “Library" displayed in Chinese characters when using  "Chinese, Simplied" as  Preferred Language.

[image]
 
[image]

tmontes, 2020-02-06:
@Salemone,

Thanks for sharing such detailed finding. I looked into the current in development version, and it does not seem affected by the language preference you refer to. Version 1.0.3, however, is. I'm not sure if any fix for it will ever be put out, other than, of course, in what will become 1.1 (currently 1.1.0alpha2).

However, if you feel "brave enough", you can "hack" your Mu installation to mitigate the failure you are observing. Here's what you could do:

* Use a text editor to edit the file at `/Applications/mu-editor.app/Contents/Resources/app/mu/app.py`.
* Navigate to line 66 where you should find the following code:

  ```
      print(_('Logging to {}').format(LOG_FILE))
  ```

* Comment it out by prepending a `#` character. You should end up with:

  ```
      # print(_('Logging to {}').format(LOG_FILE))
  ```

* Save the file and exit your editor.

This "hack" should let the Mu startup process continue and, hopefully, make it work well enough in your environment, with your preferred language. Let us know about your progress, will you? :)

Salemone, 2020-02-06:
> @Salemone,
>
> Thanks for sharing such detailed finding. I looked into the current in development version, and it does not seem affected by the language preference you refer to. Version 1.0.3, however, [...]

@tmontes,
I tried your "hack" and it works perfect 👍
Thank you very much !

tmontes, 2020-02-06:
> I tried your "hack" and it works perfect 👍
> Thank you very much !

Glad it worked for you. Enjoy Mu! :)