mu-editor/mu #938: No output and crash on Mac OS

pkeen, 2019-10-10:
If you are reporting a bug, we would like to know:

* What steps you took to make this happen,

I ran Mu alongside IDLE which was already running when I opened Mu.

* What actually happened,

No output in bottom window when trying to run or debug code.

Eventually the program crashed.

 Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Mu 1.0.2 Mac OS Mojave 10.14.6

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

I'm sorry this maybe a mess to work through:

2019-10-10 15:35:02,183 - mu.logic:702(new) INFO: Added a new tab.
2019-10-10 15:36:09,698 - mu.logic:695(toggle_theme) INFO: Toggle theme to: night
2019-10-10 15:36:13,327 - mu.logic:695(toggle_theme) INFO: Toggle theme to: contrast
2019-10-10 15:36:15,294 - mu.logic:695(toggle_theme) INFO: Toggle theme to: day
2019-10-10 15:36:41,359 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: /Users/pete/mu_code
2019-10-10 15:37:08,988 - mu.interface.main:263(get_save_path) DEBUG: Getting save path: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:08,988 - mu.logic:905(check_for_shadow_module) INFO: Checking path "/Users/pete/docs/mu_code/randomint.py" for shadow module.
2019-10-10 15:37:08,988 - mu.logic:871(save_tab_to_file) INFO: Saving script to: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:08,988 - mu.logic:872(save_tab_to_file) DEBUG: import random

for i in range(5):
    print(random.randint(1,10))
2019-10-10 15:37:15,316 - mu.modes.python3:201(stop_script) DEBUG: Stopping script.
2019-10-10 15:37:15,922 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:15,923 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2019-10-10 15:37:15,923 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 15:37:15,923 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 15:37:15,923 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 15:37:17,240 - mu.modes.python3:201(stop_script) DEBUG: Stopping script.
2019-10-10 15:37:17,823 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:17,823 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2019-10-10 15:37:17,823 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 15:37:17,823 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 15:37:17,824 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 15:37:26,882 - mu.modes.python3:201(stop_script) DEBUG: Stopping script.
2019-10-10 15:37:27,552 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:27,553 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2019-10-10 15:37:27,553 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 15:37:27,553 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 15:37:27,553 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 15:37:37,096 - mu.modes.python3:201(stop_script) DEBUG: Stopping script.
2019-10-10 15:37:37,748 - mu.modes.python3:214(debug) INFO: Starting debug mode.
2019-10-10 15:37:37,778 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/pete/mu_code
2019-10-10 15:37:37,781 - mu.modes.debugger:109(start) DEBUG: import random

for i in range(5):
    print(random.randint(1,10))
2019-10-10 15:37:37,790 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/mu_code/randomint.py
2019-10-10 15:37:37,790 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 15:37:37,791 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 15:37:37,791 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 15:37:42,210 - mu.debugger.client:221(output) DEBUG: Debugger client error.
2019-10-10 15:37:42,211 - mu.debugger.client:222(output) DEBUG: [Errno 32] Broken pipe
2019-10-10 15:37:43,236 - mu.debugger.client:221(output) DEBUG: Debugger client error.
2019-10-10 15:37:43,236 - mu.debugger.client:222(output) DEBUG: [Errno 32] Broken pipe
2019-10-10 15:37:43,751 - mu.debugger.client:221(output) DEBUG: Debugger client error.
[... 267 lines cut ...]
end = 6
while i <= end:
    n =  i
    i += 1
    
print(n)
2019-10-10 18:11:12,430 - mu.logic:1178(autosave) INFO: Autosave detected and saved changes in /Users/pete/docs/projects/mu_code/addingEnd.py.
2019-10-10 18:11:16,441 - mu.modes.python3:201(stop_script) DEBUG: Stopping script.
2019-10-10 18:11:16,960 - mu.logic:871(save_tab_to_file) INFO: Saving script to: /Users/pete/docs/projects/mu_code/addingEnd.py
2019-10-10 18:11:16,960 - mu.logic:872(save_tab_to_file) DEBUG: n = 0
i = 0
end = 6
while i <= end:
    n += i
    i += 1
    
print(n)
2019-10-10 18:11:16,977 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/projects/mu_code/addingEnd.py
2019-10-10 18:11:16,978 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2019-10-10 18:11:16,978 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 18:11:16,978 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 18:11:16,978 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 18:15:43,194 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/pete/Library/Logs/mu/mu.log

Thank you for contributing to Mu! :-)

tmontes, 2019-10-10:
Thanks @pkeen for logging this. I just edited your message for increased log readability. :)

tmontes, 2019-10-10:
Log observation notes:
* Module `helloworld.py`.
* Saved at `/Users/pete/docs/projects/mu_code`.
* Contents is `print('hello world')`.
* Exception while debugging (log snippet):
```
2019-10-10 17:54:35,951 - mu.modes.python3:214(debug) INFO: Starting debug mode.
2019-10-10 17:54:35,983 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/pete/mu_code
2019-10-10 17:54:35,986 - mu.modes.debugger:109(start) DEBUG: print('hello world')
2019-10-10 17:54:35,999 - mu.interface.panes:651(start_process) INFO: Running script: /Users/pete/docs/projects/mu_code/helloworld.py
2019-10-10 17:54:35,999 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-10-10 17:54:36,000 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/pete/mu_code
2019-10-10 17:54:36,000 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-10-10 17:54:39,390 - root:94(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 1244, in debug_toggle_breakpoint
    self.modes['debugger'].toggle_breakpoint(line, tab)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/debugger.py", line 195, in toggle_breakpoint
    bps = self.debugger.breakpoints(tab.path)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/debugger/client.py", line 247, in breakpoints
    return self.bp_index.get(normalised, {})
AttributeError: 'Debugger' object has no attribute 'bp_index'
2019-10-10 17:54:47,505 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-10-10 17:54:47,506 - root:112(run) INFO: uname_result(system='Darwin', node='Petes-MacBook-Pro.local', release='18.7.0', version='Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64', machine='x86_64', processor='i386')
```
* If `AttributeError: 'Debugger' object has no attribute 'bp_index'` was raised...
* ...in `return self.bp_index.get(normalised, {})`  at line 247 of `mu/debugger/client.py`,
* Then it stands to reason that `self.bp_index = {}` at line 317 of the same module, in the `on_bootstrap` method, never ran.

Question (not having the full details of the debugger inner workings in my head, now):
* How is it possible that the debugger is (apparently) running...
* ...but `on_bootstrap` wasn't called?

Certainly prompts further investigation. I could not reproduce the failure on my macOS 10.12.6 system, with the same software versions as @pkeen.

carlosperate, 2021-04-09:
Hi @pkeen,

We've recently released a couple of Mu betas with significant changes. Would you be able to install the latest version and check if you are still experiencing this problem?
https://github.com/mu-editor/mu/releases

Thanks!

carlosperate, 2022-07-16:
Closing due to lack of activity.

closed 2022-07-16 by carlosperate