mu-editor/mu #940: macOS Catalina | New packaging requirements?

tmontes, 2019-10-14:
Just saw this article passing through my regular RSS feeds:

* [Mac Python Distribution Post Updated for Catalina and Notarization](https://glyph.twistedmatrix.com/2019/10/catalina-notarize-python.html)...

Which essentially links to an Catalina updated version of a previous packaging article:

* [Tips And Tricks for Shipping a PyGame App on the Mac](https://glyph.twistedmatrix.com/2018/01/shipping-pygame-mac-app.html).

Both very well written by @glyph, clearly pointing to a "dragons ahead" warning WRT supporting macOS Catalina Mu users. The gist:

* New code signing requirement.
* Needs to be submitted to Apple, somehow.
* A new entitlements file seems to be needed.
* All executable files must now be signed.
* ...and more?

The article uses a different tool for the first packaging step (`py2app` where we use `briefcase`) but from a cursory read the signing / getting the packaged `.app` bundle ready for distribution seems tool agnostic.

PS: macOS Catalina has just been released, earlier this month. As time progresses, we can expect more and more Mu mac users to use Catalina. Actually, #936 is a report from a Catalina user -- wondering how Mu worked for the reporter, there.

glyph, 2019-10-14:
Always nice to hear that my writing is proving useful to people!  Just a few minor notes:

> * Needs to be submitted to Apple, somehow.

The "somehow" are the `altool` and `stapler` invocations in the library - there's no out-of-band communication other than the email notifying you that you *can* run `stapler`.

> * ...and more?

This should really be it :)

> The article uses a different tool for the first packaging step (`py2app` where we use `briefcase`) but from a cursory read the signing / getting the packaged `.app` bundle ready for distribution seems tool agnostic.

Your reading here is correct.  The `zip` / `codesign` / `altool` / `xcrun` invocations are fairly agnostic to how the bundle was actually built; these ought to work with non-Python stuff, for that matter, as long as they happen to require that one extra entitlement that Python needs.

> PS: macOS Catalina has just been released, earlier this month. As time progresses, we can expect more and more Mu mac users to use Catalina. Actually, #936 is a report from a Catalina user -- wondering how Mu worked for the reporter, there.

Upgrade curves on new macOS releases are shockingly fast; you're likely to see a *majority* of users on Catalina in the not too distant future.

tmontes, 2019-10-14:
Thanks for chiming in @glyph, and thanks for the article. We will soon fight this fight. :)

glyph, 2019-11-11:
@tmontes  I’m interested to hear how it goes!

tmontes, 2020-11-17:
A quick update + thought:

* Much progress has been made towards this -- see https://github.com/mu-editor/pup.
* Of course time (and Apple) does not stand still and now we're faced with macOS Big Sur.
* Once our dependencies (in particular Qt) handle Big Sur nicely, I wonder if `pup`'s efforts are enough...

:-)

tmontes, 2021-02-02:
Closing this:

* Addressed in PRs #1234 and #1250, recently merged.
* Needs real-world testing -- but that's the next stage of the adventure. :-)

closed 2021-02-02 by tmontes