mu-editor/mu #941: Using Mu's Python to create and use venv on Windows.

ntoll, 2019-10-14:
So I've been tearing my hair out over this today.

What I'm trying to do:

Using just the Python version installed with Mu, try to create a virtualenv and run scripts from within it by calling the virtualenv's Python executable directly.

I realise that the default Python that comes with Mu does NOT have the necessary `venv`, `ensurepip` and `pip` modules in the expected places. As a result I've just copied them over manually from a regular Python install. These shortcomings can be solved in the same way we ensure `tkinter` is available during packaging.

Two strange things happen:

1. If I try to do `"C:\Program Files\Mu\Python\python.exe" -m venv foo` I get complaints about `ensurepip` not working. This is, I believe due to file system permissions (i.e. as a regular user, I can't write to anything within "C:\Program Files". :-/ Any idea how I might get around this? (HINT: I believe the answer is "you can't".)

2. I can create a virtualenv with: `"C:\Program Files\Mu\Python\python.exe" -m venv --without-pip foo`

However, this second approach doesn't include the expected bin directory and other stuff I'm used to seeing on Linux. I want Mu to use the Python executable found *within* the venv's directory, but it doesn't appear to be there..! Obviously, in order for Mu to run scripts from within the context of this virtualenv I need to use the (apparently missing) Python executable associated with the virtualenv.

In any case, when I `Source\activate` the venv I can see `python --version` fine, but when I try to run a script or drop into the REPL then I get the following type of error message: 

```
Fatal Python error: Py_Initialize: unable to load the file system codec
ImportError: No module named 'encodings'

Current thread 0x00001db4 (most recent call first):
```

As per advice on the web, I've checked my `PYTHONPATH` and `PYTHONHOME` envars (I don't have either). So I'm at a loss as to what could be happening and don't know enough about Python and Windows to know where to start.

My question:

1. How can we workaround/use Mu's version of Python to create a `venv` given the above situation?
2. If this is not possible, what other options could we use?

All this is due to our need for trying to install PyperCard into a venv in order to get around the current problem of it not working if we specify a target directory for `pip`. I've done all the prep work in a code spike on the `use-venv` branch (which works for me on Linux). I wanted to check this approach works on Windows, yet I've not been able to even get Mu's default venv to be created (if not already available) on start up (see: https://github.com/mu-editor/mu/blob/use-venv/mu/logic.py#L673 for how I try to do this). Hence all the faffing about described above.

I realise I may be flogging a dead horse here, but any help, guidance and suggestions would be *really* appreciated. I'm afraid I'm getting more and more despondent about this and my lack of progress on Mu in general. My guess is that I'm stymied, so I'd love to explore other options that may be available.

I hope this is comprehensive enough detail for what I'm trying to achieve and why it's not working and/or impossible. If not, please don't hesitate to give me a poke via this ticket.

As always, thank you in advance for any help, support and ideas you folks may come up with.

tjguk, 2019-10-14:
Just cutting a branch to see what happens when I try... 😀

tjguk, 2019-10-14:
Well, the good news is that, by tweaking just one string, I've been able to install openpyxl into a mu-created virtualenv. I haven't gone any further than that for now (no more time this evening) but the key thing you need to know is that, in a venv on Windows, the directory which holds all the binaries is called "scripts/" and not "bin/". To test, I've simply changed it directly in line 54 of logic.py but obviously some conditional logic would be needed

tjguk, 2019-10-14:
NB I'm sure this isn't the end of the story. This was in a dev setup, ie an existing virtualenv. I haven't tried building an installer and running from there.

tmontes, 2019-10-14:
@ntoll, @tjguk,

Took this for a spin. From source, with @tjguk's fix (hardcoded, too), managed to install `pypercard` on both macOS and Windows, successfully running a minimal, single card application.

Now fighting a bit with the Windows installer generation, stupidly complaining that it can't find the icon file that is precisely in the path it says it isn't... Hmmm, you have to love packaging! :)

More feedback soon (I hope!)

tmontes, 2019-10-14:
Packaged macOS result:
* Non successful.
* Motive: venv was created, but bundled Python does not include `pip`, so venv has no `pip`!
* Message in Mu's dialog box: `.../Library/Application Support/mu/mu_venv/bin/python3: No module named pip`
* Bundled Python is `https://github.com/mu-editor/mu_portable_python_macos/releases/download/0.0.6/python3-reduced.tar.gz`.
* Also tried `briefcase`'s default Python, hoping with wouldn't be as reduced but it failed too, in the same exact way.

Planned fix:
* Have our bundled Python include `pip`. :)

tmontes, 2019-10-14:
Packaged (and installed) Windows result:
* Non successful.
* Motive: venv was not created.
* Crashes when clicking the cog wheel button.
* Log contains the following interesting messages:
```
... Creating virtualenv: C:\Users\test\AppData\Local\python\mu\mu_venv
... Could not create virtualenv.
... Command ['C:\Users\test\AppData\Local\Mu\Python\pythonw.exe', '-m', 'venv', 'C:\Users\test\AppData\Local\python\mu\mu_venv'] returned non-zero exit status 1.
```
(messages copied by hand, may have minor typos -- copy/paste from VM not active)
* Motive: bundled Python does not have the `venv` module (nor `pip`/`setuptools`, AFAICT).

Planned fix:
* Have our bundled Python include `venv`, `pip`, `setuptools`. :)

tmontes, 2019-10-14:
Lastly, on a tangent, running from source:
* `Run` and `Debug` seem to be using the venv's Python. Cool! :)
* The `REPL` is still using Mu's Python but, apparently, with the venv's site-packages added to `sys.path`. I did not look at the code... Is this planned? Feels a bit brittle...

(over and out, for the day... but this looks promising!!!) :)

tmontes, 2019-10-14:
Prologue (for the day)
* I fiddled around with this a bit more.
* On macOS packaged Mu, things seem to kind of work if the REPL pane is open. In this case, the installed package list includes all of Mu's own packages. Installed pypercard sucessfully and ran a minimal single card app. Debug fails. Not sure where the packages are actually being installed. Didn't investigate further.
* On Windows I'm not so sure. But I know that at some point (no code/packaging changes), after clicking/opening/closing things around, I managed to install something... (????)

(now over and out for the day... I need some sleep) :)

PS: Great work @ntoll! :)

ntoll, 2019-10-15:
@tjguk @tmontes thank you so much for your efforts with this and I'm so relieved that things actually work :tm:. I have to admit that by the end of yesterday I wasn't in a good place -- a day of frustrations which ended in apparent failure. Thanks to you folks I see I was so close (but couldn't see the wood for the trees).

@tjguk why (oh why???) is the venv layout different on Windows than on *nix..? I'm sure there's a good reason (which I can't currently see), so I'd love to know more if at all possible.

@tmontes regarding the REPL using Mu's Python with the venv's site-packages: that is brittle, but I'm yet to figure out how to get the qtconsole kernel to run with a specific Python interpreter. I was in the "make it work" phase, but it now looks like we're moving to the "make it work properly" effort. ;-) My guess is @takluyver probably knows the answer to this question (it's not immediately obvious from the documentation here [https://qtconsole.readthedocs.io/en/stable/config_options.html] how to do this... I suspect `kernel_name` has something to do with it since this is supposed to replace `kernel_cmdList` which appears to be the things we need). ;-)

@takluyver, any pointers here..? TL;DR, how do I specify the Python interpreter to use for the kernel used by the qtconsole REPL..? I just tried setting the `kernel_name` to point to the wanted Python interpreter but I've obviously misunderstood what it's for since I get the following message:

```
jupyter_client.kernelspec.NoSuchKernel: No such kernel named /home/ntoll/.local/share/mu/mu_venv/bin/python3
```

Obviously, `kernel_name` refers to a type of kernel (e.g. "python"). But then I assume I need to get some sort of reference to this kernel instance and tell it to use my virtualenv's interpreter. How..? (Thank you!!!)

tjg-global, 2019-10-15:
Hey @ntoll, glad we could lift you up a little :) There's still obviously a ways to go here but at least there is a way to go!

re the venv layout: AFAIK the answer goes back to the fact that Python's *own* layout is different on *nix & Windows. (Very different). So when virtualenv -- which was, and is, a 3rd-party product -- was developed it presumably made sense to the developer[s] to echo the layout on the respective systems: /bin for *nix; /scripts for Windows. When venv was implemented in Python 3, the development obviously followed the well-established layout.

It's an example of something you see all over the Python world, and presumably in other cross-platform ecosystems, definitely including GUI frameworks: do you try to keep *your* environment consistent, no matter what the surrounding environment; or do you try to follow the surrounding environment, regardless of possible confusion to the cross-platform users of your environment?

At one point a few years ago Van Lindberg was pushing quite hard for core Python to switch to /bin on Windows, but got pushback -- not least from me in a somewhat heated discussion at PyCon UK :)

ntoll, 2019-10-15:
@tjguk (or should that be @tjg-global?), yes I needed a lift after yesterday and I agree there's still a lot to do, not least getting the REPL to work correctly with the venv. I've spent some time looking into this but the Jupyter docs are *huge* and that one essential piece of information I need (how to pass in the path to the Python executable to the Python3 kernel) still eludes me. I've ended up asking on their Gitter forums and I hope when the west coast wakes up someone might be kind enough to point me in the right direction. (Aside, this isn't an easy thing to Google, since there are so many name collisions in the Jupyter/IPython "namespace" -- due to the nature of the project).

Thank you also for explaining the venv layout. This reminds me of that saying.... "when in Rome" etc.

The questions you pose about cross platform ecosystems are indeed difficult to answer. In an ideal world there would be a sort of line in the stack of abstractions. Above it, everything thing is the same across platforms. Below it, things become more "native". Of course, such an ideal world doesn't exist and, to be honest with you, I like the fact that so many flowers "bloom" (as in why the "blooming" hell is it like that?). ;-) I also feel it's a sort of a sign of respect that cross platform ecosystems accommodate such quirks. Diversity and choice are (generally) a good thing (except perhaps when it comes to root canal options).

In any case, I thought it'd be an interesting story and part of me is relieved you pushed back at Van's suggestion. While the Zen of Python talks about there being one obvious way to do things, that depends entirely upon your point of view of what the obvious way is.

:-)

Regarding packaging the other things needed for this thing to land (cc/@tmontes), I believe the process by which we can make this happen is already in place (via the scripts used to make tkinter work). We just need to expand their scope a bit.

As you may tell from the tone of this reply, I'm a lot more hopeful this afternoon.

As always, thank you all for your help..!

takluyver, 2019-10-15:
>  how do I specify the Python interpreter to use for the kernel used by the qtconsole REPL..? I just tried setting the kernel_name to point to the wanted Python interpreter but I've obviously misunderstood what it's for since I get the following message:

The kernel name either refers to a [kernel spec](https://jupyter-client.readthedocs.io/en/stable/kernels.html#kernel-specs), which will describe a path to a Python interpreter, or the `python3` kernel will fall back to using the same interpreter as the process trying to start it (if there's no `python3` kernel spec).

To run a kernel in an environment, you have two options:

- Create a kernelspec referring to that environment (e.g. `env/bin/python -m ipykernel install --user --name mu-env --display-name "Python (Mu virtualenv)"`), and use the name you gave it (e.g. `mu-env`) to launch the kernel.
- Launch a kernel manually inside the environment (`env/bin/python -m ipykernel`), and then connect the console to that. There's a `--existing` flag when using the console as a standalone application; I'm not sure what the API is like for embedding it.

This is less than ideal for working with environments, I know. When we designed this mechanism, we were imagining one kernel per language (or maybe per major version of a language). It was a long discussion and by the time we got to considering environments, we were all too tired to want to tackle it. We're working to make it more flexible now, but it's hard to change something like this.

ntoll, 2019-10-15:
@takluyver as I mentioned over on the Jupyter Gitter channel, you're an evergreen, always helpful and formidably talented chap, and I can only but thank you (again (again (again))) for your help with the Mu/Jupyter cross over. My hope is folks find it easier to migrate to Jupyter from Mu as a result of this work we're putting in.

My guess is that creating a kernelspec, as you first suggest, is the best way to do this. Can I assume some sort of black box will figure out where to find such a kernel spec..?

That's tomorrow's efforts sorted out..! :-)

Finally @takluyver, I'll be in Edinburgh on December 4th to give a presentation at the BCS's evening meeting. I believe it's a free event and if you're about, I'd love the opportunity to buy you a beer as a token of thanks. I'm talking about programming education related stuff (as usual). It'll be a bit different from when you last saw me speak because I hope to show examples of beginners' work and the steps we needed to put in place to help them get so far with their CodeGrades.

takluyver, 2019-10-15:
Thanks @ntoll! It would be great to see you again, and I'll try to ensure I'm around then. I have a nagging feeling that it's me who owes you a beer from last time, though. :slightly_smiling_face: :beers: 

> Can I assume some sort of black box will figure out where to find such a kernel spec..?

There's a search path, a lot like for Python imports: user kernelspecs under the home directory take precedence over locations in the Python env where the frontend is running, and systemwide locations have lowest priority. The `--user` option in the example command I gave would do a user install.

If you want it to be more self-contained, you can also add to the search path with the `JUPYTER_PATH` environment variable, following common conventions for search paths - directories separated by `os.pathsep`. These locations have highest priority. If a directory in the search path contains a file like `<directory>/kernels/mu-env/kernel.json`, that defines a kernel called `mu-env`.

ntoll, 2019-10-16:
@tjguk @tmontes hmm... I feel I'm going around in circles on Windows and I think your original solutions were "false positives". Try this:

1) Install Mu from an installer (e.g. the latest alpha).
2) Copy over the `venv`, `ensurepip` and `pip` modules into the expected places in Mu's installed version of Python 3.6.2 (i.e. into the `\Python\python36.zip` directory).
3) From the command line create a virtualenv with the Python interpreter that comes **packaged with Mu**. This is the important bit. DO NOT in any way allow your system Python to do this, or do it via running Mu in a virtualenv. For example:

```
C:\Program Files\Mu\Python> python -m venv --without-pip C:\Users\ntoll\test_venv
```

4) Within the new virtualenv, try running the interpreter. I still get:

```
C:\Users\ntoll\test_venv\Scripts> python
Fatal Python error: Py_Initialize: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00001eec (most recent call first):
```

I believe you folks may have been running Mu from within a virtualenv or with your installed system Python. I believe the problem is down to us using the version of Python packaged with Mu.

Why is this important? Well, the version of Python packaged with Mu is what Mu will use as `sys.executable` when making the new virtualenv. As you can see, it results in the errors I describe above.

Help!?!?! Can you reproduce what I'm seeing? If so, any idea how to fix this?

ntoll, 2019-10-16:
In any case... thanks for the rubber duck :duck: :bug: opportunity. Please feel free to tell me the many number of ways I'm being an idiot about all this and there's actually a really simple way to do this sort of thing by doing X, Y and Z instead.

tjg-global, 2019-10-16:
@ntoll Glad to hear there's a solution. If I'm honest, it feels hacky and I'm sure there must be A Better Way (tm). But if so be as I ever work out what that would be, I'll raise a PR 😄 . For now, let's just celebrate a working solution 🍾

ntoll, 2019-10-16:
Finally (sorry), why is `python36.zip` not actually a zip file..? See the comments at the start of this NSIS related file: https://github.com/mu-editor/mu/blob/master/package/mu_nsist.py#L8 ;-)

ntoll, 2019-10-16:
@tjg-global quite. As you'll see in the code (I'll probably finish tomorrow since I'm out playing in an orchestra this evening), I'll make sure it's commented and explained. :+1:

ntoll, 2019-10-18:
OK... updated code on the https://github.com/mu-editor/mu/tree/use-venv branch. **UNTESTED**

tjguk, 2020-12-17:
Closed by #1072

closed 2020-12-17 by tjguk